espressif/esp-idf #6728: RS485 Half-Duplex Mode RTS management issue

madmax4889, 2021-03-16:
I believe I found an issue with the way RTS is managed when a UART is in RS485 Half Duplex mode.

More specifically, the problem happens if a call to wait_tx_done() times-out before the TX_DONE interrupt has fired. The problem seems to be that the wait_tx_done() function will disable the TX_DONE interrupt if it times-out. Since the TX_DONE interrupt handling code is the only place where RTS is set to 1, the consequence of disabling the TX_DONE interrupt is that RTS stays at 0, which means that the transceiver's Driver Output to remain enabled even though transmission is complete.

alisitsyn, 2021-03-17:
@madmax4889,

You are right about the reason for this issue. This issue is known and the MR has been opened to fix this. Unfortunately there is a delay with merging of these MRs. 
The options below can be used to fix this issue in you project :
1. Change the `wait_tx_done()` call to `uart_wait_tx_idle_polling()` to ensure transmission is done. In this case the task is not blocked during waiting.
2. Change the timeout define components/freemodbus/port/port.h::MB_SERIAL_TX_TOUT_TICKS  to maximum time for transmission of longest allowed frame buffer can be changed to 2200 ms for example.
3. Chenge the `wait_tx_done()` function like below to not clear the TX_DONE interrupt when timeout triggered:
```
/**************************************************************/
esp_err_t uart_wait_tx_done(uart_port_t uart_num, TickType_t ticks_to_wait)
{
  ...
    //take 2nd tx_done_sem, wait given from ISR
    res = xSemaphoreTake(p_uart_obj[uart_num]->tx_done_sem, (portTickType)ticks_to_wait);
    if(res == pdFALSE) {
        UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
        uart_hal_disable_intr_mask(&(uart_context[uart_num].hal), UART_INTR_TX_DONE);
// =================== change is below =========================
        // The workaround: if the RS485 mode is used the TX_DONE interrupt
        // should not be disabled after timeout waiting transmission done
        // (TX_DONE interrupt is used to control RTS line after transmission in RS485 mode)
        if (!UART_IS_MODE_SET(uart_num, UART_MODE_RS485_HALF_DUPLEX)) {
            uart_hal_disable_intr_mask(&(uart_context[uart_num].hal), UART_INTR_TX_DONE);
        }
        UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
        xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
        return ESP_ERR_TIMEOUT;
    }
    xSemaphoreGive(p_uart_obj[uart_num]->tx_mux);
    return ESP_OK;
}
```
The patch is below:
[patch to fix wait_tx_done issue](https://github.com/espressif/esp-idf/files/6155139/0001_fix_uart_wait_tx_done_rs485.diff.txt)

Please apply changes and let me know the results of testing. Thank you.

madmax4889, 2021-03-17:
Thanks @alisitsyn,

Your message confirms the workaround(s) that I had been considering.

alisitsyn, 2021-03-17:
@madmax4889,

Unfortunately I can not let you know when the MR will be merged. I can inform once it is merged.

Alvin1Zhang, 2021-07-27:
Thanks for reporting and sorry for late reply, the fix is available https://github.com/espressif/esp-idf/commit/a1b5c382181fa26844b201684dcaa89556696044, feel free to reopen.

closed 2021-07-27 by Alvin1Zhang

umair-jarullah, 2025-10-30:
@alisitsyn 
> 1. Change the `wait_tx_done()` call to `uart_wait_tx_idle_polling()` to ensure transmission is done. In this case the task is not blocked during waiting.

I just want to clarify the statement “the task is not blocked during waiting.”
Since uart_wait_tx_idle_polling() uses a while (!uart_hal_is_tx_idle(...)); loop, it keeps the CPU busy until transmission completes.
So while the task isn’t blocked (in the RTOS sense, i.e., it doesn’t yield), it still occupies the CPU entirely.
Did you mean “not blocked by a semaphore” or “non-blocking in terms of CPU usage”?
Just want to ensure I understand your intended behavior correctly.