espressif/esp-idf #6729: Can't set file metadata (modification date, creation date, access date) on ESP32

jonathan84clark, 2021-03-16:
I need to set the creation date, modified date and access date on files saved to the SD card. I attempted to use the following method without success:

`    int year = 2021;
	int month = 3;
	int mday = 15;
	int hour = 8;
	int min = 6;
	int sec = 0;
	
    FILINFO fno;

    fno.fdate = (WORD)(((year - 1980) * 512U) | month * 32U | mday);
    fno.ftime = (WORD)(hour * 2048U | min * 32U | sec / 2U);

    f_utime("/sd", &fno);`

I am using this code right after I initialize the SD card for use. But I still show a creation date on all my files of December 31, 1970.

Any help  on this would be great!

Thanks,

Jonathan L Clark

igrr, 2021-03-16:
Hi Jonathan, there are a couple of things that may help:

1. We recommend using C standard library or POSIX functions when working with filesystems. In this case, you can use [`utime`](https://linux.die.net/man/2/utime). The benefit of using such functions is that your code doesn't need to change if you decide to switch from FAT to some other supported filesystem type later (like SPIFFS). It also makes your code more portable, potentially allowing you to test parts of your code on PC.
2. Neither `f_utime` nor `utime` can change the file attributes recursively. I think if you check the return value of f_utime, you will find that it is non-zero, indicating that the function hasn't completed successfully.
   To change the file attributes recursively, you need to implement a loop, iterating over files using `opendir`/`readdir` functions, and calling `utime` for each file. Be sure to check the return values, and look at `errno` if errors occur — this can give you a hint if something goes wrong.

jonathan84clark, 2021-03-16:
Hi iggr,

Thanks for the response. That makes sense, I thought that changing the file metadata had to be done to the entire filesystem, in other words the entire SD card would have a date/time stamp that gets set on boot and all files that get written after that would pickup that time. What you are describing makes a lot more sense. I will give that a shot! Thanks for the help!

igrr, 2021-03-22:
> the entire SD card would have a date/time stamp that gets set on boot and all files that get written after that would pickup that time.

@jonathan84clark You can do that as well — just set the system time using `settimeofday`, and the files created afterwards will have correct timestamps. If you have internet connection, you can set time from SNTP, refer to examples/protocols/sntp.

tom-borcin, 2021-07-20:
Hi @jonathan84clark, did you manage to resolve your issue following the steps provided by @igrr?

jonathan84clark, 2021-07-20:
Yes, thanks for the help! Setting the file time stamp with utime worked perfectly.

tom-borcin, 2021-07-21:
Thank you @jonathan84clark, I am closing the issue now.

closed 2021-07-21 by tom-borcin