espressif/esp-idf #6730: esp-idf Docker on armv7f build error

iot49, 2021-03-16:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.2
- Build System:         docker build
- Operating System:     Linux|macOS

## Problem Description

Building an esp-idf v4.2 docker image on an Ubuntu armv7f cpu raises

```
ERROR: tool cmake does not have versions compatible with platform linux-armel
```

Presumably it's because `esp-idf/tools/tools.json` lacks a `cmake` entry for linux-arm (~ line 380 in the file). This may be since `https://github.com/Kitware/CMake` has no binaries for arm architectures.

Before exploring other ways to build cmake in the Docker image I'd like to check if espressif has plans already or otherwise if there are special requirements for cmake.

### Expected Behavior

Dockerfile uses an approach for building cmake on arm.

### Actual Behavior

Build process fails.

### Steps to reproduce

1. Download [Dockerfile](https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile) and [entrypoint.sh](https://github.com/espressif/esp-idf/blob/master/tools/docker/entrypoint.sh)
2. Run `docker build - < Dockerfile -t "esp-idf:v4.2" --build-arg IDF_CLONE_BRANCH_OR_TAG=release/v4.2`
3. Lock for the error message ...

Alvin1Zhang, 2021-03-17:
Thanks for reporting.

sidwarkd, 2021-09-27:
I would like to +1 this issue because I want to use Raspberry Pis as device programming stations. Currently there is no version of the IDF docker image compatible with arm so this isn't possible. Would be nice if there was an ARM variant of the image to allow running esptool via docker on the Pi.

igrr, 2021-09-27:
Hi @sidwarkd  we are working on a new toolchain release to address this issue. However for the purposes of programming, you should be able to install esptool package on your Raspberry PI without installing the whole esp-idf.

Esptool can be installed using pip: `pip install esptool`.

If you save the project build directory as an artifact, you can then flash the project on an RPi using a command like
```
cd build; esptool.py --port /dev/ttyUSB0 --chip esp32 write_flash @flash_args
```
This will write the bootloader, partition table and the app using the arguments stored in `flash_args` file (produced by IDF build system).

sidwarkd, 2021-09-27:
Thanks for the super fast response @igrr. I was just looking into this. I would love to capture the entire build env in a docker image along with the build artifacts so all a programming station has to do is pull the latest docker image and run the esptool. Thanks for the tip on the `flash_args` I will look into that. In the meantime I'm working on creating a basic python image with just esptool installed in it as you suggest. That should also decrease the size significantly.

sidwarkd, 2021-09-28:
Leaving this here in case it can help someone else. I was able to get the debian:buster-slim image working with esptool installed as a pip package to handle my programming needs. At the time of this writing there is an issue with the debian:bullseye variants that cause lots of errors when run on the Raspberry Pi.