espressif/esp-idf #6732: Websocket frame receiving / unmasking fails

zinke-ct-video, 2021-03-17:
## Environment

- Module or chip used:  ESP32-WROVER-IB
- IDF version: v4.4-dev-479-g1067b2870
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt

## Problem Description

I have a HTTP server using the websocket server component. My websocket handler function receives a new connection, but the actual websocket frame content is malformed.

### Expected Behavior

Websocket frame content should be received as it has been sent by the browser (client).

### Actual Behavior

Websocket frame cannot be received. Two types of error messages occur:
1. Frame is not properly masked
2. Frame could not be received

The content of the websocket frame is invalid. The length of the received frame does not match length of the frame sent by the browser. I have verified this using wireshark.

### Steps to reproduce

1. Start the server
2. Connect a websocket client and send some text data

### Code to reproduce this issue

This is my websocket frame receive handler function:

```cpp

esp_err_t websocket_handler(httpd_req_t *req) {
    httpd_ws_frame_t ws_pkt;
	memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
	ws_pkt.type = HTTPD_WS_TYPE_TEXT;

	// determine the actual packet length
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK) {
    	ESP_LOGE(TAG, "httpd_ws_recv_frame failed: %s", esp_err_to_name(ret));
        return ret;
    }

    ESP_LOGI(TAG, "Websocket input len=%d:", ws_pkt.len);
    ws_pkt.payload = malloc(ws_pkt.len + 1); // extra byte for string NULL-termination
    assert(ws_pkt.payload);

    // First receive the full message
    ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
    if (ret != ESP_OK) {
    	ESP_LOGE(TAG, "httpd_ws_recv_frame failed: %s", esp_err_to_name(ret));
        free(ws_pkt.payload);
    	return ret;
    }

    ESP_LOGI(TAG, "Websocket input is, type=%d, len=%d:", ws_pkt.type, ws_pkt.len);
    ESP_LOG_BUFFER_HEXDUMP(TAG, ws_pkt.payload, ws_pkt.len, ESP_LOG_INFO);

    if (ws_pkt.type == HTTPD_WS_TYPE_TEXT) {
    	worker_process_websocket_input(ctx, ws_pkt.payload, ws_pkt.len, ws_pkt.payload);
    	return ESP_OK;
    }

    free(ws_pkt.payload);
    return ESP_OK;
}
```

I assume that commit cd521d4ae3301058da106eac1df5e396fec280fe introduced this bug.

## Debug Logs

```
I (578759) Websockets: Websocket input len=1:
I (578759) Websockets: Websocket input is, type=8, len=1:
I (578759) Websockets: 0x3ffd38e4   57                                                |W|
W (578759) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
E (578759) Websockets: httpd_ws_recv_frame failed: ESP_ERR_INVALID_STATE
I (580299) Websockets: Websocket input len=1:
I (580299) Websockets: Websocket input is, type=7, len=1:
I (580299) Websockets: 0x3ffd38e4   d1                                                |.|
I (581829) Websockets: Websocket input len=1:
I (581829) Websockets: Websocket input is, type=8, len=1:
I (581829) Websockets: 0x3ffd38e4   3d                                                |=|
I (581839) Websockets: Websocket input len=30:
I (581839) Websockets: Websocket input is, type=7, len=30:
I (581839) Websockets: 0x3ffd3a64   41 5b 4e 19 13 05 0d 18  06 33 3c 2f 41 4d 4e 1c  |A[N......3</AMN.|
I (581839) Websockets: 0x3ffd3a74   02 13 0d 01 10 43 56 37  18 43 19 1c 07 00        |.....CV7.C....|
I (583359) Websockets: Websocket input len=1:
I (583359) Websockets: Websocket input is, type=8, len=1:
I (583359) Websockets: 0x3ffd38e4   c9                                                |.|
I (583369) Websockets: Websocket input len=10:
I (583369) Websockets: Websocket input is, type=5, len=10:
I (583369) Websockets: 0x3ffd38e4   41 5b 4e 19 13 05 0d 18  06 33                    |A[N......3|
W (583369) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
I (584899) Websockets: Websocket input len=1:
I (584899) Websockets: Websocket input is, type=9, len=1:
I (584899) Websockets: 0x3ffd38e4   25                                                |%|
I (584899) Websockets: Websocket input len=98:
I (584899) Websockets: Websocket input is, type=0, len=98:
I (584899) Websockets: 0x3ffd2af4   5b 4e 19 52 05 0d 18 47  33 3c 2f 00 4d 4e 1c 43  |[N.R...G3</.MN.C|
I (584899) Websockets: 0x3ffd2b04   13 0d 01 51 43 56 37 59  43 19 1c 46 00 18 09 70  |...QCV7YC..F...p|
I (584899) Websockets: 0x3ffd2b14   31 2f 4e 18 15 1e 19 47  4d 4e 0a 4b 13 01 1b 43  |1/N....GMN.K...C|
I (584899) Websockets: 0x3ffd2b24   13 09 39 52 05 0d 18 47  32 18 0d 56 14 1f 4e 18  |..9R...G2..V..N.|
I (584899) Websockets: 0x3ffd2b34   15 1e 19 47 4d 4e 0d 4e  04 1e 18 00 5b 18 1e 57  |...GMN.N....[..W|
I (584899) Websockets: 0x3ffd2b44   04 40 4e 47 17 09 02 56  43 56 18 50 14 09 11 7f  |.@NG...VCV.P....|
I (584899) Websockets: 0x3ffd2b54   4d 4e                                             |MN|
I (584909) Websockets: Websocket input len=104:
W (589909) httpd_txrx: httpd_sock_err: error in recv : 11
W (589909) httpd_ws: httpd_ws_recv_frame: Failed to receive payload
E (589909) Websockets: httpd_ws_recv_frame failed: ESP_FAIL```

Alvin1Zhang, 2021-03-18:
Thanks for the very detailed report and sorry for the inconvenience, we will look into.

yuanjianmin, 2021-03-22:
@zinke-ct-video Please add following code in your `websocket_handler` entrance.
```
if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
```
Please refer the example code https://github.com/espressif/esp-idf/commit/a1d5cfc26023c64e973d5b891677a53302e993a0#diff-80d24d0597f5bd4698dcbd4628c92ef67f1427dbe6e89b83c3bd3a02678d4765

zinke-ct-video, 2021-03-22:
Thank you very much, this solved my problem.

By the way, would be really helpful if documentation for websocket server would be added under [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#websocket-server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#websocket-server). Maybe a short note that behaviour of the websocket handler function has changed.

closed 2021-03-22 by zinke-ct-video