espressif/esp-idf #6733: otatool.py crashes when imported as external python lib

etchove, 2021-03-17:
## Problem Description

When importing `otatool.py` as an python lib, it crashes if path is set in main program.

### Code to reproduce this issue

```python
import os
import sys

PORT="COM8"

idf_path = os.environ["IDF_PATH"]
sys.path.append(os.path.join(idf_path, "components", "esptool_py", "esptool"))
sys.path.append(os.path.join(idf_path, "components", "partition_table"))
sys.path.append(os.path.join(idf_path, "components", "app_update"))

import otatool

esp_ota = otatool.OtatoolTarget("COM8")
esp_ota.switch_ota_partition(0)
```

## Debug Logs

```
Traceback (most recent call last):
  File "autoprog_esp.py", line 53, in <module>
    esp_ota.switch_ota_partition(0)
  File "C:/###/esp-idf\components\app_update\otatool.py", line 106, in switch_ota_partition
    sys.path.append(PARTTOOL_DIR)
NameError: name 'PARTTOOL_DIR' is not defined
```

## Proposal

Remove line 106 in `otatool.py` as it is already in path at the to of the file or from external program.
```python
sys.path.append(PARTTOOL_DIR)
```

Alvin1Zhang, 2021-03-18:
Thanks for reporting and sharing the proposal, we will look into.

Alvin1Zhang, 2021-03-24:
Thanks for reporting, we have a fix now under internal reviewing, and the issue will close automatically once the fix is available on GitHub. Thanks.

closed 2021-04-01 by espressif-bot