espressif/esp-idf #6734: esp-idf/components/esp_wifi/src/phy_init.c:217: undefined reference to `phy_dig_reg_backup'

mvarchdev, 2021-03-17:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-479-g1067b2870
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes (VSCode w/ESP-IDF extension)
- Power Supply:         USB

## Problem Description

Cant compile program from newest version. It compiled in older versions.

## Debug Logs

```
/home/marosvarchola/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(phy_init.c.obj):(.literal.esp_phy_disable+0x10): undefined reference to `phy_dig_reg_backup'
/home/marosvarchola/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(phy_init.c.obj): in function `phy_digital_regs_store':
/home/marosvarchola/esp/esp-idf/components/esp_wifi/src/phy_init.c:210: undefined reference to `phy_dig_reg_backup'
/home/marosvarchola/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(phy_init.c.obj): in function `phy_digital_regs_load':
/home/marosvarchola/esp/esp-idf/components/esp_wifi/src/phy_init.c:217: undefined reference to `phy_dig_reg_backup'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

igrr, 2021-03-17:
Can you try running `git submodule update --init` in ESP-IDF directory?

mvarchdev, 2021-03-17:
Ouch, my bad, thank you very much!

closed 2021-03-17 by mvarchdev