espressif/esp-idf #6735: Error on gpio_example_main.c

jlnbotello, 2021-03-17:
```
static void IRAM_ATTR gpio_isr_handler(void* arg){
    uint32_t gpio_num = (uint32_t) arg;                        // HERE: Casting error. Should be  *((uint32_t*) arg) 
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
}
```
[Link bug line](https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c#L46)

chegewara, 2021-03-17:
Hi @jlnbotello 
actually not. As you can see the arg passed is value, not pointer:
`gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);`

You can see it by adding `ets_printf("GPIO: %d\n", gpio_num);` in ISR function.

https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c#L97

jlnbotello, 2021-03-17:
You're right! Thanks for the quick response. I didn't realise about the input. Even though, I think is not a good practice to misrepresent the responsibility of a pointer (hold addresses). 
I mean whatever type you want to pass to the callback

closed 2021-03-17 by jlnbotello