espressif/esp-idf #6737: NimBLE 128-bits UUIDs are reversed

DurandA, 2021-03-18:
When declaring a 128-bits BLE UUID (i.e. for a service or a characteristic) then the UUID is reversed.

Here is an example:
```
#define GATT_RS_UUID 0x09,0x46,0x02,0x00,0xc9,0xee,0x41,0x95,0x81,0xce,0x3e,0xc8,0x9c,0x3d,0x6e,0x3b

static const struct ble_gatt_svc_def gatt_svr_svcs[] = {
    {
        /* Service: Reflow */
        .type = BLE_GATT_SVC_TYPE_PRIMARY,
        .uuid = BLE_UUID128_DECLARE(GATT_RS_UUID),
        ...
```

With this definition, the UUID will be `3b6e3d9c-c83e-ce81-9541-eec900024609`. Is this the normal behavior of NimBLE or is this an issue with the ESP-IDF port?

dhrishi, 2021-03-18:
@DurandA  Upstream NimBLE behavior. See this example from the upstream for reference: https://github.com/apache/mynewt-nimble/blob/master/apps/bleprph/src/gatt_svr.c#L37-L40

Please close the issue if it addresses your concern.

DurandA, 2021-03-18:
@dhrishi Thanks.

closed 2021-03-18 by DurandA