espressif/esp-idf #6738: ESP32S2 Reboots upon interrupts

Shravani100, 2021-03-18:
Dear all,

I am using ESP32S2 and GPIO interrupts in my application.But whenever Interrupt arrives ESP32S2 reboots. What may be the issue?
Please help.

Alvin1Zhang, 2021-03-18:
@Shravani100 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Shravani100, 2021-03-18:
Development Kit: ESP32S2 saola-1 Develpement Board
- Kit version :v1.2
- Module or chip used:  ESP32-S2-WROOM
- IDF version :4.3
- Build System: idf.py
- Compiler version :  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Ubuntu 20.04 on vbox
- Using an IDE?: No
- Power Supply: USB
Please check attatched log files and the code

[log.txt](https://github.com/espressif/esp-idf/files/6163093/log.txt)
[gpio_interrupts1.zip](https://github.com/espressif/esp-idf/files/6163095/gpio_interrupts1.zip)

felmue, 2021-03-18:
Hello @Shravani100 

I haven't look at everything, but I am pretty sure that calling `ESP_LOGI()` from within the interrupt function causes the reboot. You'll need to set a flag in the interrupt function and then check for that flag in the main loop.

Thanks
Felix

Shravani100, 2021-03-18:
Hi @felmue 
Thank you so much for the reply.Finally It is working now.

Highly appreciated.

felmue, 2021-03-18:
Hello @Shravani100 

You are welcome. I am glad to hear you got it working to your liking.

Cheers
Felix

Alvin1Zhang, 2021-03-19:
@felmue Thanks for sharing the notes. @Shravani100 Thanks for reporting and glad it works, will close now, feel free to reopen.

closed 2021-03-19 by Alvin1Zhang