espressif/esp-idf #6740: UART reading issue without any driver errors

pbruynbroeck, 2021-03-18:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROVER-IE
- IDF version: v4.0.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: [no]
- Power Supply:         external 3.3V

## Problem Description

We have a GNSS receiver that sending frame over UART to the ESP. After several hours of operation, the ESP start to read wrong values.
In our application, we can check if the frame is valid thank to a CRC. Here, I plot the total number of CRC error over the time (in ms):

[image]

As you can see, after ~20h, we start to have a lot of reading error. After the last CRC err, the behavior returned to normal.

We are using  the default `uart_driver` like in the `uart_events_example_main.c`. In the left, I print data received in `uart event queue`, and in the right all data that read on the UART with a raspberry.

[image]

If you want more data:
- [esp.log](https://github.com/espressif/esp-idf/files/6164860/esp.log)
- [gnss.log](https://github.com/espressif/esp-idf/files/6164861/gnss.log)

As I mentioned in the title, in the `uart event queue`, all data have the type `UART_DATA`. We haven't any driver errors.

Alvin1Zhang, 2021-03-19:
@pbruynbroeck Thanks for reporting and sharing the logs, we will look into.

dzarda, 2021-12-11:
I'm also having issues with UART reception yielding data corruption.

People have reported RX corruption in higher data throughput (same as me) but your described behavior is interesting. Could there be some increased system load at the 20 h mark?

Have you gained additional knowledge since?