espressif/esp-idf #6741: Avoid crashing when BMP180(I2C) is not connected to ESP32-WROOM-32D

Yoimer, 2021-03-18:
Hello over there. ESP-IDF crashes when BMP180 is not plugged to the ESP32. The code I am using is this:
https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/bmp180_test/main/main.c
https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/bmp180_test/CMakeLists.txt

The library  code is based in: https://github.com/UncleRus/esp-idf-lib/tree/master/components/bmp180

Is there any way the error checking does not endlessly crashes the esp32 if sensor is not plugged?
This is the error I get. 
`C:\esp-idf-v4.2\pressure>idf.py  monitor
Executing action: monitor
Choosing default port b'COM4' (use '-p PORT' option to set a specific serial port)
Running idf_monitor in directory c:\esp-idf-v4.2\pressure
Executing "C:\Users\Yoimer\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\esp-idf-v4.2\tools/idf_monitor.py -p COM4 -b 115200 --toolchai
n-prefix xtensa-esp32-elf- c:\esp-idf-v4.2\pressure\build\example-bmp180.elf -m 'C:\Users\Yoimer\.espressif\python_env\idf4.2_py3.8_env\Scripts\python
.exe' 'C:\esp-idf-v4.2\tools\idf.py'"...
←[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx←[0m
←[0;33m--- Using \\.\COM4 instead...←[0m
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7016
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (29) boot: compile time 09:39:20
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0711c ( 28956) map
I (123) esp_image: segment 1: paddr=0x00017144 vaddr=0x3ffb0000 size=0x020a8 (  8360) load
I (127) esp_image: segment 2: paddr=0x000191f4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/esp-idf-v4.2/components/freertos/xtensa/xtensa_vectors.S:1730

I (132) esp_image: segment 3: paddr=0x00019600 vaddr=0x40080404 size=0x06a18 ( 27160) load
I (153) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15ca4 ( 89252) map
0x400d0020: _stext at ??:?

I (187) esp_image: segment 5: paddr=0x00035ccc vaddr=0x40086e1c size=0x044c4 ( 17604) load
0x40086e1c: vTaskPrioritySet at C:/esp-idf-v4.2/components/freertos/tasks.c:1703

I (201) boot: Loaded app from partition at offset 0x10000
I (201) boot: Disabling RNG early entropy source...
I (202) cpu_start: Pro cpu up.
I (205) cpu_start: Application information:
I (210) cpu_start: Project name:     example-bmp180
I (216) cpu_start: App version:      v4.2-dirty
I (221) cpu_start: Compile time:     Mar 18 2021 09:37:58
I (227) cpu_start: ELF file SHA256:  10fdf973a6626195...
I (233) cpu_start: ESP-IDF:          v4.2-dirty
I (238) cpu_start: Starting app cpu, entry point is 0x40081668
0x40081668: call_start_cpu1 at C:/esp-idf-v4.2/components/esp32/cpu_start.c:287
[... 181 lines cut ...]
I (233) cpu_start: ESP-IDF:          v4.2-dirty
I (238) cpu_start: Starting app cpu, entry point is 0x40081668
0x40081668: call_start_cpu1 at C:/esp-idf-v4.2/components/esp32/cpu_start.c:287

^C
C:\esp-idf-v4.2\pressure>
Traceback (most recent call last):
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 1087, in <module>
    main()
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 991, in main
    monitor.main_loop()
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 538, in main_loop
    self.handle_serial_input(data)
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 581, in handle_serial_input
    self._print(line + b'\n')
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 830, in _print
    console_printer(string)
  File "C:\Users\Yoimer\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages\serial\tools\miniterm.py", line 65, in write_bytes
    self.byte_output.write(byte_string)
  File "C:\esp-idf-v4.2\tools/idf_monitor.py", line 1051, in write
    b = bytes([b])
KeyboardInterrupt

Aborted!
`

negativekelvin, 2021-03-18:
Use ESP_ERROR_CHECK_WITHOUT_ABORT or your own error handling code

Yoimer, 2021-03-18:
@negativekelvin Thanks a lot for replying. Let me try what you suggested and I will come with the updates in a few hours.

Yoimer, 2021-03-18:
@negativekelvin Thanks a lot for replying. Let me try what you suggested and I will come with the updates in a few hours.

> Use ESP_ERROR_CHECK_WITHOUT_ABORT or your own error handling code

Yoimer, 2021-03-19:
@negativekelvin Thanks for your support. All is working now.

closed 2021-03-19 by Yoimer

Alvin1Zhang, 2021-03-19:
@negativekelvin Thanks for sharing the debugging notes, @Yoimer glad to hear all is working now.