espressif/esp-idf #6743: WiFi task triggering watchdog

KaeLL, 2021-03-18:
## Environment

- Module or chip used:  ESP32
- Build System: Make
- IDF version: Latest release/v4.1 as well as latest release/v4.2 as of right now.
- Power Supply: external 5V

## Problem Description

WiFi intermittently starts to spin on the core it's running on, disconnecting from the AP and triggering the TWDT, exactly like #2370.
I've reproduced this issue with BLE disabled, but having both WiFi and BLE enabled (coexist also enabled) makes it easier. There is no useful log or backtrace because the problem is more easily reproducible on my test rack with 32 boards working simultaneously a few cm away from each other (mentioning it in case it's relevant), and by the time I plug the cable to check the log, it's spitting only the watchdog error. Power supply issues have been discarded as I've tested it and they are working fine. Also tested keeping all WiFi stuff on IRAM but to no avail. Below there are 4 out of about 2 dozen core dumps I selected that have subtle differences on the WiFi task stack, as well as the sdkconfig used for the tests.

Let me know if there is more info I can provide.

## Other items if possible

<details>
    <summary>sdkconfig</summary>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG=y
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=4
[... 1972 lines cut ...]
.coredump.tasks.data 0x3ffef17c 0x170 RW
.coredump.tasks.data 0x3ffeefd0 0x1a4 RW
.coredump.tasks.data 0x3ffb7a50 0x170 RW
.coredump.tasks.data 0x3ffb7800 0x248 RW
.coredump.tasks.data 0x3ffd26a8 0x170 RW
.coredump.tasks.data 0x3ffd24b0 0x1f0 RW
.coredump.tasks.data 0x3ffdf270 0x170 RW
.coredump.tasks.data 0x3ffea560 0x20c RW
.coredump.tasks.data 0x3ffdfda0 0x170 RW
.coredump.tasks.data 0x3ffcd590 0x1e4 RW
.coredump.tasks.data 0x3ffda8c4 0x170 RW
.coredump.tasks.data 0x3ffda6a0 0x21c RW
.coredump.tasks.data 0x3ffb5770 0x170 RW
.coredump.tasks.data 0x3ffb5590 0x1d8 RW
.coredump.tasks.data 0x3ffbaaa0 0x170 RW
.coredump.tasks.data 0x3ffba900 0x198 RW
.coredump.tasks.data 0x3ffb7d7c 0x170 RW
.coredump.tasks.data 0x3ffba080 0x1b8 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

</details>

Alvin1Zhang, 2021-03-19:
Thanks for reporting and sharing the logs, we will look into.

espxiehang, 2021-03-19:
Hi @KaeLL , Can you test this version? I think it can solve this issue.

```
commit b449909b352fb68ff49f79c0edd0b49dd064b2f6

    Fix controller task watchdog in Wi-Fi test
```

espxiehang, 2021-03-19:
Hi @KaeLL ， It is suggessed that you use the latest release / v4.3 verification.
If this issue still exists, please provide test log and recurrence method to me.
thank you!

KaeLL, 2021-03-19:
@espxiehang how about backporting the fix to release/v4.{1,2}, especially release/4.1? When performing the tests with release/v4.2 I had another issue in which the firmware would freeze and completely stop logging. That left me wishing I didn't have to change IDF versions so soon, and I'm on a tight schedule. These backports will happen anyway, right?

AxelLin, 2021-03-19:
The commit b449909b352fb68ff49f79c0edd0b49dd064b2f6 is a fix in BT library,
but @KaeLL reported "I've reproduced this issue with BLE (and coex) disabled".
I doubt if it can completely fix the issue.

KaeLL, 2021-03-20:
Nice catch @AxelLin .I actually checked the commit a couple hours ago and saw that it's a BT lib fix, but was distracted enough to not connect the dots.  
Here are a few dumps of the version running only WiFi, with BLE and coex __disabled__. All seem to be triggered by the watchdog except for Dump 5, which seems to be a Divide by 0.

<details>
    <summary>Dump 1</summary>

```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffb4c6c, name: 'IDLE0', GDB name: 'process 1073433708'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x40009203
pc             0x40087cb7	0x40087cb7 <invoke_abort+27>
lbeg           0x0	0
lend           0x0	0
lcount         0x0	0
sar            0x5	5
ps             0x60021	393249
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8008851c	-2146925284
a1             0x3ffbf810	1073477648
a2             0x6	6
a3             0x2f	47
a4             0xc	12
a5             0xffffffff	-1
a6             0x0	0
a7             0xfffffffe	-2
a8             0x0	0
a9             0x3ffbf7f0	1073477616
a10            0x0	0
a11            0x3ffbf7a1	1073477537
a12            0x3ffbf74f	1073477455
a13            0x30	48
a14            0x0	0
a15            0x3ffbf759	1073477465

==================== CURRENT THREAD STACK =====================
#0  0x40087cb7 in invoke_abort () at /opt/esp/idf/components/esp32/panic.c:159
#1  0x4008851c in abort () at /opt/esp/idf/components/esp32/panic.c:174
[... 1173 lines cut ...]
.coredump.tasks.data 0x3ffd1c94 0x164 RW 
.coredump.tasks.data 0x3ffd18c0 0x3cc RW 
.coredump.tasks.data 0x3ffd91cc 0x164 RW 
.coredump.tasks.data 0x3ffdf870 0x174 RW 
.coredump.tasks.data 0x3ffd8c18 0x164 RW 
.coredump.tasks.data 0x3ffdd2d0 0x3e4 RW 
.coredump.tasks.data 0x3ffb5e38 0x164 RW 
.coredump.tasks.data 0x3ffb5cb0 0x180 RW 
.coredump.tasks.data 0x3ffdc470 0x164 RW 
.coredump.tasks.data 0x3ffdc260 0x208 RW 
.coredump.tasks.data 0x3ffb1368 0x164 RW 
.coredump.tasks.data 0x3ffb11d0 0x190 RW 
.coredump.tasks.data 0x3ffb0900 0x164 RW 
.coredump.tasks.data 0x3ffb0740 0x1b8 RW 
.coredump.tasks.data 0x3ffafdc0 0x164 RW 
.coredump.tasks.data 0x3ffafbe0 0x1d8 RW 
.coredump.tasks.data 0x3ffbb1f4 0x164 RW 
.coredump.tasks.data 0x3ffbaff0 0x1fc RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

</details>

fito-jaeuklee, 2021-03-22:
I have same issue ...
In my case IDF version is v4.2  and use ble & wifi coex function. (wifi - transmit GPS data / BLE - transmit Heart rate data)

Watchdog reboot happen rarely, when i use only wifi watchdog.
but use both of them, watchdog is constantly occur.

espxiehang, 2021-03-22:
Hi @KaeLL @AxelLin , We will synchronize that fix to release/v4.1.{1,2} as soon as possible.
It solves the problem of watchdog when coexisting.
In WiFi only, if watchdog appears, can you provide me with a recurrence method？

KaeLL, 2021-03-22:
I'll see what I can do, but it might take a couple of days on my part.

KaeLL, 2021-03-22:
Not sure if this is totally related, though I hope it is, but testing with everything enabled (BLE and MQTT over WiFi) against the `release/v4.3` (9a2d251) yielded this
```E (223684075) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (223684075) task_wdt:  - IDLE (CPU 0)
E (223684075) task_wdt: Tasks currently running:
E (223684075) task_wdt: CPU 0: tiT
E (223684075) task_wdt: CPU 1: IDLE
E (223684075) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40173F76:0x3FFBF900 0x40084F0A:0x3FFBF920 0x0004001E:0x00000000  |<-CORRUPTED
0x40173f76: task_wdt_isr at /esp-idf/components/esp_common/src/task_wdt.c:189

0x40084f0a: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

E (223684075) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008EE11:0x3FFC0130 0x40084F0A:0x3FFC0150 0x0004001E:0x00000000  |<-CORRUPTED
0x4008ee11: esp_crosscore_isr at /esp-idf/components/esp32/crosscore_int.c:77

0x40084f0a: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
```
Other boards had the same log, except sometimes for the CPU which the LwIP task is spinning on.

KaeLL, 2021-03-25:
@espxiehang ETA on that backport, specially for release/v4.1?

espxiehang, 2021-03-26:
Hi @KaeLL ，We have merge this change to release / v4.1.
It should be able to synchronize to GitHub soon.

KaeLL, 2021-03-26:
@espxiehang I'm talking about espressif/esp32-wifi-lib@e7fc97c
The issue I'm having isn't necessarily related to bluetooth, though it worsens it.

KaeLL, 2021-03-27:
@espxiehang I managed to test a few boards with the code compiled against rev. 1067b2870, and I'm still having the exact same issue as initially, along with the LwIP task also going through the same issue

```
E (5664408) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5664408) task_wdt:  - IDLE (CPU 0)
E (5664408) task_wdt: Tasks currently running:
E (5664408) task_wdt: CPU 0: tiT
E (5664408) task_wdt: CPU 1: IDLE
E (5664408) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4017607E:0x3FFBF860 0x40085172:0x3FFBF880 0x0004001E:0x00000000  |<-CORRUPTED

E (5664408) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008ECB5:0x3FFC0090 0x40085172:0x3FFC00B0 0x0004001E:0x00000000  |<-CORRUPTED
```
Not sure if it's relevant, but the WiFi functions are not in IRAM (which still isn't an excuse for the behavior).

espxiehang, 2021-03-29:
Hi @KaeLL  ，Can you give me its call stack information. Refer to the following command:
`xtensa-esp32-elf-addr2line -piaf -e *.elf 0x4017607E:0x3FFBF860 0x40085172:0x3FFBF880 0x0004001E:0x00000000`

In addition, please update WiFi lib espressif/esp32-wifi-lib@e7fc97c7

KaeLL, 2021-03-29:
@espxiehang

<details>
<summary>Here's the backtrace of another build with debug stuff on</summary>

```
E (496190) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (496190) task_wdt:  - IDLE (CPU 1)
E (496190) task_wdt: Tasks currently running:
E (496190) task_wdt: CPU 0: IDLE
E (496190) task_wdt: CPU 1: wifi
E (496190) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4017C32B:0x3FFBF860 0x40085729:0x3FFBF880 0x401D3B4F:0x3FFBBB70 0x4017B94F:0x3FFBBB90 0x40093D19:0x3FFBBBB0 
0x4017c32b: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:191

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x401d3b4f: cpu_ll_waiti at /esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /esp-idf/components/esp_pm/pm_impl.c:802

0x4017b94f: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:63

0x40093d19: prvIdleTask at /esp-idf/components/freertos/tasks.c:3819 (discriminator 1)

E (496190) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F859:0x3FFC0090 0x40085729:0x3FFC00B0 0x400DC00A:0x3FFD3730 0x400DC3B6:0x3FFD3750 0x400D9509:0x3FFD3770 0x401D73CA:0x3FFD3790 
0x4008f859: esp_crosscore_isr at /esp-idf/components/esp32/crosscore_int.c:77

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400dc00a: wifi_hw_stop at ??:?

0x400dc3b6: wifi_stop_process at ??:?

0x400d9509: ieee80211_ioctl_process at ??:?

0x401d73ca: ppTask at ??:?
```
</details>
<details>
<summary>And here's the sdkconfig used for this build</summary>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
[... 1196 lines cut ...]
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
# end of Wear Levelling

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16
CONFIG_WIFI_PROV_AUTOSTOP_TIMEOUT=30
# end of Wi-Fi Provisioning Manager

#
# Supplicant
#
CONFIG_WPA_MBEDTLS_CRYPTO=y
# CONFIG_WPA_WAPI_PSK is not set
# CONFIG_WPA_DEBUG_PRINT is not set
# CONFIG_WPA_TESTING_OPTIONS is not set
# CONFIG_WPA_WPS_WARS is not set
# CONFIG_WPA_11KV_SUPPORT is not set
# end of Supplicant
# end of Component config
```

</details>

KaeLL, 2021-03-29:
@espxiehang as for updating the wifi lib, I was already using IDF on commit 1067b2870 since before my penultimate comment, which already includes espressif/esp32-wifi-lib@e7fc97c and other 3 commits after it.
I said
> I managed to test a few boards with the code compiled against rev. 1067b28, and I'm still having the exact same issue as initially

AxelLin, 2021-03-30:
> In addition, please update WiFi lib [espressif/esp32-wifi-lib@e7fc97c](https://github.com/espressif/esp32-wifi-lib/commit/e7fc97c7)

Hi @espxiehang 

According to your comment below:
https://github.com/espressif/esp-idf/issues/6744#issuecomment-809213061

The watchdog issue in master tree was caused by https://github.com/espressif/esp-idf/commit/deb94f4
but this bad commit does not exist in v4.1 and v4.2 branches at all.
This makes me wondering if this is a different issue. (since this issue was initially reported against v4.1 and v4.2)

espxiehang, 2021-03-30:
Hi @KaeLL ，From the log analysis you provided, WiFi lib espressif/esp32-wifi-lib@e7fc97c can solve this issue.
I guess there's `E (4092778) wifi:ieee80211_ioctl.c 1443` printing before I trigger watchdog.

Hi @AxelLin ,  The issue of  `E (4092778) wifi:ieee80211_ioctl.c 1443` only exists in master and release/v4.3. Fix commit espressif/esp32-wifi-lib@e7fc97c
Bluetooth watchdog issue exists in all versions. Fix commit espressif/esp32-bt-lib@63e1129

As for the v4.1 and v4.2 watchdog issue, can you provide me with more information.
 commit id、backtrace、log , etc

KaeLL, 2021-03-30:
@espxiehang Along with boards that restarted (apparently) because of the watchdog, I had 3 instances of another issue which I'm contemplating the chances of being related to this one in any way.

<details>
    <summary>Backtrace 1</summary>

```
E (56296789) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (56296789) task_wdt:  - IDLE (CPU 0)
E (56296789) task_wdt: Tasks currently running:
E (56296789) task_wdt: CPU 0: esp_timer
E (56296789) task_wdt: CPU 1: IDLE
E (56296789) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4017C2E7:0x3FFBF8F0 0x40085729:0x3FFBF910 0x4000BFED:0x3FFBAEC0 0x4009657E:0x3FFBAED0 0x400936D4:0x3FFBAEF0 0x400D1846:0x3FFBAF30 0x40082722:0x3FFBAF50 0x400829B7:0x3FFBAF90 0x400D1261:0x3FFBAFC0 0x400D12A6:0x3FFBAFF0
0x4017c2e7: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:191

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009657e: vPortExitCritical at /esp-idf/components/freertos/port/xtensa/port.c:472

0x400936d4: xQueueSemaphoreTake at /esp-idf/components/freertos/queue.c:1509

0x400d1846: semphr_take_wrapper at /esp-idf/components/esp_wifi/esp32/esp_adapter.c:304

0x40082722: coex_bb_reset_unlock at ??:?

0x400829b7: coex_timer_ts_end_handler at ??:?

0x400d1261: timer_process_alarm at /esp-idf/components/esp_timer/src/esp_timer.c:366

0x400d12a6: timer_task at /esp-idf/components/esp_timer/src/esp_timer.c:392 (discriminator 1)

E (56296789) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F859:0x3FFC0120 0x40085729:0x3FFC0140 0x4000BFED:0x3FFBC270 0x4009657E:0x3FFBC280 0x4017C58B:0x3FFBC2A0 0x4017C597:0x3FFBC2D0 0x4017B8FD:0x3FFBC2F0 0x40093D19:0x3FFBC310
0x4008f859: esp_crosscore_isr at /esp-idf/components/esp32/crosscore_int.c:77

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009657e: vPortExitCritical at /esp-idf/components/freertos/port/xtensa/port.c:472

0x4017c58b: esp_task_wdt_reset at /esp-idf/components/esp_system/task_wdt.c:338

0x4017c597: idle_hook_cb at /esp-idf/components/esp_system/task_wdt.c:88

0x4017b8fd: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x40093d19: prvIdleTask at /esp-idf/components/freertos/tasks.c:3819 (discriminator 1)
```

</details>
<details>
    <summary>Backtrace 2</summary>

```
E (51183188) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (51183188) task_wdt:  - IDLE (CPU 0)
E (51183188) task_wdt: Tasks currently running:
E (51183188) task_wdt: CPU 0: esp_timer
E (51183188) task_wdt: CPU 1: IDLE
E (51183188) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4017C2E7:0x3FFBF8F0 0x40085729:0x3FFBF910 0x4009023A:0x3FFBAF30 0x400826F5:0x3FFBAF50 0x400829B7:0x3FFBAF90 0x400D1261:0x3FFBAFC0 0x400D12A6:0x3FFBAFF0
0x4017c2e7: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:191

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009023a: coex_is_in_isr_wrapper at /esp-idf/components/esp_wifi/esp32/esp_adapter.c:661

0x400826f5: coex_bb_reset_unlock at ??:?
[... 48 lines cut ...]
0x400d1261: timer_process_alarm at /esp-idf/components/esp_timer/src/esp_timer.c:366

0x400d12a6: timer_task at /esp-idf/components/esp_timer/src/esp_timer.c:392 (discriminator 1)

E (58172149) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F859:0x3FFC0120 0x40085729:0x3FFC0140 0x4000BFED:0x3FFBC270 0x4009657E:0x3FFBC280 0x4017C58B:0x3FFBC2A0 0x4017C597:0x3FFBC2D0 0x4017B8FD:0x3FFBC2F0 0x40093D19:0x3FFBC310
0x4008f859: esp_crosscore_isr at /esp-idf/components/esp32/crosscore_int.c:77

0x40085729: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009657e: vPortExitCritical at /esp-idf/components/freertos/port/xtensa/port.c:472

0x4017c58b: esp_task_wdt_reset at /esp-idf/components/esp_system/task_wdt.c:338

0x4017c597: idle_hook_cb at /esp-idf/components/esp_system/task_wdt.c:88

0x4017b8fd: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x40093d19: prvIdleTask at /esp-idf/components/freertos/tasks.c:3819 (discriminator 1)
```

</details>

The reason I thought there must be some correlation is because my app doesn't make use of the `esp_timer`, but the WiFi stack does. Any thoughts?

espxiehang, 2021-03-31:
Hi @KaeLL ，Thank you for your feedback. Can you test it with 'release/v4.1'  e71033b  version.

KaeLL, 2021-03-31:
@espxiehang why? I was still having the issue in previous revisions of the wifi lib on master, which has a more up-to-date version of the wifi lib than release/v4.1. espressif/esp32-wifi-lib@e7fc97c seems to have fixed the issue apart from the `esp_event` issue I'm still having. So from where I see, the issue still exists on release/v4.1 and the fix should be backported.

espxiehang, 2021-04-01:
Hi @KaeLL ，Because e71033b version contains fix for Bluetooth watchdog issue .
The  esp_event issue has not been backported to release/v4.1 version.

KaeLL, 2021-04-01:
@espxiehang Very well. I'll test it and see what happens despite having stated that BLE doesn't even need to be enabled for the issue to happen.
> The esp_event issue has not been backported to release/v4.1 version.

What?? Did you mean `esp_timer`? And did you also mean **the fix** for the issue wasn't backported? And I **didn't** have the esp_timer issue on release/v4.{1,2}, only on master + espressif/esp32-wifi-lib@e7fc97c applied. Which means either you know and I don't that these issues are affecting the 2 release branches, or I have no idea what you're talking about in your last statement.

espxiehang, 2021-04-01:
@KaeLL I'm sorry I didn't describe it right.
Because there is the following print in your backtrace, I suspect it is caused by Bluetooth watchdog issue.
So I hope you can test the new version and look forward to your feedback.
```
0x4009023a: coex_is_in_isr_wrapper at /esp-idf/components/esp_wifi/esp32/esp_adapter.c:661
0x400826f5: coex_bb_reset_unlock at ??:?
0x400829b7: coex_timer_ts_end_handler at ??:?
```

KaeLL, 2021-04-01:
@espxiehang but those are from the app compiled with the master branch of IDF, not release/v4.1. I'll do the tests anyway just out of curiosity.

espxiehang, 2021-04-06:
Hi @KaeLL ,How about the test results with this version? Does watchdog issue still exist?

KaeLL, 2021-04-06:
@espxiehang Sorry for taking so long to get back to you. Yes, the issue seems to be resolved in the current version of release/v4.1. Thank you, and feel free to close the issue.

KaeLL, 2021-04-12:
@espxiehang so it looks like I spoke too soon and the problem is still happening on latest release/v4.1 (a921852) with the exact same symptoms.
@Alvin1Zhang can you reopen the issue, please?

Here are 3 core dumps which look similar enough to others I posted before. There are no logs because there was no backtraces being printed, likely due to the fact that the app was compiled with `/O2`, so there was only the usual watchdog error telling the wifi task had not reset the watchdog in time.

<details>
<summary>Dump 1</summary>

```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbd118, name: 'IOsTask', GDB name: 'process 1073467672'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x40009203
pc             0x4008f61b	0x4008f61b <invoke_abort+27>
lbeg           0x0	0
lend           0x0	0
lcount         0x0	0
sar            0x5	5
ps             0x60b20	396064
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8008feb0	-2146894160
a1             0x3ffce820	1073539104
a2             0x80040004	-2147221500
a3             0x2f	47
a4             0xc	12
a5             0xffffffff	-1
a6             0x0	0
a7             0xfffffffe	-2
a8             0x0	0
a9             0x3ffce800	1073539072
a10            0x0	0
a11            0x3ffce7b1	1073538993
a12            0x3ffce75f	1073538911
a13            0x30	48
a14            0x0	0
a15            0x3ffce769	1073538921

==================== CURRENT THREAD STACK =====================
[... 682 lines cut ...]
.coredump.tasks.data 0x3ffbcb84 0x164 RW
.coredump.tasks.data 0x3ffbc9f0 0x18c RW
.coredump.tasks.data 0x3ffbaa44 0x164 RW
.coredump.tasks.data 0x3ffba8a0 0x19c RW
.coredump.tasks.data 0x3ffb7d48 0x164 RW
.coredump.tasks.data 0x3ffba080 0x1b8 RW
.coredump.tasks.data 0x3ffcd6c8 0x164 RW
.coredump.tasks.data 0x3ffee250 0x1a8 RW
.coredump.tasks.data 0x3ffb7a48 0x164 RW
.coredump.tasks.data 0x3ffb7800 0x240 RW
.coredump.tasks.data 0x3ffdd9d4 0x164 RW
.coredump.tasks.data 0x3ffdd7e0 0x1ec RW
.coredump.tasks.data 0x3ffd1968 0x164 RW
.coredump.tasks.data 0x3ffd1770 0x1f0 RW
.coredump.tasks.data 0x3ffb5758 0x164 RW
.coredump.tasks.data 0x3ffb55a0 0x1b0 RW
.coredump.tasks.data 0x3ffe19b4 0x164 RW
.coredump.tasks.data 0x3ffdf130 0x1f8 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

</details>

Alvin1Zhang, 2021-04-13:
@KaeLL Thanks for sharing the updates, reopened now.

KaeLL, 2021-04-13:
@Alvin1Zhang thanks.
It does seem that building the app with `-O2` might be the cause. I have a 40 boards test running for a little less than 6hrs now, and since the problem generally takes way less to happen, the correlation is starting to look like causation.

KaeLL, 2021-04-13:
Alright, so `-O2` really does seem to trigger the issue while `-Os` seems to work fine. Now, the ball is in your court, espressif.

AxelLin, 2021-04-14:
@KaeLL 
Which compiler version are you using?

KaeLL, 2021-04-14:
@AxelLin The one you get by running the latest release/v4.1's install.sh. I'm on my phone and won't be in front of a computer for the next 9 hours, so I can't tell you specifically.

espxiehang, 2021-04-14:
Hi @KaeLL ,Can you provide me with a reproduction method?

KaeLL, 2021-04-14:
@AxelLin esp-2020r3-8.4.0

@espxiehang For the purposes of this test, the mixing the ble_ibeacon with mqtt/ssl example with the following sdkconfig should do. I can't attest if this setup works just yet, as my test rack is busy. I will do that as soon as I can. I should note that my test setup is 40 esp32's clumped up together ~11cm apart, all using WiFi+BLE scan, just in case that plays a role in forcing the issue to happen.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6312388/sdkconfig.zip)

espxiehang, 2021-04-15:
OK, I'll try to reproduce it.

Amey112, 2021-09-28:
Hey, I am having the same problem, though I am using platformIO and Arduino framework, do you know of a way to resolve it?

My packages are:
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)

AxelLin, 2021-10-30:
> OK, I'll try to reproduce it.

@espxiehang 
Any update?

TianaESP, 2021-11-16:
@KaeLL Hi, can you please try with the latest version v4.3? thanks.

KaeLL, 2021-11-16:
Not really. I haven't had that problem for a while, so I moved past it and considered it fixed for all intents and purposes. I'm won't be able to perform a real test for a few days, maybe a week or 2, because I'm in the process of moving to a new place. I'll eventually finish and do the test, but I can't give you a specific time of when that'll happen.

KaeLL, 2022-01-06:
@Alvin1Zhang I ran some tests against branch release/v4.1 (the branch the stable version of the app is currently being compiled with) and wasn't able to reproduce it all.
@TianaESP As there are a few changes that have to be made to compile the app with 4.3, and bumping IDF version isn't a priority for me right now, I'll have to postpone that test, as I'm currently working on another project as well.
Feel free to close the issue, and I'll ask to reopen if I come across this issue again, with more info.

Alvin1Zhang, 2022-03-08:
Thanks for reporting and sharing the updates, feel free to reopen.

closed 2022-03-08 by Alvin1Zhang

bill1davis, 2022-03-21:
@Alvin1Zhang We saw this in 4.3.1 during a overnight run last night.  Wifi was connected in client mode, ble was continuously scanning.

ESP-IDF 4.3.1
Chip (Rover Module) ESP32-D0WD v3
xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
Optimization -Os
Power External 5v

Backtrace:0x401D8726:0x3FFBEA80 0x40084901:0x3FFBEAA0 0x4021ACBA:0x3FFDDF00 0x40214127:0x3FFDDF20 0x40203A19:0x3FFDDF40 0x40204169:0x3FFDDF60 0x40202DD8:0x3FFDDF80 0x40249351:0x3FFDDFA0

E (236090541) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008E615:0x3FFBF080 0x40084901:0x3FFBF0A0 0x400026F3:0x3FFD89F0 0x400FD353:0x3FFD8A00 0x40102B1D:0x3FFD8A30 0x401032FE:0x3FFD8A70 0x400F4B69:0x3FFD8AA0 0x400F3B6A:0x3FFD8AD0 0x400F4A5B:0x3FFD8D30

E (236095541) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (236095541) task_wdt: - IDLE (CPU 0)

E (236095541) task_wdt: Tasks currently running:
E (236095541) task_wdt: CPU 0: wifi
E (236095541) task_wdt: CPU 1: IDLE
E (236095541) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401D8726:0x3FFBEA80 0x40084901:0x3FFBEAA0 0x4021ACBA:0x3FFDDF00 0x40214127:0x3FFDDF20 0x40203A19:0x3FFDDF40 0x40204169:0x3FFDDF60 0x40202DD8:0x3FFDDF80 0x40249351:0x3FFDDFA0

E (236095541) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008E615:0x3FFBF080 0x40084901:0x3FFBF0A0 0x4000BFED:0x3FFCA880 0x4009B695:0x3FFCA890 0x401D89E6:0x3FFCA8B0 0x401D89F7:0x3FFCA8E0 0x400DC566:0x3FFCA900 0x40099468:0x3FFCA920

E (236100541) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (236100541) task_wdt: - IDLE (CPU 0)
E (236100541) task_wdt: Tasks currently running:
E (236100541) task_wdt: CPU 0: wifi
E (236100541) task_wdt: CPU 1: PDA
E (236100541) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401D8726:0x3FFBEA80 0x40084901:0x3FFBEAA0 0x4021ACBA:0x3FFDDF00 0x40214127:0x3FFDDF20 0x40203A19:0x3FFDDF40 0x40204169:0x3FFDDF60 0x40202DD8:0x3FFDDF80 0x40249351:0x3FFDDFA0

E (236100541) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008E615:0x3FFBF080 0x40084901:0x3FFBF0A0 0x4000BFED:0x3FFCA880 0x4009B695:0x3FFCA890 0x401D89E6:0x3FFCA8B0 0x401D89F7:0x3FFCA8E0 0x400DC566:0x3FFCA900 0x40099468:0x3FFCA920

MaxwellAlan, 2022-03-31:
hi @bill1davis 
Can you use below cmd to parse the backtrace:
xtensa-esp32-elf-add2line -pfiaC -e build/*.elf 0x401D8726:0x3FFBEA80 0x40084901:0x3FFBEAA0 0x4021ACBA:0x3FFDDF00 0x40214127:0x3FFDDF20 0x40203A19:0x3FFDDF40 0x40204169:0x3FFDDF60 0x40202DD8:0x3FFDDF80 0x40249351:0x3FFDDFA0
xtensa-esp32-elf-add2line -pfiaC -e build/*.elf 0x4008E615:0x3FFBF080 0x40084901:0x3FFBF0A0 0x400026F3:0x3FFD89F0 0x400FD353:0x3FFD8A00 0x40102B1D:0x3FFD8A30 0x401032FE:0x3FFD8A70 0x400F4B69:0x3FFD8AA0 0x400F3B6A:0x3FFD8AD0 0x400F4A5B:0x3FFD8D30
Thanks.

bill1davis, 2022-03-31:
Thanks for replying @MaxwellAlan.  We aren't 100%  certain of the build hash where the error occurred.  I checked out a few hashes around the day it occurred.  This is the trace that makes the most sense, but I don't guarantee it is right.  We will pay attention to the build if this occurs again and post a new trace.
```

xtensa-esp32-elf-addr2line -pfiaC -e build/caretaker5.elf 0x401D8726:0x3FFBEA80 0x40084901:0x3FFBEAA0 0x4021ACBA:0x3FFDDF00 0x40214127:0x3FFDDF20 0x40203A19:0x3FFDDF40 0x40204169:0x3FFDDF60 0x40202DD8:0x3FFDDF80 0x40249351:0x3FFDDFA0

0x401d8726: task_wdt_isr at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:179 (discriminator 1)
0x40084901: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4021acba: hal_mac_rx_set_policy at ??:?
0x40214127: ic_set_trc at ??:?
0x40203a19: wifi_reset_mac at ??:?
0x40204169: wifi_init_process at ??:?
0x40202dd8: ieee80211_parse_action at ??:?
0x40249351: ppTask at ??:?
-------------------
xtensa-esp32-elf-addr2line -pfiaC -e build/caretaker5.elf 0x4008E615:0x3FFBF080 0x40084901:0x3FFBF0A0 0x400026F3:0x3FFD89F0 0x400FD353:0x3FFD8A00 0x40102B1D:0x3FFD8A30 0x401032FE:0x3FFD8A70 0x400F4B69:0x3FFD8AA0 0x400F3B6A:0x3FFD8AD0 0x400F4A5B:0x3FFD8D30

0x4008e615: esp_crosscore_isr at /home/dev/esp/esp-idf/components/esp32/crosscore_int.c:77
0x40084901: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x400026f3: ?? ??:0
[filenames and addresses in our application stack I would rather not put in the cloud, you can approach me privately if these matter.  The code in question does a good deal of floating point calculations, so core 1 in the second wdt backtrace was busy]
```

MaxwellAlan, 2022-04-07:
Hi @bill1davis 
 Seems these backtrace not helpful, waitting for new reproduction backtrace.

bill1davis, 2022-04-07:
@MaxwellAlan 

Ok.  Sorry, and thanks for looking at it.  Give me a few days to reproduce it with a known build.

bill1davis, 2022-04-07:
@MaxwellAlan Happened today after about an hour.  Reiterating setup.  We are connected to our device which is a wifi client.  There are open sockets.  We are also continuously BLE scanning for a third party pulse oxymeter.  When we get a inquiry complete, we rescan.

xtensa-esp32-elf-addr2line -pfiaC -e build/caretaker5.elf 0x401D9EDE:0x3FFBEA80 0x400848D9:0x3FFBEAA0 0x4021B7DA:0x3FFDE260 0x40214C67:0x3FFDE280 0x40204505:0x3FFDE2A0 0x40204C59:0x3FFDE2C0 0x402038B8:0x3FFDE2E0 0x40249E7D:0x3FFDE300
0x401d9ede: task_wdt_isr at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:189
0x400848d9: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4021b7da: hal_init at ??:?
0x40214c67: ic_enable at ??:?
0x40204505: wifi_hw_start at ??:?
0x40204c59: wifi_start_process at ??:?
0x402038b8: ieee80211_ioctl_process at ??:?
0x40249e7d: ppTask at ??:?

xtensa-esp32-elf-addr2line -pfiaC -e build/caretaker5.elf 0x4008E5ED:0x3FFBF080 0x400848D9:0x3FFBF0A0 0x4000BFED:0x3FFCACD0 0x4009B66D:0x3FFCACE0 0x401DA19E:0x3FFCAD00 0x401DA1AF:0x3FFCAD30 0x400DC53E:0x3FFCAD50 0x40099440:0x3FFCAD70 
0x4008e5ed: esp_crosscore_isr at /home/dev/esp/esp-idf/components/esp32/crosscore_int.c:77
0x400848d9: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4000bfed: ?? ??:0
0x4009b66d: vPortExitCritical at /home/dev/esp/esp-idf/components/freertos/port/xtensa/port.c:473
0x401da19e: esp_task_wdt_reset at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:336
0x401da1af: idle_hook_cb at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:88
0x400dc53e: esp_vApplicationIdleHook at /home/dev/esp/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)
0x40099440: prvIdleTask at /home/dev/esp/esp-idf/components/freertos/tasks.c:3839

**LOG Before, during, after first  occurrence.**
I (4280631) ble_client: Ignore Spo2 [ECGRec_202130/C867399	ECGRec_202130/C867399] wrong mac [cc:ce:c0:7c:f4:ea]
I (4288761) ble_client: Inq Cmpl while scanning for SpO2. Reset recovery timer. Rescan.
I (4288761) ble_client: Setting Scan Params
I (4288771) ble_client: Will Scan For Orphaned SpO2 Forever
I (4288781) ble_client: scan start success
bcn_timout,ap_probe_send_startI (4291111) ct5_wifi: Unhandled wifi event 21
pm stop, total sleep time: 2557604488 us / 4282819426 us
W (4293601) ct5_wifi: WIFI_EVENT_STA_DISCONNECTED expected = 0 reason = 200
E (4293611) ct5_wifi: Restarting Wifi
hmac tx: ifx0 stop, discardhmac tx: ifx0 stop, discardhmac tx: ifx0 stop, discardhmac tx: ifx0 stop, discardTX Q not empty: 500, TXQ_BLOCK=17fflmac stop hw txqW (4303571) ui: windowManagerLoop: WindowManager queue timed out
E (4303611) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (4303611) task_wdt:  - IDLE (CPU 0)
E (4303611) task_wdt: Tasks currently running:
E (4303611) task_wdt: CPU 0: wifi
E (4303611) task_wdt: CPU 1: IDLE
E (4303611) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401D9EDE:0x3FFBEA80 0x400848D9:0x3FFBEAA0 0x4021B7DA:0x3FFDE260 0x40214C67:0x3FFDE280 0x40204505:0x3FFDE2A0 0x40204C59:0x3FFDE2C0 0x402038B8:0x3FFDE2E0 0x40249E7D:0x3FFDE300 
0x401d9ede: task_wdt_isr at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:189

0x400848d9: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4021b7da: hal_init at ??:?

0x40214c67: ic_enable at ??:?

0x40204505: wifi_hw_start at ??:?

0x40204c59: wifi_start_process at ??:?

0x402038b8: ieee80211_ioctl_process at ??:?

0x40249e7d: ppTask at ??:?

E (4303611) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008E5ED:0x3FFBF080 0x400848D9:0x3FFBF0A0 0x4000BFED:0x3FFCACD0 0x4009B66D:0x3FFCACE0 0x401DA19E:0x3FFCAD00 0x401DA1AF:0x3FFCAD30 0x400DC53E:0x3FFCAD50 0x40099440:0x3FFCAD70 
0x4008e5ed: esp_crosscore_isr at /home/dev/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400848d9: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009b66d: vPortExitCritical at /home/dev/esp/esp-idf/components/freertos/port/xtensa/port.c:473

0x401da19e: esp_task_wdt_reset at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:336

0x401da1af: idle_hook_cb at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:88

[... 30 lines cut ...]

Backtrace:0x4008E5ED:0x3FFBF080 0x400848D9:0x3FFBF0A0 0x4000BFED:0x3FFCACD0 0x4009B66D:0x3FFCACE0 0x401DA19E:0x3FFCAD00 0x401DA1AF:0x3FFCAD30 0x400DC53E:0x3FFCAD50 0x40099440:0x3FFCAD70 
0x4008e5ed: esp_crosscore_isr at /home/dev/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400848d9: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4009b66d: vPortExitCritical at /home/dev/esp/esp-idf/components/freertos/port/xtensa/port.c:473

0x401da19e: esp_task_wdt_reset at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:336

0x401da1af: idle_hook_cb at /home/dev/esp/esp-idf/components/esp_common/src/task_wdt.c:88

0x400dc53e: esp_vApplicationIdleHook at /home/dev/esp/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)

0x40099440: prvIdleTask at /home/dev/esp/esp-idf/components/freertos/tasks.c:3839

E (4312721) logger: Could not queue other vitals. Measurements lost!!!
E (4313211) logger: Could not queue bbb vitals. Measurements lost!!!
W (4313571) ui: windowManagerLoop: WindowManager queue timed out
E (4313611) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (4313611) task_wdt:  - IDLE (CPU 0)
E (4313611) task_wdt: Tasks currently running:
E (4313611) task_wdt: CPU 0: wifi
E (4313611) task_wdt: CPU 1: IDLE
E (4313611) task_wdt: Print CPU 0 (current core) backtrace

MaxwellAlan, 2022-04-08:
Hi @bill1davis 
From your backtrace, this issue is caused by coex, can you try this commit: 72e4b5ac5b00b35980dfd225ceea2ad7c1cb7812 in release/v4.3

bill1davis, 2022-04-11:
@MaxwellAlan 

Thank you!  40 hour run with no watchdog.  That seems to be it.