espressif/esp-idf #6744: task_wdt: Task watchdog got triggered

AxelLin, 2021-03-18:
## Environment

Module or chip used: ESP32-WROOM-32E
IDF version: v4.4-dev-479-g1067b28707e5
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux
Power Supply: USB

## Problem Description

After updating ESP-IDF, my application start hitting below issue on some devices.

Backtrace:0x400FD7CE:0x3FFB0C30 0x400833CD:0x3FFB0C50 0x400902E2:0x3FFC7540 0x40093DC6:0x3FFC7560 0x400942AC:0x3FFC7590 0x401A5D4E:0x3FFC75B0 0x401A5DAE:0x3FFC75D0 0x40091F8A:0x3FFC75F0 
0x400fd7ce: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:191

0x400833cd: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400902e2: wifi_rf_phy_enable at ??:?

0x40093dc6: pm_dream at ??:?

0x400942ac: pm_tbtt_process at ??:?

0x401a5d4e: pm_update_next_tbtt at ??:?

0x401a5dae: pp_timer_do_process at ??:?

0x40091f8a: ppTask at ??:?

E (50008619) task_wdt: Print CPU 1 backtrace

Backtrace:0x400828E5:0x3FFB1230 0x400833CD:0x3FFB1250 0x4000BFED:0x3FFBE470 0x4008D012:0x3FFBE480 0x400FDA1B:0x3FFBE4A0 0x400FDA27:0x3FFBE4D0 0x400D5E16:0x3FFBE4F0 0x4008A8DC:0x3FFBE510 
0x400828e5: esp_crosscore_isr at /home/axel/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400833cd: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4008d012: vPortExitCritical at /home/axel/esp/esp-idf/components/freertos/port/xtensa/port.c:472

0x400fda1b: esp_task_wdt_reset at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:338

0x400fda27: idle_hook_cb at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:88

0x400d5e16: esp_vApplicationIdleHook at /home/axel/esp/esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x4008a8dc: prvIdleTask at /home/axel/esp/esp-idf/components/freertos/tasks.c:3819

E (50010980) esp-tls: Failed to open new connection
E (50010985) TRANSPORT_BASE: Failed to open a new connection
E (50013619) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (50013619) task_wdt:  - IDLE (CPU 0)
E (50013619) task_wdt: Tasks currently running:
E (50013619) task_wdt: CPU 0: wifi
E (50013619) task_wdt: CPU 1: IDLE
E (50013619) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400FD7CE:0x3FFB0C30 0x400833CD:0x3FFB0C50 0x400902E2:0x3FFC7540 0x40093DC6:0x3FFC7560 0x400942AC:0x3FFC7590 0x401A5D4E:0x3FFC75B0 0x401A5DAE:0x3FFC75D0 0x40091F8A:0x3FFC75F0 
0x400fd7ce: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:191

0x400833cd: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400902e2: wifi_rf_phy_enable at ??:?

0x40093dc6: pm_dream at ??:?

0x400942ac: pm_tbtt_process at ??:?

0x401a5d4e: pm_update_next_tbtt at ??:?

0x401a5dae: pp_timer_do_process at ??:?

0x40091f8a: ppTask at ??:?

E (50013619) task_wdt: Print CPU 1 backtrace

Backtrace:0x400828E5:0x3FFB1230 0x400833CD:0x3FFB1250 0x4000BFED:0x3FFBE470 0x4008D012:0x3FFBE480 0x400FDA1B:0x3FFBE4A0 0x400FDA27:0x3FFBE4D0 0x400D5E16:0x3FFBE4F0 0x4008A8DC:0x3FFBE510 
0x400828e5: esp_crosscore_isr at /home/axel/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400833cd: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4008d012: vPortExitCritical at /home/axel/esp/esp-idf/components/freertos/port/xtensa/port.c:472

0x400fda1b: esp_task_wdt_reset at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:338

0x400fda27: idle_hook_cb at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:88

0x400d5e16: esp_vApplicationIdleHook at /home/axel/esp/esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x4008a8dc: prvIdleTask at /home/axel/esp/esp-idf/components/freertos/tasks.c:3819

E (50016020) esp-tls: empty esp_tls parameter
E (50016020) esp-tls: Failed to open new connection
E (50016020) TRANSPORT_BASE: Failed to open a new connection
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40138d75  PS      : 0x00060330  A0      : 0x80138dac  A1      : 0x3ffe1d70  
0x40138d75: esp_tls_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport_ssl.c:109 (discriminator 3)

A2      : 0x3ffbd1c0  A3      : 0x00000000  A4      : 0x00000f2b  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x3ffbd160  A8      : 0x00000600  A9      : 0x3ffe1d20  
A10     : 0x3ffbd1c0  A11     : 0x3f424b92  A12     : 0x3f424cf6  A13     : 0x02fb2f14  
A14     : 0x3f424b92  A15     : 0x3ffde528  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000006e8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace:0x40138d72:0x3ffe1d70 0x40138da9:0x3ffe1d90 0x401a3e0f:0x3ffe1db0 0x400e9847:0x3ffe1dd0
0x40138d72: esp_tls_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport_ssl.c:109 (discriminator 3)

0x40138da9: tcp_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport_ssl.c:124

0x401a3e0f: esp_transport_connect at /home/axel/esp/esp-idf/components/tcp_transport/transport.c:181

0x400e9847: esp_mqtt_task at /home/axel/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1340

ELF file SHA256: 6f65f3e279944105

Rebooting...
ets Jun  8 2016 00:22:57

### Expected Behavior
It was working fine before updating latest esp-idf.

### Actual Behavior
Device reboots.

### Steps to reproduce

My device keeps publishing data to mqtt broker.
Just run the application for a while, it hit the issue.

KaeLL, 2021-03-18:
Seems similar enough to #6743.

Alvin1Zhang, 2021-03-19:
Thanks for reporting and sorry for the inconvenience, we will look into.

AxelLin, 2021-03-19:
Hi @david-cermak 
Thanks for the quick response with the fix.
I will rebuild my application with the fix and check if it can run over the weekend.

Thanks,
Axel

AxelLin, 2021-03-19:
I got two devices reboot now.
The esp_reset_reason() shows 6 (Reset by SLC module, reset digital core).

AxelLin, 2021-03-20:
E (4092778) wifi:ieee80211_ioctl.c 1443

E (4097779) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (4097779) task_wdt:  - IDLE (CPU 0)
E (4097779) task_wdt: Tasks currently running:
E (4097779) task_wdt: CPU 0: wifi
E (4097779) task_wdt: CPU 1: IDLE
E (4097779) task_wdt: Aborting.

abort() was called at PC 0x400fc464 on core 0
0x400fc464: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:184 (discriminator 1)

Backtrace:0x4008902e:0x3ffb0b80 0x40089775:0x3ffb0ba0 0x4009043e:0x3ffb0bc0 0x400fc464:0x3ffb0c30 0x400833cd:0x3ffb0c50 0x400902e2:0x3ffc73a0 0x40093dc6:0x3ffc73c0 0x400942ac:0x3ffc73f0 0x401a5d56:0x3ffc7410 0x401a5db6:0x3ffc7430 0x40091f8a:0x3ffc7450
0x4008902e: panic_abort at /home/axel/esp/esp-idf/components/esp_system/panic.c:365

0x40089775: esp_system_abort at /home/axel/esp/esp-idf/components/esp_system/system_api.c:112

0x4009043e: abort at /home/axel/esp/esp-idf/components/newlib/abort.c:46

0x400fc464: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:184 (discriminator 1)

0x400833cd: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400902e2: wifi_rf_phy_enable at ??:?

0x40093dc6: pm_dream at ??:?

0x400942ac: pm_tbtt_process at ??:?

0x401a5d56: pm_update_next_tbtt at ??:?

0x401a5db6: pp_timer_do_process at ??:?

0x40091f8a: ppTask at ??:?

AxelLin, 2021-03-22:
Hi @espxiehang 
I reproduce the watchdog issue by simply running mqtt to publish data to broker.
I think you can modify the mqtt example code to repeating publish data.

If you test in master tree you need below fix to avoid possible panic (it's different issue from the watchdog issue)
https://github.com/espressif/esp-idf/issues/6744#issuecomment-802869633

BTW, I'm wondering if below log helps
https://github.com/espressif/esp-idf/issues/6744#issuecomment-803357073
Especially this line:
**E (4092778) wifi:ieee80211_ioctl.c 1443**

I run 7 testing devices during last weekend, 6 devices reboot due to hitting the watchdog issue.
So I think it simply can happen. But I don't know how much time is required to reproduce it.
(I don't have a better way to quickly reproduce it).

espxiehang, 2021-03-22:
Hi @AxelLin ， We are dealing with this issue and will fix it in the near future.

KaeLL, 2021-03-25:
@espxiehang any ETA?

negativekelvin, 2021-03-25:
https://github.com/espressif/esp32-wifi-lib/commit/e7fc97c714706757a39f05e17faa10b4d1c00b1d

AxelLin, 2021-03-26:
> Hi @AxelLin
> 
> Thanks for the report. I think this is related to the recent rework in [e615d10](https://github.com/espressif/esp-idf/commit/e615d10b4f19d7a55988a685dd4a0a4d1d1931e2) which make use of `esp-tls` even for non-tls connection (which is your use case, too)
> 
> There's one bug which caused the crash above is fixed here:
> [tcp_transport-Fix-error-handling-of-esp_tls_init.patch.txt](https://github.com/espressif/esp-idf/files/6171714/tcp_transport-Fix-error-handling-of-esp_tls_init.patch.txt)
> 

Hi @david-cermak 

I still got below error after apply wifi-lib update and tcp_transport-Fix-error-handling-of-esp_tls_init.patch.txt

W (9518498) TRANSPORT_BASE: Poll timeout or error, errno=Success, fd=53, timeout_ms=10000
E (9518506) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
W (9518524) MQTT_CLIENT: Publish: Losing qos0 data when client not connected
E (9536985) esp-tls: couldn't get hostname for :ingics.ddns.net: getaddrinfo() returns 202, addrinfo=0x0
E (9536985) esp-tls: Failed to open new connection
E (9536990) TRANSPORT_BASE: Failed to open a new connection
E (9536996) MQTT_CLIENT: Error transport connect

And sometimes got below if wifi is disconnected.
E (1452193) TRANSPORT_BASE: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 53
E (1452204) MQTT_CLIENT: Poll read error: 119, aborting connection

It's pretty confused about seeing "ssl_poll_read" in the message as I just use tcp in my test.

AxelLin, 2021-03-29:
@espxiehang 

This issue is marked as done but the fix is not yet synced to the esp-idf master tree,
and the v4.1 and v4.2 have been reported having the similar watchdog issue.
The v4.3-beta2 was just released, but it's not clear to me if it's impacted
by this watchdog issue. So I cannot find a stable branch to continue testing.
I'd like to check out the commit before this regression so I can test other functions.

May I know which commit causes this regression?

espxiehang, 2021-03-29:
Hi @AxelLin 
`E (4092778) wifi:ieee80211_ioctl.c 1443`
On the WiFi watchdog issue, fix commit is espressif/esp32-wifi-lib@e7fc97c

For another issue, please refer to the description of david-cermak.
_Thanks for the report. I think this is related to the recent rework in e615d10 which make use of esp-tls even for non-tls connection (which is your use case, too)_

AxelLin, 2021-03-29:
> Hi @AxelLin
> `E (4092778) wifi:ieee80211_ioctl.c 1443`
> On the WiFi watchdog issue, fix commit is [espressif/esp32-wifi-lib@e7fc97c](https://github.com/espressif/esp32-wifi-lib/commit/e7fc97c)

I know the fix, but what I want to know is which commit causes the regression.
(I don't think I can take the fix to different branches)

> For another issue, please refer to the description of david-cermak.
> _Thanks for the report. I think this is related to the recent rework in [...]

I cannot revert this commit, it causes several conflicts.
i.e.  multiple issues mixed in the master branch.
So I'm trying to find a stable branch to continue work.

espxiehang, 2021-03-29:
Hi @AxelLin 
What is the commit that causes the `E (4092778) wifi:ieee80211_ioctl.c 1443` issue is espressif/esp-idf@deb94f4.
Generally, the version with a tag is relatively stable.  So you can try the v4.3-beta1 version.
It is recommended that you wait for the next version of 4.3 tag.

AxelLin, 2021-03-29:
> Hi @AxelLin
> What is the commit that causes the `E (4092778) wifi:ieee80211_ioctl.c 1443` issue is [deb94f4](https://github.com/espressif/esp-idf/commit/deb94f4).
> Generally, the version with a tag is relatively stable. So you can try the v4.3-beta1 version.

Thanks for the information.

The v4.3-beta1 has a regression that causes OTA failure with flash encryption.
v4.3-beta2 already includes 8f951ea341721767731e18a260847f6306fc68b7.
So I think I will use v4.3-beta2 and revert 8f951ea341721767731e18a260847f6306fc68b7.

> It is recommended that you wait for the next version of 4.3 tag.

Yes, but I don't think it will be available soon.

I appreciate your support, you guys really fix the issues pretty soon.
However, it really takes very long time to sync with github (Usually 3+ weeks)
I'm wondering if it is possible to speedup syncing with github especially for the panic/reboot/hangup issues.

AxelLin, 2021-03-29:
> The v4.3-beta1 has a regression that causes OTA failure with flash encryption.
> v4.3-beta2 already includes 8f951ea341721767731e18a260847f6306fc68b7.
> So I think I will use v4.3-beta2 and revert 8f951ea341721767731e18a260847f6306fc68b7.

Oops, Just found I cannot revert the commit 8f951ea341721767731e18a260847f6306fc68b7 in wifi lib.
Otherwise, esp_wifi_init will fail and abort.
Well, the only option is v4.3-beta1 + local fix manually for the OTA issue.

espxiehang, 2021-03-29:
It is suggested to wait for the next tag version. We have done a very detailed test and it will be very stable.

AxelLin, 2021-04-13:
> Hi @AxelLin
> 
> since `esp-tls` needs about 2K RAM to create a non-tls connection. This is something we have to look into and try to optimize.

On the second thought, this is actually a bug/regression. 
It takes extra 2K RAM per connection.
So if you have 5 connections at the same time, it takes extra 10K memory.

AxelLin, 2021-04-15:
> It is suggested to wait for the next tag version. We have done a very detailed test and it will be very stable.

Just to remind the v4.3 branch has been broken for 1 month due to this wifi watchdog issue,
and this issue blocks other testing against v4.3 branch because devices frequently reboot.

closed 2021-04-21 by AxelLin

gl-chenxiaosheng, 2022-07-14:
@[espxiehang](https://github.com/espxiehang) Has this problem been fixed in v4,2 ? Moreover, this fix is now merged in which the stable IDF version, the stable version what it is ?