espressif/esp-idf #6745: Sharing LAN8720A MDC/MDIO  lines with I2C lines ?

JB-DX, 2021-03-19:
I use the ESP32 along with ethernet LAN8720A.
Because running out of free ports I hooked up a I2C unit to the MDC/MDIO lines.
Now I 'm puzzled how to call the I2C functions at the software side.
I found emac_esp32_read_phy_reg() in esp_eth_mac_esp32.c but that returns only one byte per call and I am not sure how to call it from the application side. However I can't find  the underlying functions  in order to transfer a byte string to/from the I2C unit.

Has anyone ever tried that and provide some clues?

suda-morris, 2021-03-19:
I think @yorickvP would have made a workable PHY management over IIC, thanks to his [PR](https://github.com/espressif/esp-idf/pull/5903).
I think currently we can't call I2C functions in the PHY driver without touching the IDF code. One possible solution could be extend the `esp_eth_config_t` structure to provide two callbacks, e.g. `esp_err_t eth_phy_reg_read(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value)`. This can help users to override the default implementation.

yorickvP, 2021-03-19:
We have a custom eth_phy driver that doesn't use eth_phy_reg_read/eth_phy_reg_write, but rather some i2c commands. Having the eth_phy implementation be overridable would be useful for cases such as this.

JB-DX, 2021-03-22:
Thanks for your replies !
I checked  LAN8720A closer and only now realized, that it's not I2C compatible, so my idea won't work. It first looked just too tempting.

I second  yorickvP  suggestion to make it overridable .

yorickvP, 2021-03-26:
MDIO and I2C are pretty similar. You may be able to connect the pin to both
devices and switch the pin mode when you want to output MDIO or I2C. I'd
recommend testing this first.

closed 2021-04-27 by suda-morris