espressif/esp-idf #6747: MQ2 sensor component for ESP-IDF

Yoimer, 2021-03-19:
Hello everybody. Is there any MQ2 library available for ESP-IDF?.

Currently there are these amazing libraries to work great on Arduino-ESP32 https://github.com/miguel5612/MQSensorsLib.

Is there any possible way to import those Arduino libraries and use them as native components on ESP-IDF, MQ2 specifically?. 

Thanks in advance

leeebo, 2021-05-07:
@Yoimer Thanks for your advice. But I think ESP-IDF tends to support more basic and core components, while ESP-IoT-Solution is more suitable to support such external sensor drivers. MQ series will be considered to add to https://github.com/espressif/esp-iot-solution/tree/master/components/sensors

Yoimer, 2021-05-07:
@leeebo Hello. Thanks for responding. To make things a bit easier. Is there any way you could import these [MQSensorsLib](https://github.com/miguel5612/MQSensorsLib) from Arduino ESP32 to ESP-IDF as a component?

That Arduino library happens to be very reliable.

Looking forward to hearing from you soon.

Thanks in advance.

Alvin1Zhang, 2024-11-29:
Thanks for reporting, sorry for the very slow turnaround, let us track in https://github.com/espressif/esp-iot-solution/issues, feel free to reopen.

closed 2024-11-29 by Alvin1Zhang