espressif/esp-idf #6748: ble_mesh_provisioner example

franz-ms-muc, 2021-03-19:
as by Description the "https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner" example does Provision the Examples https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node. 

and this works. 

i was using 3 Devices, 1x onoff_client, 2x onoff_server, and it happens as follows: 

all 3 Devices are provisioned by the Provisioner.

the 2 onoff_server then get a "on" for the RED LED. 

Problem: 
the onoff_client does NOT work, 

Expected: Press a Switch on the onoff_client to Operate the onoff_server. 
same as if provisioned by App.

franz-ms-muc, 2021-03-20:
https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/ble_mesh_on_atom/tree/feature/cont/ble_mesh_provisioner
this works as explected but i do not like the fact that i need to store so much locally which might be already in the lib inside. ... this can be done for sure nicer.

InfiniteYuan, 2021-06-28:
Sorry for the late reply. 

The ble_mesh_provisioner example only add the App key to the Generic OnOff Server model. So you need to add App key to the Generic OnOff Client model.

closed 2025-03-13 by Alvin1Zhang