espressif/esp-idf #6749: Sdcard write throughput drops when used in conjunction with the HTTP Server

bbernhard, 2021-03-19:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.2
- Build System:  idf.py
- Compiler version: 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

Hi everybody, 

I am using a small SDIO breakout board to connect a 8GB SDHC/SDXC card to my ESP32-DevKitC. While the write performance is pretty good (~3MB/s) when tested isolated, it significantly drops (~0.3MB/s) when the Sdcard is used in conjunction with the HTTP Server. 

Please see the two example programs below. 

### Expected Behavior

For a Sdcard connected in 4bit SDIO mode, writing to a FAT filesystem I would expect  a throughput of ~3MB/s

### Actual Behavior

Under certain circumstances the throughput drops to ~0.3MB/s

### Steps to reproduce

**Example program 1 (the fast one)**

[This](https://gist.github.com/bbernhard/9c8cab8fce23398784ca6eda2446b40d) example program here fills a 32k buffer 1000 times with `A`s and writes it to the Sdcard. 

That's the output of the program: 

```
I (340) example: Initializing SD card
I (340) example: Using SDMMC peripheral
I (350) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Name: SU08G
Type: SDHC/SDXC
Speed: 40 MHz
Size: 7580MB
I (560) example: Opening perf.txt for writing
I (10960) example: File written
I (10960) example: write time = 8106
I (10960) example: total time = 10293
I (10960) example: Card unmounted
```

**Example program 2 (the slow one)**

[This](https://gist.github.com/bbernhard/bba404e297e6e136d54a1eb1b58f2fcd) example program reads data from a HTTP post request and writes the data to the Sdcard. 

That's the output of the program:

```
I (2606) example: Starting server on port: '80'
I (2606) example: Registering URI handlers
I (4326) example: Initializing SD card
I (4326) example: Using SDMMC peripheral
I (4326) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Name: SU08G
Type: SDHC/SDXC
Speed: 40 MHz
Size: 7580MB
I (129216) example: DONE
I (129216) example: write time = 76780
I (129216) example: total time = 124692
```

As it can be seen in the output, the write time here is 76 seconds, whereas the write time in the example program above is just 8 seconds.  

### Steps to reproduce

* Connect a SDIO breakout board to the ESP32-DevKitC v4 (in 4bit SDIO mode)
* run the first example program ("the fast one"). -> you should see a throughput of ~3MB
* run the second example program ("the slow one") and post a file via curl (e.g: `curl -X POST -F 'data=@"/tmp/perf.txt"' -H 'Expect:' 192.168.2.154/upload`) -> you should see a significant drop in throughput. 

# Additional Information
* the same hardware was used in both tests
* the same sdkconfig was used in both tests
* the same Sdcard was used in both tests
* the same amount of bytes were written to the Sdcard in both tests

Does anyone have an idea what's going on here? I am stuck at this for weeks now and I am running out of ideas of what I could try to nail the problem. Am I doing something wrong here?

Thanks a lot,
Bernhard

igrr, 2021-03-19:
Hi Bernhard, from a brief look, only a difference in buffer sizes is apparent.
In the first case you have a [32 kB buffer](https://gist.github.com/bbernhard/9c8cab8fce23398784ca6eda2446b40d#file-main-c-L99), which is large and multiple of the allocation_unit_size specified in FAT settings (16kB). This allows FAT to write directly to the card, bypassing internal buffers.
In the second case, the buffer is [500 bytes](https://gist.github.com/bbernhard/bba404e297e6e136d54a1eb1b58f2fcd#file-main-c-L100), which isn't aligned to the SD card sector size (512 bytes) and much less than the FAT cluster size. Writing in 500 byte blocks requires the FAT filesystem to do read-modify-write operation sector by sector, which is quite slow.
Please try changing the buffer size in the 2nd example to match the 1st example and see if the results become closer.

bbernhard, 2021-03-19:
Hi Ivan,

thanks for the fast reply, very much appreciated!

I _think_ the buffers are the same in both examples. I am collecting the small 500 byte chunk in a 32k buffer (`totalBuf`) and writing the 32k Buffer to the Sdcard (see https://gist.github.com/bbernhard/bba404e297e6e136d54a1eb1b58f2fcd#file-main-c-L128). 

Or am I missing something here?

igrr, 2021-03-19:
Thanks for pointing this out, I have overlooked the 2nd buffer!

It seems that the amount written from the larger buffer may be less than 32k, if the next small buffer would overflow the large one. Because of this, each subsequent buffer will not be aligned to cluster boundary, potentially requiring more time due to a read-modify-write. This shouldn't reduce performance as much as 10 times, though.

I have modified the 2nd example to fill data directly into the larger 32k buffer, since it is already allocated. The example also tries to write in exactly 32k chunks. The write speed, if we consider only the time of `write` operation, is now very similar between the two examples:

1st (writing 32768000 bytes, same code as yours, only added kB/sec logging):
```
I (8960) example: write time = 6342, kB/sec=5045.73
I (8960) example: total time = 8551, kB/sec=3742.25
```

2nd (writing 33554432 bytes, modified code [here](https://gist.github.com/igrr/338dfcb448df374b286892d2d35db631/revisions)):
```
I (84609) example: write time = 6221, kB/sec=5393.73
I (84609) example: total time = 76202, kB/sec=440.33
```

Note that the total time in the 2nd case is still significantly lower. I expect that reading from socket needs to be happen in parallel with writing to the SD card, to reach higher speeds. You can send the data coming from the upload request into a [ring buffer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#ring-buffers) and run a separate task to move data from ring buffer to a file.

bbernhard, 2021-03-20:
Hi Ivan,

you were right on the money with that! 

Originally I was using two tasks (one task to receive the data and one task to write the data to the Sdcard), but when I did that, I experienced the same problems as above. The write speed was just terrible. 

I always knew that I wasn't filling up the buffers up to the last byte, but I always thought that this was "good enough" and can't be the reason for the slow writing. I've never been more wrong (I've also read up a bit on how a Sdcard & the fat filesystem works and now it makes perfectly sense to me).

I rewrote my two task based application today and made sure that every buffer is full up to the last byte before writing it to the disk. With that I was able to improve the performance of my application by a factor of 4. Receiving + writing a 18MB file now takes ~13 seconds instead of ~55 seconds. 

I guess I could probably still improve that a bit by playing with the buffer size of my fifo/ringbuffer and by some code improvements. But with that performance I am already REALLY happy. 

Thank you so much @igrr for taking the time to help me fix this issue. Very much appreciated!

igrr, 2021-03-21:
Good to know you got it working! I'm going to close this issue now.

If your code isn't proprietary, would you mind posting a link to a gist with the new, faster version? I'm sure this will help those who may stumble upon this issue report in the future.

closed 2021-03-21 by igrr

bbernhard, 2021-03-21:
> If your code isn't proprietary, would you mind posting a link to a gist with the new, faster version? I'm sure this will help those who may stumble upon this issue report in the future.

Currently it's still closed source, but I am planning to change the license soon (just need to get some legal things in order). I'll add a link here once it's open source :)