espressif/esp-idf #6751: Coredump partition initialization broken

KaeLL, 2021-03-19:
[This](https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/components/espcoredump/src/core_dump_checksum.c#L145) call with these parameters breaks the app since [this](https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/components/espcoredump/src/core_dump_checksum.c#L131) doesn't do NULL check.
Also, why the SHA256 variant does the calculation on spot, but the CRC32 one doesn't?
@o-marshmallow thoughts?

o-marshmallow, 2021-03-21:
Hello @KaeLL ,
Thank your for your message, it is indeed a bug. The fact is I already noticed this bug and created an internal _Merge Request_ in order to fix it with other bugs related to core dump's fake stack. Once it is merged and passes internal tests, it will be push to Github.

Regarding your second question, I am not sure to understand: all the checksum calculation are performed in `esp_core_dump_checksum_update` function. For both CRC32 and SHA256, external functions are used. (`esp_rom_crc32_le` and `mbedtls_sha256_update_ret` respectively)

KaeLL, 2021-03-22:
@o-marshmallow Good to know the bug's already been fixed.
And disregard the second question, I mistakenly assumed `mbedtls_sha256_finish_ret` was doing the calculation.

closed 2021-04-01 by espressif-bot