espressif/esp-idf #6757: ESP-IDF and Toolchain version from AWS FreeRTOS ESP32 V202002.00

Yoimer, 2021-03-20:
Hello there. I need to test the AWS FreeRTOS ESP32 V202002.00 zile file automatically generated on AWS Console. However, I don't find which exact version of ESP-IDF and Toolchain is required to do so. I created a Github repo https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00 which has all the files from the zip file. Even I got a version file, https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/esp-idf/version.txt which says `v3.3-163-g601a03e`, however I don't find that tag release from `espressif/esp-idf`.
Could you please, help me out telling me the right tools I need to tests the demos? I plan to work on this using Windows.
Thanks in advance.

chegewara, 2021-03-20:
This is esp-idf API issue tracker, not @Yoimer issue tracker, not even AWSFreeRTOS issue tracker.
Here you can find few useful links:
https://github.com/espressif/esp-idf/issues/new/choose

https://github.com/aws/amazon-freertos/issues?q=is%3Aissue+is%3Aopen+esp32

Yoimer, 2021-03-21:
@chegewara Ok thanks. I will ask my question there

Alvin1Zhang, 2021-03-22:
@chegewara Thanks for the notes, @Yoimer will close now.

closed 2021-03-22 by Alvin1Zhang