espressif/esp-idf #6758: "Header fields are too long for server to interpret"   form chrom

cqwuxiaolong, 2021-03-21:
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
chrom:"Header fields are too long for server to interpret"  
IE:正常

negativekelvin, 2021-03-21:
https://github.com/espressif/arduino-esp32/issues/2983

Alvin1Zhang, 2021-03-22:
@negativekelvin Thanks for sharing the notes, @cqwuxiaolong would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2021-03-23:
@cqwuxiaolong 

Please try increasing HTTP max header size, ref:

https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/components/esp_http_server/Kconfig#L3-L8

Please feel free to re-open in case there is any issue

closed 2021-03-23 by mahavirj