espressif/esp-idf #6761: Receiving events after esp_event_handler_instance_unregister

saloid, 2021-03-22:
## Environment

- Development Kit:      custom PCB
- Module or chip used:  ESP32-WROVER-B
- IDF version : v4.2-408-gc9cf7bcb0
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: VSCode
- Power Supply:        external 5V
- 
## Problem Description
I continue receive events even after `esp_event_handler_instance_unregister`.

### Expected Behavior
Stop receiving events after instance unregistering

### Actual Behavior
Event handler called even after unregistering

### Steps to reproduce

1. register events handler with `esp_event_handler_instance_register`. I am using custom base and `ESP_EVENT_ANY_ID`
2. post some events with any ID - all works as it should
3. delete (unregister) event handler with `esp_event_handler_instance_unregister`. Make sure that return value == 0
4. post to event again. Handler from unregistered event instance will be called, but it shouldn't!

Alvin1Zhang, 2021-03-23:
Thanks for reporting.

0xjakob, 2021-06-28:
@saloid Thanks for bringing this up and sorry for the long delay!
In fact, you need to provide the correct base and event id to `esp_event_handler_instance_unregister()` to successfully unregister, even if you use the event handler instance-based functions. The reason for this the current implementation of `esp_event` as well as compatibility to the `old` API where you don't have an instance object for each registration.

0xjakob, 2021-06-28:
@saloid OK, disregard the comment before. I just read the documentation again and it's probably a bug. Let me investigate a bit and I'll come back to you.

saloid, 2021-06-28:
If i'll try to unregister with same base - I'm getting the same issue

0xjakob, 2021-06-30:
@saloid Do you have a reproducer for this? I just tested and it works fine for me registering/unregistering a handler with `ESP_EVENT_ANY_ID` events if using the same base.

0xjakob, 2021-06-30:
We still need to think if we change the behavior or require users to keep a record of the registrations and allow only specific de-registrations. Registering all handlers of some base or even the whole event loop has the risk of accidentally de-registering events from user users on the same system (e.g. Wifi) form the event loop. This needs to be carefully thought through.

saloid, 2021-06-30:
@0xjakob, I'll prepare example on this week. Thank you for dealing with this issue.

0xjakob, 2021-07-07:
Hi @saloid,

do you have any updates regarding the reproducer?

Regarding the original issue: We decided against changing the implementation to unregister *all* corresponding events when using `ESP_EVENT_ANY_ID` or `ESP_EVENT_ANY_BASE`. This might easily create new bugs where users accidentally unregister events from other users or components. E.g., WiFi uses the esp_event.

saloid, 2021-07-07:
Sorry for delay. I'm still working on small example. Looks like that this issue doesn't appear outside of my project. I maybe it's because of newer version of IDF or something else. I will investigate more tomorrow.

saloid, 2021-07-08:
I don't see this error im my full project to. Maybe it was fixed since that time. Thank you for your support.

And regarding unregistering with `ESP_EVENT_ANY_ID` - it works now as it should (at least as I expect it). Here is my example (which was supposed to be a reproduce): https://github.com/saloid/esp_reproduce_IDFGH-4969/blob/master/src/main.c

Alvin1Zhang, 2021-07-09:
Reopen since we have a documentation fix under internal reviewing now, the issue will close automatically once the fix is available on GitHub. Thanks.

closed 2021-07-25 by espressif-bot