espressif/esp-idf #6762: gatt_security_server removed the bonded device

rolan82, 2021-03-22:
## Environment

- Module or chip used:  ESP32-WROOM-32U
- IDF version:     v4.4-dev-479-g1067b2870
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    Linux
- Using an IDE?: VSCode
- Power Supply:         dc/dc 3.3V

## Problem Description

I observe that my gatt_security_server has been removed the bonded device. I have previously successfully paired with my phone using a pincode (I am using NRF connect Nordic).
Then I started making NRF connect / disconnect with my esp32. On some of the attempts, I receive an invitation to enter the pincode. I see that the pair in the phone has been deleted.

### Expected Behavior
I think that if the connection is broken due to a bad connection, then the pair should not be destroyed and the bonded devices should not be deleted.

### Steps to reproduce
I started researching the problem. For tests, I used the example  [gatt_security_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server). 
[my test code](https://gist.github.com/rolan82/8be44be649b5f4b910de8593ec6cf716) . I made some minor changes to the example:
1. when creating a pair, you need to enter a pincode
```cpp
esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT; // ESP_IO_CAP_NONE; // set the IO capability to No output No input
```
2. changed connection parameters
```cpp
        case ESP_GATTS_CONNECT_EVT:
            ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_CONNECT_EVT");
            /* start security connect with peer device when receive the connect event sent by the master */
            esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);

            esp_ble_conn_update_params_t conn_params = {};
            memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
            /* For the iOS system, please refer to Apple official documents about the BLE connection parameters restrictions. */
            conn_params.latency = 0;
            conn_params.max_int = 0x10;//20; // max_int = 0x20*1.25ms = 40ms
            conn_params.min_int = 0x6;//10; // min_int = 0x10*1.25ms = 20ms
            conn_params.timeout = 400;  // timeout = 400*10ms = 4000ms
            //start sent the update connection parameters to the peer device.
            esp_ble_gap_update_conn_params(&conn_params);

            break;
```
**Steps to reproduce:**
1. Successfully pair with your phone using pincode using NRF
2. Disconnect the antenna from esp32 to simulate poor connection with the phone
3. Try to disconnect / connect your phone to esp32 and get unpairing

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
https://gist.github.com/rolan82/8be44be649b5f4b910de8593ec6cf716

## Debug Logs
I have attached the logs [fail_pairing.log](https://github.com/espressif/esp-idf/files/6184028/fail_pairing.log)
```
show a successful connection attempt
I (106177) SEC_GATTS_DEMO: remote BD_ADDR: 61ca9e33460e
I (106177) SEC_GATTS_DEMO: address type = 1
I (106177) SEC_GATTS_DEMO: pair status = success
I (106187) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (106197) SEC_GATTS_DEMO: Bonded devices number : 2
...
then a few unsuccessful and at the end I get a prompt to re-pair and enter the picode
I (153617) SEC_GATTS_DEMO: The passkey Notify number:123456

```
Cheers
rolan82

Alvin1Zhang, 2021-03-23:
Thanks for reporting and sharing the details, we will look into.

xulongzhuang, 2021-03-24:
@rolan82 , I followed your steps to reproduce the problem you encountered, but unfortunately, I did not reproduce it. The environment is the same as you described above. I set the distance between the mobile phone and ESP32-WROOM-32U until ESP_GATTS_DISCONNECT_EVT was printed and started broadcasting, and then I reconnected. Repeat the above process, I did not encounter unpairing situation ，the following is part of my log :

> I (426918) SEC_GATTS_DEMO: 4c 5e e1 cf 88 71 
I (427648) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=16, min_int=6, max_int=16, latency=0, conn_int=36, timeout=500
I (428148) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=0, min_int=0, max_int=0, latency=0, conn_int=12, timeout=400
I (430088) SEC_GATTS_DEMO: ESP_GATTS_WRITE_EVT, write value:
I (430088) SEC_GATTS_DEMO: 01 
I (434258) SEC_GATTS_DEMO: ESP_GATTS_WRITE_EVT, write value:
I (434258) SEC_GATTS_DEMO: 01 
I (443598) SEC_GATTS_DEMO: ESP_GATTS_WRITE_EVT, write value:
I (443608) SEC_GATTS_DEMO: 01 
I (445598) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (445618) SEC_GATTS_DEMO: advertising start success
I (810838) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (812848) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=16, min_int=6, max_int=16, latency=0, conn_int=36, timeout=500
I (813448) SEC_GATTS_DEMO: remote BD_ADDR: 4c5ee1cf8871
I (813448) SEC_GATTS_DEMO: address type = 1
I (813448) SEC_GATTS_DEMO: pair status = success
I (813448) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (813458) SEC_GATTS_DEMO: Bonded devices number : 1
I (813458) SEC_GATTS_DEMO: Bonded devices list : 1
I (813468) SEC_GATTS_DEMO: 4c 5e e1 cf 88 71 
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
I (825238) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8
I (825258) SEC_GATTS_DEMO: advertising start success
I (855568) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (857578) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=16, min_int=6, max_int=16, latency=0, conn_int=36, timeout=500
I (858538) SEC_GATTS_DEMO: remote BD_ADDR: 4c5ee1cf8871
I (858538) SEC_GATTS_DEMO: address type = 1
I (858538) SEC_GATTS_DEMO: pair status = success
I (858538) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (858548) SEC_GATTS_DEMO: Bonded devices number : 1
I (858548) SEC_GATTS_DEMO: Bonded devices list : 1
I (858558) SEC_GATTS_DEMO: 4c 5e e1 cf 88 71 
I (859118) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=0, min_int=0, max_int=0, latency=0, conn_int=12, timeout=400
I (868698) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x1f
I (868718) SEC_GATTS_DEMO: advertising start success
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
I (916048) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (917038) SEC_GATTS_DEMO: remote BD_ADDR: 4c5ee1cf8871
I (917038) SEC_GATTS_DEMO: address type = 1
I (917038) SEC_GATTS_DEMO: pair status = success
I (917038) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (917048) SEC_GATTS_DEMO: Bonded devices number : 1
I (917048) SEC_GATTS_DEMO: Bonded devices list : 1
I (917058) SEC_GATTS_DEMO: 4c 5e e1 cf 88 71 
I (917928) SEC_GATTS_DEMO: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT, status=0, min_int=6, max_int=16, latency=0, conn_int=12, timeout=400

Therefore, is it because my reproducing steps are missing? Please describe in more detail, and can you provide a complete gatt_security_server log?

rolan82, 2021-03-24:
@xulongzhuang, thanks for helping solve my problem

I am able to reproduce the problem regularly (about 1 time out of 10 attempts). What am I doing:
1. Disconnected the antenna from esp32
2. Placed the phone at such a distance where it is still possible to start the connection process
3. At the moment of connection, I move the phone to a distance where it is not possible to successfully complete the connection.

What components do I need to include to make the logging more informative?

xulongzhuang, 2021-03-24:
@rolan82 ，I have reproduced the problem you encountered. The simple way is to extend their distance to a position where they cannot be connected during connection authentication. The following is the log of unbinding or unpairing: 
```
E (153856) BT_BTM: Device not found

I (153856) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8
I (153856) SEC_GATTS_DEMO: remote BD_ADDR: 51514d53d991
I (153866) SEC_GATTS_DEMO: address type = 1
I (153866) SEC_GATTS_DEMO: pair status = fail
I (153876) SEC_GATTS_DEMO: fail reason = 0x66
I (153876) SEC_GATTS_DEMO: Bonded devices number : 0

I (153886) SEC_GATTS_DEMO: Bonded devices list : 0
```
I think "fail reason = 0x66" is important, which may help us find the key to the problem. If you just want to not delete the pairing or binding information, please remove the bta_dm_remove_sec_dev_entry (bda) function(About line 4696) call in [bta_dm_act.c ](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/bta/dm/bta_dm_act.c)，This might help you

rolan82, 2021-03-24:
@xulongzhuang  Thanks for the quick response. i will check this solution

rolan82, 2021-03-25:
@xulongzhuang  I found that the pair gets deleted when I get the following codes: 0x66, 0x63, 0x52.
Is it correct if I don't call bta_dm_remove_sec_dev_entry for these cases?
```cpp
if (sec_event.auth_cmpl.fail_reason! = 0x66
 && sec_event.auth_cmpl.fail_reason! = 0x63
 && sec_event.auth_cmpl.fail_reason! = 0x52) {
    bta_dm_remove_sec_dev_entry (bda);
 }
```
Or do I need to do something else?

xulongzhuang, 2021-06-22:
@rolan82 
0x66 and other errors may also be encountered during the first pairing, if you perform the above operations. This kind of established connection is insecure. Maybe their keys have not been distributed, Therefore, this device will be removed from the underlying IDF code and paired again next time  
You can also make a judgment without removing the device. But this may have unknown security risks