espressif/esp-idf #6763: allow the ICMP echo-respond-buffer to be modified before replying

JB-DX, 2021-03-22:
My app needs to read and modify a ping reply message before the reply is sent back. This way a custom ping-sender can examine specifics of the ping target.

So a callback is required that passes a pointer to the message buffer, before it gets sent.

Currently I have added this to icmp.c
```
//---------------------------------------
// added global function pointer:
bool (*onIcmpMessage)(u8_t *msg);   //  returns: TRUE if Msg has been modified
...

icmp_send_response() {
 [...]
  // my addition follws  right after this existing line 
      iecho = (struct icmp_echo_hdr *)p->payload;

    if (onIcmpMessage && onIcmpMessage(p->payload))  // these 3 lines are added 
     {
       iecho->chksum= 0;
       iecho->chksum= inet_chksum_pbuf(p);
     }
 [...]
//---------------------------------------

```
usage example in the app:
```
//---------------------------------------
  onIcmpMessage= IcmpMessageEvt;

static bool IcmpMessageEvt(u8_t *b)
{
    b += 20;
    if (!memcmp(b, "key", 4))
     {
      u8_t cmd = b[5];
      if (cmd == '?')   // if  info query then ...
       {
        b[4] = '!';  // set reply cmd
        b[5] = myFlags;
        return true;
     }
    }
  return false;  
}
//---------------------------------------

```

david-cermak, 2021-04-01:
Hi @JB-DX 

I'm sorry but we have to turn down this feature request. It would mean update of lwip in a non standard way (which we generally try to avoid) but most importantly this change would imply a violation the ICMP protocol defined in https://tools.ietf.org/html/rfc792.
Please check the description of echo request and echo replies at page 15:
```
      The data received in the echo message must be returned in the echo
      reply message.
```

It seems that you're trying to misuse ICMP as an application protocol. Would it help to use some UDP socket for that or a mdns discovery if you're on the local link?

JB-DX, 2021-04-01:
We have used this ping-data-method with our previous h/w for 15 years and never experienced any problems. We use it to indentify whether it is our type of terminal or someone else responding to a ping. Also some parameters like port number get transferred back to the ping sender. So we always get important settings back even if port or TCP/UDP is blocked by firewalls or routers etc. along the way. UDP broadcast is not an option as it is limited to the subnet.

This method has helped us fixing LAN configuration problems at many customers sites. It's just a very reliable method.
It is not used to replace TCP or UDP for normal data exchange

Why not let the app writer decide to (mis-)use or not use it?

david-cermak, 2021-04-06:
@JB-DX I believe that ICMP tunneling is a powerful technique, but, as mentioned already, we generally try to avoid system level modification and encouraging users to break standard internet control protocols.

> Why not let the app writer decide to (mis-)use or not use it?

I think it's not a big problem for apps to misuse (even system) protocols, if they do so in the user space. The suggested technique would however result in a non-trivial modification of lwip and IDF.

It is possible to achieve the same with
* custom modification of IDF (we could help to post a patch, but that won't be an official SDK update)
* lwip update to allow for ICMP response injection (similar to `LWIP_HOOK_...` macro to enable the app layer to inject a custom handle. Would need the lwip upstream to accept the changes)
* complete application level API with `SOCK_RAW` sockets (we can also help to provide such example)

Clearly, the last option seems to be the most preferred, as it needs no SDK modification nor any system protocol injection into the lwip. IDF supports the raw API:
```c
/**
 * LWIP_RAW==1: Enable application layer to hook into the IP layer itself.
 */
#define LWIP_RAW                        1
```
Which allows the application to receive the raw ICMP packets and respond to them independently from the system layers. (apps might need to disable the system ICMP, though). We would only need to create a raw socket on the ICMP protocol:
```c
socket(AF_INET, SOCK_RAW, IP_PROTO_ICMP);
```

JB-DX, 2021-04-08:
I'm not sure how complicated the last option would be on the app side. 
In any case you could leave it unchanged, because I can easily add the code myself into icmp.c as I have done in the past. The only drawback: I have to do it during each IDF update. But I can live with that.

david-cermak, 2021-04-09:
Thanks for the feedback, will close the issue as no change to IDF components.

 > how complicated the last option would be on the app side.
Very simple, just spin a thread and `readfrom()` the raw socket (no need for a separate thread, if your app already has some housekeeping task running), but we can help here.

> only drawback: I have to do it during each IDF update. 
Another drawback is that you would use unofficial/unsupported IDF. This is not recommended.

@JB-DX Please let us know if you like to receive this example demonstrating raw socket operations with ICMP,  we could post the code here (it is not suitable as a standard IDF example)

closed 2021-04-09 by Alvin1Zhang

JB-DX, 2021-04-09:
@david-cermak  Yes I would like to get an example, so that I could try that.

david-cermak, 2021-04-10:
Here is a very simple example for testing this approach (no task, error handling and setting socket options omitted, only IPv4)
```c
#include "esp_event.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"
#include "lwip/sockets.h"
#include "lwip/inet_chksum.h"
#include "lwip/icmp.h"
#include "lwip/ip.h"

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(example_connect());

    static char buf[256];
    int len;
    struct sockaddr_storage from;
    int fromlen = sizeof(from);
    int sock = socket(AF_INET, SOCK_RAW, IP_PROTO_ICMP);
    while ((len = recvfrom(sock, buf, sizeof(buf), 0, (struct sockaddr *)&from, (socklen_t *)&fromlen)) > 0) {
        if (from.ss_family == AF_INET && len >= sizeof(struct ip_hdr) + sizeof(struct icmp_echo_hdr)) {
            struct icmp_echo_hdr *icmp = (struct icmp_echo_hdr *) (buf + sizeof(struct ip_hdr));
            // modify buf[sizeof(struct ip_hdr) + sizeof(struct icmp_echo_hdr) ...]
            icmp->chksum = icmp->type = 0;
            icmp->chksum = inet_chksum(icmp, len - sizeof(struct ip_hdr));
            sendto(sock, icmp, len - sizeof(struct ip_hdr), 0, (struct sockaddr *) &from, sizeof(from));
        }
    }
}
```

Disable default ICMP config from project makefile:
```cmake
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(icmp-socket-reply)

idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DLWIP_ICMP=0")
```

JB-DX, 2021-05-03:
Sorry for the long silence, too many other projects needed my attention.
I just implemented the above suggestion, and it works very well. That's really the better approach on the long run. 
Thank you *very* much !!

JB-DX, 2023-12-30:
after migrating to idf v5.1.2.dirty the above solution does not work anymore.
Now two ping reply messages are sent, the first one with the original content, the second with the modified content.
How can I prevent the first message?

JB-DX, 2024-01-17:
I finally found the reason: 
this has no more effect:
```
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DLWIP_ICMP=0")
```
instead ICMP has to be disabled in menuconfig