espressif/esp-idf #6765: Task watchdog got triggered at vTaskDelayUntil

whong0951, 2021-03-23:
## Environment
Module: ESP32-WROOM-32U
IDF version: ESP-IDF v4.2-198-g2aafcd9b1
Operating system: Ubuntu
## Traceback

abort() was called at PC 0x400d4634 on core 0 Backtrace:0x40090702:0x3ffbe2c0 0x40090e6d:0x3ffbe2e0 0x40097b7a:0x3ffbe300 0x400d4634:0x3ffbe370 0x400829a1:0x3ffbe390 0x4000bfed:0x3ffdf0d0 0x400910a1:0x3ffdf0e0 0x40081d36:0x3ffdf100 0x40081d5d:0x3ffdf120 0x400928ef:0x3ffdf140 0x400dc1b0:0x3ffdf160 ELF file SHA256: 00cb80ddaead8bae

0x400d4634: task_wdt_isr at /home/whong/workspace/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)

0x40090702: panic_abort at /home/whong/workspace/mower-display_tube/build/../components/esp_system/panic.c:363

0x40090e6d: esp_system_abort at /home/whong/workspace/mower-display_tube/build/../components/esp_system/system_api.c:106

0x40097b7a: abort at /home/whong/workspace/esp/esp-idf/components/newlib/abort.c:46

0x400d4634: task_wdt_isr at /home/whong/workspace/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)

0x400829a1: _xt_lowint1 at /home/whong/workspace/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400910a1: vPortExitCritical at /home/whong/workspace/esp/esp-idf/components/freertos/xtensa/port.c:419

0x40081d36: esp_crosscore_int_send at /home/whong/workspace/esp/esp-idf/components/esp32/crosscore_int.c:104

0x40081d5d: esp_crosscore_int_send_yield at /home/whong/workspace/esp/esp-idf/components/esp32/crosscore_int.c:115

0x400928ef: vTaskDelayUntil at /home/whong/workspace/esp/esp-idf/components/freertos/tasks.c:1419

0x400dc1b0: process_task at /home/whong/workspace/mower-display_tube/build/../main/src/app/rw_display_process.c:372

## Phenomenon
The machine resets occasionally, the frequency is very low.

projectgus, 2021-03-23:
Hi @whong0951,

Can you give any information about the pattern your task uses when calling `vTaskDelayUntil(arg)`?

Is it possible that the argument is always in the past, ie `xTaskGetTickCount() <= arg`? If this is true then it's possible to trigger Task WDT, as the task is never actually suspended.

Do you also have the [interrupt watchdog timer](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-reference/kconfig.html#config-esp-int-wdt) enabled in your project config?

whong0951, 2021-03-23:
> Hi @whong0951,
>
> Can you give any information about the pattern your task uses when calling `vTaskDelayUntil(arg)`?
>
> Is it possible that the argument is always in the past, ie `xTaskGetTickCount() <= [...]

*****
Hi projectgus,
I call this function in my main loop, just like this：
```
    static void process_task(void *args)
    {
        TickType_t previous_waketime = xTaskGetTickCount();
        uint32_t notify_value = 0;
        while (1)
        {
            if (data_pri->running_flag == RW_TRUE)
            {
                vTaskDelayUntil(&previous_waketime, pdMS_TO_TICKS(1000 / PROCESS_CONTROL_RATE_HZ));
                ……
            }
        }
    }
```
and `previous_waketime` hasn't appeared elsewhere. So I think it  wouldn't appear `xTaskGetTickCount() <= arg`.
I enable the interrupt watchdog timeout (300ms) and the task watchdog timeout (5s).

Tanks for your help.

projectgus, 2021-03-23:
@whong0951 ,

Thanks for the update, agree that this looks OK.

What is the value of `PROCESS_CONTROL_RATE_HZ`, and what is the project config setting for [CONFIG_FREERTOS_HZ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-hz)?

Is it possible that this CPU core is sometimes so busy that it is constantly processing data, so process_task is running behind or at maximum capacity?

whong0951, 2021-03-24:
@projectgus 
`#define PROCESS_CONTROL_RATE_HZ 100`
`CONFIG_FREERTOS_HZ=1000`
This block is executed per 10ms. And it takes much less than 10ms. Unless something wrong was happened, like it is blocked in interrupt...
I am trying to reproduce this phenomenon to get more information...
I enabled ble, wifi, mqtt client in my project.

Alvin1Zhang, 2021-06-08:
@whong0951 Thanks for reporting and sharing progress, would you please help share if any updates for this issue? Thanks.

whong0951, 2021-06-08:
@Alvin1Zhang 
I have never seen this again. I thought it may be caused by some bug in my project. Perhaps … memory leak.
I will close this issue. Thanks for ur help.

closed 2021-06-08 by whong0951

Alvin1Zhang, 2021-06-09:
Glad to hear the issue didn't happen, and thanks for sharing updates.