espressif/esp-idf #6766: If BLE authentication fails with error code 0x66, then device bond information gets removed

Digitelektro, 2021-03-23:
## Environment
- IDF version v4.2
- BLE stack: bluedroid

## Problem Description

I have a wearable device with an ESP32 working as a BLE peripheral with bond enabled. As the user walks around the BLE central which is collecting the data, at a certain distance the connections is very unstable. At this point the device may disconnect and reconnect continuously. As the signal is too weak, sometimes it disconnects during the authentication phase. When this happens, the next time when the device is in range, it wants to pair again, and from the log of the central device (which is not an ESP32) I see that one party may lose its key, and passkey has to be given again.

I spent many hours to find what causes this behavior, and I think I found the root cause. But I am not sure what would be the proper way to fix this issue as I don’t see what side effects will arise by removing the related codes.
So first problem in file bta_dm_act.c, see comments for the appropriate line:

```
static UINT8 bta_dm_ble_smp_cback (tBTM_LE_EVT event, BD_ADDR bda, tBTM_LE_EVT_DATA *p_data) {
....
case BTM_LE_COMPLT_EVT:
        bdcpy(sec_event.auth_cmpl.bd_addr, bda);
#if BLE_INCLUDED == TRUE
        BTM_ReadDevInfo(bda, &sec_event.auth_cmpl.dev_type, &sec_event.auth_cmpl.addr_type);
#endif
        p_name = BTM_SecReadDevName(bda);
        if (p_name != NULL) {
            BCM_STRNCPY_S((char *)sec_event.auth_cmpl.bd_name, p_name, (BD_NAME_LEN));
        } else {
            sec_event.auth_cmpl.bd_name[0] = 0;
        }
        if (p_data->complt.reason != 0) {
            sec_event.auth_cmpl.fail_reason = BTA_DM_AUTH_CONVERT_SMP_CODE(((UINT8)p_data->complt.reason));
            /* delete this device entry from Sec Dev DB */
            bta_dm_remove_sec_dev_entry (bda);    //Problem is comming from here, removing this line fixes the issue
        } else {
            sec_event.auth_cmpl.success = TRUE;
            if (!p_data->complt.smp_over_br) {

            }
        }
        sec_event.auth_cmpl.auth_mode = p_data->complt.auth_mode;
        if (bta_dm_cb.p_sec_cback) {
            //bta_dm_cb.p_sec_cback(BTA_DM_AUTH_CMPL_EVT, &sec_event);
            bta_dm_cb.p_sec_cback(BTA_DM_BLE_AUTH_CMPL_EVT, &sec_event);
        }

        break;
....
}
```

Other problem is in btc_dm.c. If I don't remove the line marked with a comment, after ESP32 reset the bonding process is started again if there was a timeout during authentication.
```
static void btc_dm_ble_auth_cmpl_evt (tBTA_DM_AUTH_CMPL *p_auth_cmpl)
{
    /* Save link key, if not temporary */
    BTC_TRACE_DEBUG("%s, status = %d", __func__, p_auth_cmpl->success);
    bt_status_t status = BT_STATUS_FAIL;
    int addr_type;
    bt_bdaddr_t bdaddr;
    bdcpy(bdaddr.address, p_auth_cmpl->bd_addr);
    bdcpy(btc_dm_cb.pairing_cb.bd_addr, p_auth_cmpl->bd_addr);

    if (p_auth_cmpl->success) {
        status = BT_STATUS_SUCCESS;
        BTC_TRACE_DEBUG ("%s, -  p_auth_cmpl->bd_addr: %08x%04x", __func__,
                             (p_auth_cmpl->bd_addr[0] << 24) + (p_auth_cmpl->bd_addr[1] << 16) + (p_auth_cmpl->bd_addr[2] << 8) + p_auth_cmpl->bd_addr[3],
                             (p_auth_cmpl->bd_addr[4] << 8) + p_auth_cmpl->bd_addr[5]);

        // Check if need to save BLE keys
        if((p_auth_cmpl->auth_mode & SMP_AUTH_GEN_BOND) == 0) {
            return;
        }

         if (btc_storage_get_remote_addr_type(&bdaddr, &addr_type) != BT_STATUS_SUCCESS) {
            btc_storage_set_remote_addr_type(&bdaddr, p_auth_cmpl->addr_type, true);
        }
        btc_storage_set_ble_dev_auth_mode(&bdaddr, p_auth_cmpl->auth_mode, true);
        btc_dm_save_ble_bonding_keys();
    } else {
        /*Map the HCI fail reason  to  bt status  */
        switch (p_auth_cmpl->fail_reason) {
        case BTA_DM_AUTH_SMP_PAIR_AUTH_FAIL:
        case BTA_DM_AUTH_SMP_CONFIRM_VALUE_FAIL:
            btc_dm_remove_ble_bonding_keys();
            status = BT_STATUS_AUTH_FAILURE;
            break;
        case BTA_DM_AUTH_SMP_PAIR_NOT_SUPPORT:
            status = BT_STATUS_AUTH_REJECTED;
            break;
        default:
            btc_dm_remove_ble_bonding_keys();   //Removing this is also necessary
            status =  BT_STATUS_FAIL;
            break;
        }

    }

#if (CONFIG_BT_STACK_NO_LOG)
    (void) status;
#endif
    BTC_TRACE_DEBUG("%s, authentication status = %x", __func__, status);
    return;

}
```

I think the error code should be checked for the reason of the timeout, and bond information shouldn’t be deleted if only a timeout occurred.

### Expected Behavior

Already bonded device pairing data shouldn't be removed if only a timeout occurs

### Steps to reproduce
Hard to reproduce, I have a device working as a BLE central, and I keep turning it on and off and trying to turn it off at the right time. I think the right time is sometime after connection params update, which can be identified from the central’s log.

## Debug Logs

```
BLE-SEC: >>>> Authentication COMPLETE <<<<
BLE-SEC: Pair Status: FAIL
BLE-SEC:  - fail reason: 0x66
```

xulongzhuang, 2021-03-24:
@Digitelektro ，I think [IDFGH-4970](https://github.com/espressif/esp-idf/issues/6762) is similar to this issue. Using the method in it, the 0x66 error code can be reproduced, and the solution you described above is effective for the problem of binding and pairing removed. We will evaluate Whether this solution will have other effects, thank you very much for your feedback.

Digitelektro, 2021-03-24:
Thank you for the investigation. Yes, that is the same issue.

Here is my fix it that I use now:
bta_dm_act.c
```
    case BTM_LE_COMPLT_EVT:
        bdcpy(sec_event.auth_cmpl.bd_addr, bda);
#if BLE_INCLUDED == TRUE
        BTM_ReadDevInfo(bda, &sec_event.auth_cmpl.dev_type, &sec_event.auth_cmpl.addr_type);
#endif
        p_name = BTM_SecReadDevName(bda);
        if (p_name != NULL) {
            BCM_STRNCPY_S((char *)sec_event.auth_cmpl.bd_name, p_name, (BD_NAME_LEN));
        } else {
            sec_event.auth_cmpl.bd_name[0] = 0;
        }
        if (p_data->complt.reason != 0) {
            sec_event.auth_cmpl.fail_reason = BTA_DM_AUTH_CONVERT_SMP_CODE(((UINT8)p_data->complt.reason));
            /* delete this device entry from Sec Dev DB */
            bta_dm_remove_sec_dev_entry (bda);
            if(sec_event.auth_cmpl.fail_reason != 0x66) {
                bta_dm_remove_sec_dev_entry (bda);
            }
```
btc_dm.c
```
/*Map the HCI fail reason  to  bt status  */
switch (p_auth_cmpl->fail_reason) {
case BTA_DM_AUTH_SMP_PAIR_AUTH_FAIL:
case BTA_DM_AUTH_SMP_CONFIRM_VALUE_FAIL:
    btc_dm_remove_ble_bonding_keys();
    status = BT_STATUS_AUTH_FAILURE;
    break;
case BTA_DM_AUTH_SMP_PAIR_NOT_SUPPORT:
    status = BT_STATUS_AUTH_REJECTED;
    break;
case 0x66:
    break;
default:
    btc_dm_remove_ble_bonding_keys();
    status =  BT_STATUS_FAIL;
    break;
}
```

xulongzhuang, 2021-06-28:
hi，@Digitelektro
0x66 timeout will also appear in the first pairing. At this time, the exchanged key may not have been distributed yet, which makes the subsequent link encryption unreliable, so I think the developer may remove the device based on this consideration. You have made the above changes in your code, please verify whether there will be some bugs, thank you very much for your solution

Digitelektro, 2021-06-30:
Hi @xulongzhuang ,
I have been using the aforementioned solution since I published it and I never had a problem. I've never seen 0x66 during first pairing, that is might be another issue.

xulongzhuang, 2021-07-01:
Hi ,@Digitelektro
The probability of 0x66 error in the first pairing may be very small, but it can be reproduced. For example, if you take an esp-32 without an antenna, such as esp-wroom-32u, when the pairing is about to succeed, immediately put esp-wroom-32u Move far away, this error will appear

cichytom, 2024-10-31:
Hi, I have four eQ-3 bluetooth thermostats (firmware 1.49) that were detected and added in the eQ-3 integration in Home Assistant (2024.10.4). Thermostats communicate with the [GL.iNet GL-S10 v1.0 gateway](https://blakadder.com/gl-s10/). I noticed in the ESPHome (2024.7.3) logs on this gateway that there was a connection problem. After each restart of this gateway, either no thermostat will connect at all, or one, maximum three, will connect, but after the next restart there is a problem with the connection again. This is what it looks like in the logs. Is there any solution for this?

[09:22:06][D][esp32_ble_client:110]: [0] [00:1A:22:0A:0F:C3] ESP_GATTC_CONNECT_EVT
[09:22:06][D][esp32_ble_client:110]: [0] [00:1A:22:0A:0F:C3] ESP_GATTC_OPEN_EVT
[09:22:06][I][esp32_ble_client:067]: [1] [00:1A:22:15:6E:38] 0x00 Attempting BLE connection
[09:22:06][I][esp32_ble_client:333]: [0] [00:1A:22:0A:0F:C3] auth complete. remote BD_ADDR: 001a220a0fc3
[09:22:06][E][esp32_ble_client:336]: [0] [00:1A:22:0A:0F:C3] auth fail reason = 0x66
[09:22:06][D][esp32_ble_client:172]: [0] [00:1A:22:0A:0F:C3] ESP_GATTC_DISCONNECT_EVT, reason 14
[09:22:06][D][esp32_ble_client:110]: [0] [] ESP_GATTC_CLOSE_EVT
[09:22:08][I][bluetooth_proxy:282]: [0] [00:1A:22:0A:0E:A4] Connecting v3 without cache
[09:22:08][D][esp32_ble_client:110]: [2] [00:1A:22:0A:0E:ED] ESP_GATTC_WRITE_CHAR_EVT
[09:22:08][D][esp32_ble_client:110]: [2] [00:1A:22:0A:0E:ED] ESP_GATTC_NOTIFY_EVT
[09:22:09][D][esp32_ble_client:110]: [1] [00:1A:22:15:6E:38] ESP_GATTC_CONNECT_EVT
[09:22:09][D][esp32_ble_client:110]: [1] [00:1A:22:15:6E:38] ESP_GATTC_OPEN_EVT
[09:22:09][I][esp32_ble_client:067]: [0] [00:1A:22:0A:0E:A4] 0x00 Attempting BLE connection
[09:22:09][I][esp32_ble_client:333]: [1] [00:1A:22:15:6E:38] auth complete. remote BD_ADDR: 001a22156e38
[09:22:09][E][esp32_ble_client:336]: [1] [00:1A:22:15:6E:38] auth fail reason = 0x66
[09:22:09][D][esp32_ble_client:172]: [1] [00:1A:22:15:6E:38] ESP_GATTC_DISCONNECT_EVT, reason 14
[09:22:09][D][esp32_ble_client:110]: [1] [] ESP_GATTC_CLOSE_EVT
...

ambamo, 2024-11-16:
I have the same problem as cichytom. I think it's more of a device problem. I have 6 eq3 with the same firmware v1.48, 3 don't want to connect to 2 modules bluetooth-proxy
[logs_esphome-web-salon-scale_run.txt](https://github.com/user-attachments/files/17787038/logs_esphome-web-salon-scale_run.txt)
[logs_esp32-bluetooth-proxy-bureau_run.txt](https://github.com/user-attachments/files/17787039/logs_esp32-bluetooth-proxy-bureau_run.txt)

cichytom, 2024-11-18:
I found information [HERE](https://esphome.io/components/bluetooth_proxy.html) that "maximum three" results from the limitation of Bluetooth Proxy - ESPHome. I currently use [THIS](https://github.com/softypit/esp32_mqtt_eq3) app with an external antenna connected to ESP32 which works with all my thermostats.

EuleMitKeule, 2025-02-07:
Just for your information: I got the thermostats to connect again by restarting both the ESP and the thermostat (removing the battery) and then reflashing the ESPs with the new PIN.