espressif/esp-idf #6767: CMake Warning in

franz-ms-muc, 2021-03-23:


franz-ms-muc, 2021-03-23:
Building in: C:\\git2\\CTIS\\blink\\build  
cmake -G Ninja -DCMAKE\_TOOLCHAIN\_FILE=C:\\Users\\mail\\esp-idf-v4.3-beta1\\tools\\cmake\\toolchain-esp32.cmake -DCMAKE\_EXPORT\_COMPILE\_COMMANDS=ON -DIDF\_TARGET=esp32 C:\\git2\\CTIS\\blink  
\-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.30.0.windows.2")   
\-- The C compiler identification is GNU 8.4.0  
\-- The CXX compiler identification is GNU 8.4.0  
\-- The ASM compiler identification is GNU  
\-- Found assembler: C:/Users/mail/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe  
\-- Check for working C compiler: C:/Users/mail/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe  
\-- Check for working C compiler: C:/Users/mail/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works  
\-- Detecting C compiler ABI info  
\-- Detecting C compiler ABI info - done  
\-- Detecting C compile features  
\-- Detecting C compile features - done  
\-- Check for working CXX compiler: C:/Users/mail/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe  
\-- Check for working CXX compiler: C:/Users/mail/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works  
\-- Detecting CXX compiler ABI info  
\-- Detecting CXX compiler ABI info - done  
\-- Detecting CXX compile features  
\-- Detecting CXX compile features - done  
\-- Building ESP-IDF components for target esp32  
\-- Checking Python dependencies...  
Python requirements from C:\\Users\\mail\\esp-idf-v4.3-beta1\\requirements.txt are satisfied.  
\-- Project sdkconfig file C:/git2/CTIS/blink/sdkconfig  
Loading defaults file C:/git2/CTIS/blink/sdkconfig.defaults...  
\-- Found PythonInterp: C:/Users/mail/.espressif/python\_env/idf4.3\_py3.8\_env/Scripts/python.exe (found version "3.8.7")   
\-- Could NOT find Perl (missing: PERL\_EXECUTABLE)   
CMake Warning (dev) at C:/Users/mail/esp-idf-v4.3-beta1/components/mbedtls/CMakeLists.txt:113 (target\_sources):  
 Policy CMP0076 is not set: target\_sources() command converts relative paths  
 to absolute.  Run "cmake --help-policy CMP0076" for policy details.  Use  
 the cmake\_policy command to set the policy and suppress this warning.

 A private source from a directory other than that of target "mbedcrypto"  
 has a relative path.  
This warning is for project developers.  Use -Wno-dev to suppress it.

\-- App "blink" version: v1.20d-15-g0ad94c9-dirty

franz-ms-muc, 2021-03-23:
this warning comes in RED Colour. 

CMake Warning (dev) at C:/Users/mail/esp-idf-v4.3-beta1/components/mbedtls/CMakeLists.txt:113 (target\_sources):  
 Policy CMP0076 is not set: target\_sources() command converts relative paths  
 to absolute.  Run "cmake --help-policy CMP0076" for policy details.  Use  
 the cmake\_policy command to set the policy and suppress this warning.

 A private source from a directory other than that of target "mbedcrypto"  
 has a relative path.  
This warning is for project developers.  Use -Wno-dev to suppress it.

projectgus, 2021-03-25:
Hi @franz-ms-muc,

Thanks for letting us know about this! This warning is produced when running CMake 3.16 and above, because this default behaviour changed. CMake continues to use the old behaviour when building ESP-IDF, so there should be no issue apart from the warning itself (although a future CMake version will remove the old behaviour, and this might break something). So for now it should be safe to ignore.

We will update the ESP-IDF build system to remove the warning, ASAP.

Angus

closed 2021-06-06 by espressif-bot