espressif/esp-idf #6771: idf.py requires python instead of python3

hadmut, 2021-03-23:
Hi, 
I'm currently trying to build and pack the esp-idf toolchain inside an (alpine 3.13 linux-base) docker image. 

git clone --recursive https://github.com/espressif/esp-idf.git
./install.sh
. ./export.sh

Things work up to install, but export.sh fails for several reasons:

# . ./export.sh
Setting IDF_PATH to '/root/esp-idf'
Detecting the Python interpreter
Checking "python" ...
bash: python: command not found
Checking "python3" ...
Python 3.8.8
"python3" has been detected
Adding ESP-IDF tools to PATH...
WARNING: directory for tool xtensa-esp32-elf version esp-2020r3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s2-elf version esp-2020r3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s3-elf version esp-2020r3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool riscv32-esp-elf version 1.24.0.123_64eb9ff-8.4.0 is present, but tool was not found
ERROR: tool riscv32-esp-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32ulp-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32s2ulp-elf has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
WARNING: directory for tool openocd-esp32 version v0.10.0-esp32-20200709 is present, but tool was not found
ERROR: tool openocd-esp32 has no installed versions. Please run '/root/esp-idf/install.sh' to install it.
env: can't execute 'idf.py': No such file or directory

although it detects that there is python3 only and not python (deprecated old python2) idf.py still starts with
 
#!/usr/bin/env python

so it is hardcoded to use python2 which is absent.

chegewara, 2021-03-23:
https://github.com/espressif/esp-idf/issues/5908

igrr, 2021-03-24:
I think the python 2/3 problem near the end is simply the result of install script not completing successfully, it fails before it can set up a Python virtual environment. The main issue here is failure to verify that the tools installed by install.sh can run.

Errors such as `ERROR: tool openocd-esp32 has no installed versions. Please run '/root/esp-idf/install.sh' to install it.` typically indicate that the program couldn't be executed, typically due to missing shared libraries or due to architecture mismatch.

On Alpine, it is most likely the former, and most likely related to glibc. Please run `ldd` on one of the executables to check if all the shared libraries are found. E.g. `find $HOME/.espressif -name xtensa-esp32-elf-gcc | xargs ldd`.

dobairoland, 2021-03-24:
Yes, "python" will be set up after the export script finishes successfully and the hardcoding in the sheebang won't matter.

We are currently working on improving these error messages so the installer will print a more useful error message.

Please run the command suggested by igrr and then we will advise you further.

dobairoland, 2021-04-24:
I'm closing this for lack of activity. In the meantime, we have improved the error message: https://github.com/espressif/esp-idf/commit/ae6f602b6bb53c47cb938f510c1dae57eb0f58a9

closed 2021-04-24 by dobairoland

DhruvaG2000, 2021-09-07:
`find $HOME/.espressif -name xtensa-esp32-elf-gcc | xargs ldd`

gives me the output `not a dynamic executable`

dot4qu, 2022-07-15:
If seeing this on an M1 mac it's because Rosetta is not yet installed. Open an Intel app to get the prompt to install Rosetta 2.

nekopsykose, 2022-07-22:
the prebuilt binaries are linked against glibc and so won't work on a system with a different libc

listout, 2022-07-22:
@nekopsykose is there any way to build the tool chain ourselves? With musl linc. 

I tried building the esp32ulp-binutil and it errored out while compiling. On a relevant issues someone said to downgrade gcc.

For now I've been using a mix of container and chroot

nekopsykose, 2022-07-22:
> is there any way to build the tool chain ourselves?  

no idea, i assume this literal repository and the .zip in releases with all the code in it is what you want, and the instructions would be somewhere. but the issue is just that the prebuilt binaries won't work on a musl system, and that's all

listout, 2022-07-22:
> prebuilt binaries won't work on a musl system, and that's all

:(

nekopsykose, 2022-07-22:
i guess https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-setup-scratch.html tells you how? but you'd have to figure it out yourself, and any intermediate issues of needing to patch things to work with musl you'd have to do yourself.  

also all of this has nothing to do with this issue :)