espressif/esp-idf #6772: Secure Boot V2 native_ota example not working (v4.4-dev)

projectgus, 2021-03-24:
Thank you so much for the reply.
my requirement is also  **"Verify app signature on update without secure boot"** but yet this code is in progress and not supported in this release i have enable secure boot v2 for my ESP32-S2 in native-ota-example and **i face one another issue while ota update start "native_ota_example: Error: SSL data read error" with secure boot v2 enable**. Below is the steps i have followed.

- In menuconfig “Security Features” set “Enable hardware Secure Boot in bootloader” to enable Secure Boot.
- Change partition_table offset to 0x10000 as secure boot loader size is large.
- create self sign-certificate and copy into server_certs/ directory.
- Build the native_ota_example
- Flash bootloader as secure boot enable need to flash bootlaoder independently.
- Flash partition table, signed native-ota.bin using `$idf.py -p (PORT) flash`   command. and reset the board.
- It run correctly bootloader is verify the native-ota.bin signature also fine.
- Now i sign the hello_word.bin using `espsecure.py sign_data --version 2 --keyfile ./my_signing_key.pem --output ./image_signed.bin image-unsigned.bin
` command
- use the above signed binary for the ota update.
-  start https server at host PC and esp32 give below logs

> I (7859) native_ota_example: Starting OTA example
I (7859) native_ota_example: Running partition type 0 subtype 0 (offset 0x00020000)
I (10119) native_ota_example: Writing to partition subtype 16 at offset 0x120000
I (10119) native_ota_example: New firmware version: v4.4-dev-479-g1067b2870-dirty
I (10119) native_ota_example: Running firmware version: 1
I (10129) native_ota_example: esp_ota_begin succeeded
D (16849) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
**E (16849) native_ota_example: Error: SSL data read error**
E (16849) native_ota_example: Exiting task due to fatal error...

**Please help me how can i resolve this issue for native_ota example.**
Please find attached full log for more details.

**Note:** I have tried above steps secure boot v2 with **simple_ota example** then it's verify the hello_word.bin signature and update the firmware successfully.
Also if i use native_ota example without secure boot v2 then it working fine means **there is some bug with native_ota + secure boot v2 combination.** 
 
[native_ota_secure_boot_v2.log](https://github.com/espressif/esp-idf/files/6182954/native_ota_secure_boot_v2.log)

_Originally posted by @Darsh-Dev in https://github.com/espressif/esp-idf/issues/6212#issuecomment-804110101_

projectgus, 2021-03-24:
@Darsh-Dev Sorry to hear that native_ota is not working. The error you're seeing seems to be coming from the call to esp_http_client_read() and not to do with the app_update APIs or Secure Boot itself, so I'm unsure why it goes away with Secure Boot V2 disabled.

* Can you please give more details about your HTTPS server setup? Do you have any logs from the server during OTA?
* If you set the [default log level](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-log-default-level) to Debug, are you able to please re-run the OTA update with and without Secure Boot V2 enabled and post both logs?

Thanks in advance.

Alvin1Zhang, 2021-03-24:
Thanks for reporting.

Darsh-Dev, 2021-03-24:
Hi @projectgus 

Below are our HTTPS server setup steps.
- Create one directory on the host PC for HTTPS server and create a self-sign certificate using the below command 

> $ mkdir test_https_server                                                       
$ cd test_https_server                                                          
$ openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes

**Note:** When to create a self-signed certificate make sure we enter HTTPS server/host PC IP address into the Common Name parameter
```                                                                             
i.e Common Name (e.g. server FQDN or YOUR name) []:<Your htpps server ip address>
```                                                                                                                                                         
- Copy generated ca certificate into `server_certs` directory.                  
```                                                                             
$ cp test_https_server/ca_cert.pem <example path>/server_certs/                 
```                                                                             
- Run the HTTPS server using the below command
```                                                                    
$ openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070   
```                                                                    

Below is the server-side log during OTA.

> $ openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070
Using default temp DH parameters
ACCEPT
FILE:hello-world.bin

**Note:** server-side log is similar in both the case with secure boot and without secure boot. the change is at secure boot time we have replaced hello-world.bin with signed hello-world.bin.

Please find "native_ota_debug_without_secure_boot.log" working log "native_ota_debug_with_secure_boot.log" failed log with debug option enabled.
[native_ota_debug_without_secure_boot.log](https://github.com/espressif/esp-idf/files/6196164/native_ota_debug_without_secure_boot.log)
[native_ota_debug_with_secure_boot.log](https://github.com/espressif/esp-idf/files/6196165/native_ota_debug_with_secure_boot.log)

shubhamkulkarni97, 2021-03-24:
Hi @Darsh-Dev,

Thanks for reporting this issue and sharing detailed logs.

Can you try running OTA Example with a python server instead of openssl server?
Here are some tips to run local python HTTPS server: https://github.com/espressif/esp-idf/tree/master/examples/system/ota#local-https-server

Please let us know if OTA is successful with same binary file on python server.

negativekelvin, 2021-03-24:
@shubhamkulkarni97 your own code suggestion 
https://github.com/espressif/esp-idf/issues/6165#issuecomment-734794341

conflicts with https://github.com/espressif/esp-idf/blob/c49d03b8eb9cb2070960927cbd7194645c44a874/examples/system/ota/native_ota_example/main/native_ota_example.c#L212

Regarding whether return value of -1 can happen after a valid data transfer based on patch https://github.com/espressif/esp-idf/commit/4c166085fb6c6ee67361a1e7af461887659e6184

Is openssl server violating spec or is this a valid case that needs to be handled properly?

Darsh-Dev, 2021-03-25:
Hi @shubhamkulkarni97 

I am not able to run the python server, It through the below error

>$  python example_test.py build 8070
Traceback (most recent call last):
  File "example_test.py", line 11, in <module>
    import ttfw_idf
**ModuleNotFoundError: No module named 'ttfw_idf'**

I have also tried setting the "PYTHONPATH" path

> $ export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH"
$ python example_test.py build 8070
Traceback (most recent call last):
  File "example_test.py", line 11, in <module>
    import ttfw_idf
  File "/home/einfochips/project/panko/sourcecode/esp-idf-latest/esp-idf/tools/ci/python_packages/ttfw_idf/__init__.py", line 21, in <module>
    import junit_xml
**ModuleNotFoundError: No module named 'junit_xml'**

And can you please clarify why the simple_ota + secure_boot v2 example works fine with the OpenSSL HTTPS server?

shubhamkulkarni97, 2021-03-26:
Hi @Darsh-Dev,

To fix issue with example_test.py, please install python requirements in `${IDF_PATH}/tools/ci/python_packages/ttfw_idf/requirements.txt`

I think the issue you are observing in native_ota example is because of a special case that is not handled in example. This issue is only with openssl server and not with python server. To confirm this issue, I requested you to try native_ota example with python server.

@negativekelvin,
OpenSSL server works on HTTP/1.0 protocol, it causes issues while invoking message_complete callback. We had added a fix to explicitly invoke message_complete callback in case of OpenSSL. Please check https://github.com/espressif/esp-idf/commit/4dcffdb0a9ae6e91dcd5d5ac3a1677376a7bf158.

We have refactored `esp_http_client_read` to return correct error codes in all the cases to avoid dependency on errno. It is always better to rely on `esp_http_client_is_complete_data_received` for a safer side. It looks like this issue is a special case where message_complete callback is invoked with 0 data read.

Thanks for mentioning the comment in native_ota example, will update it.

Darsh-Dev, 2021-03-30:
Hi @shubhamkulkarni97,

Thanks for the input, with the python HTTPS server I resolve the native_ota + secure_boot v2 firmware update.
But I have few queries
1. Difference between the Python server and OpenSSL server
2. If I will upload the my_signed_firmware.bin on a public server, will I face a similar issue with it?
3. Suggestion is, similar type issue can be fixed in the client-side application, not in the server, correct?

Thanks

shubhamkulkarni97, 2021-03-30:
Hi @Darsh-Dev,

Thanks for conforming that OTA works with python server.

> 1. Difference between the Python server and OpenSSL server

The python server works on HTTP/1.1 protocol while OpenSSL server works on HTTP/1.0 protocol. Due to this, OpenSSL server does not send the content length of file to be downloaded. Hence, it causes issues in invoking message_complete callback.

> 2. If I will upload the my_signed_firmware.bin on a public server, will I face a similar issue with it?

It is unlikely that you would face similar issue with public servers. Because most the public servers support HTTP/1.1 protocol.

> 3. Suggestion is, similar type issue can be fixed in the client-side application, not in the server, correct?

You are correct, this issue should be fixes on client side. Can you try the patch attached below? Please let me know if it fixes the issue while performing OTA with OpenSSL server.

[0001-esp_http_client_read-Add-check-for-esp_http_client_i.patch.zip](https://github.com/espressif/esp-idf/files/6227205/0001-esp_http_client_read-Add-check-for-esp_http_client_i.patch.zip)

Thanks,
Shubham

shubhamkulkarni97, 2021-04-05:
@Darsh-Dev Did you get a chance to try the patch? Does it fix OTA with OpenSSL server?

Darsh-Dev, 2021-04-09:
Hi @shubhamkulkarni97,

Accept my apology for the delay, I have just checked the above-attached patch and it working fine.
Thank you so much for your support.

Thanks

shubhamkulkarni97, 2021-04-12:
Thanks @Darsh-Dev for confirming that the patch fixes this issue

Batov, 2021-04-15:
@shubhamkulkarni97 I have a similar issue with HTTPS OTA from OpenSSL server when size of firmware is multiple of the 289 bytes.
On last (already empty) chunk `esp_http_client_read` returns `-1` and `esp_https_ota_perform` does not set `ESP_HTTPS_OTA_SUCCESS` state and return `ESP_FAIL`. At result `esp_https_ota_finish` does not call `esp_ota_set_boot_partition`.

Tested on 4.1.1 and 4.2 too.

Your patch for `esp_http_client_read` fix it. Thanks.

closed 2021-04-27 by espressif-bot