espressif/esp-idf #6773: otatool.py improperly reads ota data seq and crc

etchove, 2021-03-24:
## Problem Description

The method `_get_otadata_info` is unpacking ota data with `">I"` format but data is packed with `"I"` format in method `switch_ota_partition`. This causes readen data to be invalidated by `is_otadata_info_valid`... so sequence allways restarts to 0.

The otatool.py still works, but doesn't respect ota seq incease.

Alvin1Zhang, 2021-03-24:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

KonstantinKondrashov, 2021-03-24:
Hi @etchove!
The fix for master is here [12900.zip](https://github.com/espressif/esp-idf/files/6199695/12900.zip)
What is your IDF version? We will provide the backports.
Thanks.

etchove, 2021-03-25:
> Hi @etchove [1]!
> The fix for master is here 12900.zip [2]

Hi,

Thank's for your bugfix, that's perfect ! I'm using idf 4.1.

Etienne

Links:
------
[1] https://github.com/etchove
[2] https://github.com/espressif/esp-idf/files/6199695/12900.zip
[3] 
https://github.com/espressif/esp-idf/issues/6773#issuecomment-806065347
[4] 
https://github.com/

closed 2021-04-27 by espressif-bot