espressif/esp-idf #6774: esp_restart() stuck and it didn't restart/reset the ESP32.

ROYMURLI, 2021-03-24:
OS- Ubuntu
IDF- v4.3-dev-2586-g526f68239

I am calling esp_restart() function after calling checking the wifi status and if the wrong password is given to the wifi then I am restarting the device in every two hours. 

```
else if(wifi_status==202 ||wifi_status==205 || wifi_status==9 || wifi_status==6 || wifi_status==23 || wifi_status==2 || wifi_status==15)
			{
				if(condition_ck)
				{
					restrt_time = xTaskGetTickCount();
					condition_ck = false;
				}

				fprintf(f, "{\"%010u\":\"WIFIWRONGPASSWD rssi = %d\"},", (uint32_t) t, wifistr.rssi);
				if((xTaskGetTickCount() - restrt_time) >= 7200000)
					esp_restart();	
			}
```

Here I am calling esp_restart() after every 2 hours. But after calling this the task in which I have defined this function getting stuck and also device is not restarted. SDKCONFIG is attached here:-
[sdkconfig1.txt](https://github.com/espressif/esp-idf/files/6197860/sdkconfig1.txt)

I am not getting any logs on calling esp_restart() to debug the issue. Let me know if I can change anything in menuconfig to get the logs for the same.

zhangyanjiaoesp, 2021-03-26:
@ROYMURLI  Please set the log level to debug in the menuconfig to see whether there is any useful logs.
`Component config -> Log output -> Default log verbosity -> Debug `

ROYMURLI, 2021-03-26:
Hi @zhangyanjiaoesp , Please find the logs attached. It randomly stuck after giving a bt error on calling esp_restart().
[debug_logs.txt](https://github.com/espressif/esp-idf/files/6211833/debug_logs.txt)

ammaree, 2021-03-26:
@ROYMURLI 
I have had the same problem for a very long time. I tried everything in my shutdown process to clean up the environment but no luck.

After gracefully stopping and deleting all my application timers, same with all queue, and then stopping all tasks 1 by one, esp_restart would STILL not be successful. At the end of "main" task all I leaving different is that I do NOT stop wifi but leave it in STA mode.

After tracking through esp_restart() logic I found that it was consistently hanging in esp_wifi_stop(), which was called from 1 of the shutdown_handlers[].

After commenting out lines 200-203 in wifi_default.c where the shutdown handler was being registered I have had a 100% success rate over about 70 restarts.

Worth giving a try, and if it helps maybe Espressif guys can look deeper into it. 
@zhangyanjiaoesp 
[image]

zhangyanjiaoesp, 2021-03-30:
@ROYMURLI  There is bt error in the log, can you describe more about how you use BT and Wi-Fi ? And what your code is doing when you call `esp_restart()` ?

zhangyanjiaoesp, 2021-03-30:
@ammaree 
If the WiFi task need to wait for data send from some queue when running `esp_wifi_stop()` , the wifi stop will be stuck, you can check you code to see if there is any code like `xQueueSend(s_example_queue, &evt, portMAX_DELAY)`, and if you change it to `xQueueSend(s_example_queue, &evt, 512)`, it will be help.

ROYMURLI, 2021-03-30:
> @ROYMURLI There is bt error in the log, can you describe more about how you use BT and Wi-Fi ? And what your code is doing when you call `esp_restart()` ?

@zhangyanjiaoesp  I am using GATTC and GATTS along with wifi. I am using gatts to advertise the name of device all and gattc to scan a ble server sensor. Along with that wifi is used to update the data on server. I am calling this esp_restart() if the wifi password is incorrect in every 2 hours as I have mentioned this on my first comment. It is still stucking there giving me error.

ROYMURLI, 2021-03-30:
> @ROYMURLI
> I have had the same problem for a very long time. I tried everything in my shutdown process to clean up the environment but no luck.
>
> After gracefully stopping and deleting all my [...]

It worked. But, not 100%. It stuck 2 times in the same way.

blueMoodBHD, 2021-03-30:
Hi @ammaree @ROYMURLI 
Could you please add those debug log and test again? To find out which shutdown callback stuck. 
At the same time, please enable task watchdog and remember to save the .elf file.
```
diff --git a/components/esp_system/system_api.c b/components/esp_system/system_api.c
index da96bf7..e8aa65b 100644
--- a/components/esp_system/system_api.c
+++ b/components/esp_system/system_api.c
@@ -67,10 +67,12 @@ void IRAM_ATTR esp_restart_noos_dig(void)
 void IRAM_ATTR esp_restart(void)
 {
     for (int i = SHUTDOWN_HANDLERS_NO - 1; i >= 0; i--) {
+        printf("shutdown %p\n", shutdown_handlers[i]);
         if (shutdown_handlers[i]) {
             shutdown_handlers[i]();
         }
     }
+    printf("All shutdown callback success!\n");
 
     // Disable scheduler on this core.
     vTaskSuspendAll();
```

ammaree, 2021-03-30:
@zhangyanjiaoesp  As said right at the start....
All my tasks are shutdown in an organised manner. That means 
a) stopping the timer that triggers the logging of sensor data to the cloud
b) When the sensor data queue is empty, deleting the timer, the queue and the MQTT TX & RX tasks.
c) similarly the actuator time is allowed till all actuator have completed all FadeIn/On/FadeOut/Off cycles fully
d) when all user tasks are completed and queues/timers deleted, then do I call esp_restart() at end of "app_main"
 
@blueMoodBHD My application does not use shutdown_handlers  at all, when checked there was only 1 shutdown handler I could find for esp_wifi_stop(). I then followed it back to source and disabled in wifi_default.c. Since then 100's of restarts and 100% success.

blueMoodBHD, 2021-03-31:
@ammaree  Bluetooth and UART will register shutdown_handlers, too. You did not use Bluetooth, did you ?

ryan4volts, 2021-05-11:
this appears to be the same as  #2101 . Though for me using the simple hello-world app with rtc 32khz crystal enabled does this.

xueyunfei998, 2021-07-08:
hi @ROYMURLI @ammaree 

Will you delete the task before calling restart?

Recently, I met a customer who was also calling restart and got stuck.
The scene is as follows:
- The client called the esp_wifi_sta_get_ap_Info, this function will check the status of WiFi at the beginning. After checking, it will be executed in the WiFi task.
- in esp_wifi_sta_get_ap_Info function, 
At the beginning, it will go to get the lock. Before unlocking, there is a high priority task to delete, as shown in the figure above. As a result, the function cannot continue to execute, that is, the mutex cannot be unlocked

- When performing WiFi_ Stop function, to get this mutex, because the front did not release, it will be stuck.

Now I want to know if there are other tasks in your program that delete tasks before calling WiFi API or restart. I want to know more details

ammaree, 2021-07-08:
@xueyunfei998 
Each and every of my application tasks are shut down in stages based on the nature of the task.

sntp:  since it only runs every 2 hours, simply exists, no queues, timers, muteness nothing to worry about.

sensors: purely responsible for scheduling regular reading of latest values from I2C and SPI connected sensors, so task simply stops and terminates itself.

events: purely reacts to events (primarily 1-Wire iButton detects) coming from sensor task, so simply stops and terminates itself.

I2C: with sensor task stopping, I2C task completes any requests still in its queue then deletes the queue and the task deletes itself.
 
telnet server: completes any processing for a current connection then simply deletes itself.

http server: exact same as telnet server.

mqtt tx: Since purely driven by a timer generating queue entries for sensor data to be logged the timer is stopped, once all entries already in queue has been transmitted, mqtt disconnects from host, the queue and timer is deleted and the task deletes itself.

mqtt rx: Once mqtt tx task is deleted the rx task is also deleted.

actuators: stops process allowing more actuators to be [re]started, allows all currently running actuators to complete all repetitions of FadeIn/ON/Out/Off cycles, when all completed, task terminates itself.

console: last task to terminate, monitors the shutdown of all other tasks and when it is the only user task remaining, deregisters 1x idle and 1x tick hook handlers, then initiates esp_restart() as the last instruction before (possibly) exiting app_main() back to main()

Only change I have made relating the restart is patching out of the registration of the shutdown handler for esp_wifi_stop(). at no stage does any tasks even try to disconnect or stop wifi...

I have not tried to re-enable the registration of esp_wifi_stop() as shutdown handler during the last 2 months but have had many 100's of successful restarts since disabling...

closed 2021-07-18 by ROYMURLI