espressif/esp-idf #6775: esp-idf-tools-setup install error

zhy102, 2021-03-24:
i download the  "Online Installer (30 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe", and setup end up with "check widows defender ..[\]" when i used it. click the button of "FUll log" :
2021-03-24 22:40:16.923   * Checking Windows Defender 
2021-03-24 22:40:16.925   Checking PSMODULEPATH for Windows Defender module
2021-03-24 22:40:16.925   Checking Windows Services Defender is enabled: (Get-MpComputerStatus).AntivirusEnabled
2021-03-24 22:40:16.925    $ powershell -ExecutionPolicy Bypass "if((Get-MpComputerStatus).AntivirusEnabled) { Exit 0 } else { Exit 1 }"
2021-03-24 22:40:18.425   Get-MpComputerStatus : 操作失败，出现以下错误: 0x800106ba
                          所在位置 行:1 字符: 5
                          + if((Get-MpComputerStatus).AntivirusEnabled) { Exit 0 } else { Exit 1  ...
                          +     ~~~~~~~~~~~~~~~~~~~~
                              + CategoryInfo          : NotSpecified: (MSFT_MpComputerStatus:ROOT\Microsoft\...pComputerStatus) [Get-MpComputerS 
                             tatus], CimException
                              + FullyQualifiedErrorId : HRESULT 0x800106ba,Get-MpComputerStatus

i try to open the Windows Defender , but nothing. please help me.

dobairoland, 2021-03-25:
> i try to open the Windows Defender , but nothing. please help me.

What do you mean by this? If your Windows is broken then please fix it before trying to install ESP-IDF.

Please link the full installation log, not just the last lines.

georgik, 2021-03-25:
@zhy102 Thank you for reporting the issue.
The warning that you see in the log indicates that Windows Defender is not enabled. Most likely, because there is another Antivirus installed. This does not mean any problem. It's just an information that registration of Exceptions to Windows Defender won't be executed at the end of installation.

The installation should proceed without any problem. Please, click just Next button.

georgik, 2021-03-25:
@zhy102 Update: Meanwhile we updated our Windows Installer, now it detects name of antivirus and displays it in System Check page. This helps us to avoid the problem with PowerShell error in the log.

The new 2.7 beta version is available here: https://github.com/espressif/idf-installer/releases/tag/online-2.7-beta-06
It would help us if you can test the installer in your environment.

zhy102, 2021-03-26:
> @zhy102 Update: Meanwhile we updated our Windows Installer, now it detects name of antivirus and displays it in System Check page. This helps us to avoid the problem with PowerShell error in the log.
> 
> The new 2.7 beta version is available here: https://github.com/espressif/idf-installer/releases/tag/online-2.7-beta-06
> It would help us if you can test the installer in your environment.
> @zhy102 Update: Meanwhile we updated our Windows Installer, now it detects name of antivirus and displays it in System Check page. This helps us to avoid the problem with PowerShell error in the log.
> 
> The new 2.7 beta version is available here: https://github.com/espressif/idf-installer/releases/tag/online-2.7-beta-06
> It would help us if you can test the installer in your environment.

thanks, but i used this new version, errors still as follows:
2021-03-26 15:43:54.410   Log opened. (Time zone: UTC+08:00)
2021-03-26 15:43:54.411   Setup version: Inno Setup version 6.1.2
2021-03-26 15:43:54.411   Original Setup EXE: C:\Users\Administrator\Desktop\esp-idf-tools-setup-online-2.7-beta-06.exe
2021-03-26 15:43:54.411   Setup command line: /SL5="$15064A,2249822,780800,C:\Users\Administrator\Desktop\esp-idf-tools-setup-online-2.7-beta-06.exe" 
2021-03-26 15:43:54.411   Compatibility mode: Yes (DetectorsMessageBoxErrors)
2021-03-26 15:43:54.411   Windows version: 10.0.18363  (NT platform: Yes)
2021-03-26 15:43:54.411   64-bit Windows: Yes
2021-03-26 15:43:54.411   Processor architecture: x64
2021-03-26 15:43:54.411   User privileges: None
2021-03-26 15:43:54.411   Administrative install mode: No
2021-03-26 15:43:54.411   Install mode root key: HKEY_CURRENT_USER
2021-03-26 15:43:54.411   64-bit install mode: Yes
2021-03-26 15:43:54.425   Created temporary directory: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp
2021-03-26 15:43:54.432   -- DLL function import --
2021-03-26 15:43:54.432   Function name: idpAddFile
2021-03-26 15:43:54.432   DLL name: files:idp.dll
2021-03-26 15:43:54.433   Extracting temporary file: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.446   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.446   Importing the DLL function.
2021-03-26 15:43:54.455   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.455   -- DLL function import --
2021-03-26 15:43:54.455   Function name: idpAddFileComp
2021-03-26 15:43:54.455   DLL name: files:idp.dll
2021-03-26 15:43:54.455   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.455   Importing the DLL function.
2021-03-26 15:43:54.455   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.455   -- DLL function import --
2021-03-26 15:43:54.455   Function name: idpAddMirror
2021-03-26 15:43:54.455   DLL name: files:idp.dll
2021-03-26 15:43:54.455   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.455   Importing the DLL function.
2021-03-26 15:43:54.456   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.456   -- DLL function import --
2021-03-26 15:43:54.456   Function name: idpAddFtpDir
2021-03-26 15:43:54.456   DLL name: files:idp.dll
2021-03-26 15:43:54.456   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.456   Importing the DLL function.
2021-03-26 15:43:54.456   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.456   -- DLL function import --
2021-03-26 15:43:54.456   Function name: idpAddFtpDirComp
2021-03-26 15:43:54.456   DLL name: files:idp.dll
2021-03-26 15:43:54.456   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.456   Importing the DLL function.
2021-03-26 15:43:54.456   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.456   -- DLL function import --
2021-03-26 15:43:54.456   Function name: idpClearFiles
2021-03-26 15:43:54.456   DLL name: files:idp.dll
2021-03-26 15:43:54.456   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.456   Importing the DLL function.
2021-03-26 15:43:54.456   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.456   -- DLL function import --
2021-03-26 15:43:54.456   Function name: idpFilesCount
2021-03-26 15:43:54.456   DLL name: files:idp.dll
2021-03-26 15:43:54.456   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
2021-03-26 15:43:54.456   Importing the DLL function.
2021-03-26 15:43:54.456   Successfully imported the DLL function. Delay loaded? No
2021-03-26 15:43:54.456   -- DLL function import --
2021-03-26 15:43:54.456   Function name: idpFtpDirsCount
2021-03-26 15:43:54.456   DLL name: files:idp.dll
2021-03-26 15:43:54.457   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\idp.dll
[... 481 lines cut ...]
                              main(sys.argv[1:])
                            File "C:\esp32-v3.3\\tools\idf_tools.py", line 1345, in main
                              action_func(args)
                            File "C:\esp32-v3.3\\tools\idf_tools.py", line 1155, in action_install_python_env
                              subprocess.check_call([sys.executable, '-m', 'virtualenv', '--no-site-packages', idf_python_env_path],
                            File "subprocess.py", line 364, in check_call
[... cut ...]
2021-03-26 15:49:14.565   User chose Yes.
2021-03-26 15:49:14.649   CurStepChanged raised an exception.
2021-03-26 15:49:14.696   Need to restart Windows? No
2021-03-26 15:49:14.703   Exception message:
2021-03-26 15:49:14.703   Message box (OK):
                          Out Of Range.
2021-03-26 15:49:14.730   User chose OK.
2021-03-26 15:49:14.730   Deinitializing Setup.
2021-03-26 15:49:14.742   Failed to remove temporary directory: C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp
2021-03-26 15:49:14.751   Log closed.

georgik, 2021-03-26:
@zhy102 Thank you for the test and log.
I see there several problems in the log. First of all it seems that your Antivirus: 360 瀹夊叏濂楄 is blocking internet connection to dl.espressif.com and github.com. Here is the part of log:
```
powershell -ExecutionPolicy Bypass Invoke-WebRequest -Uri "https://dl.espressif.com/dl/?system_check=win10.00.18363" -OutFile "C:\Users\ADMINI1\AppData\Local\Temp\is-4ROA2.tmp\check-1.txt";Invoke-WebRequest -Uri "https://github.com/espressif" -OutFile "C:\Users\ADMINI~1\AppData\Local\Temp\is-4ROA2.tmp\check-2.txt";
2021-03-26 15:45:02.890 Invoke-WebRequest : 无法连接到远程服务器
所在位置 行:1 字符: 152
+ ... check-1.txt;Invoke-WebRequest -Uri https://github.com/espressif -OutF ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
+ CategoryInfo : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest]，WebExce
ption
+ FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand
```

Please configure your antivirus so that it allows the connection. If it's not possible to configure the antivirus to allow network connection, you can download Offline installer which contains all necessary files. The Offline installer is available only for IDF 4.1.1 and 4.2. You can find it here: https://dl.espressif.com/dl/esp-idf/

The other issue is caused by old ESP-IDF v3.3 which you have installed on your machine. Please choose option Download ESP-IDF on the third screen instead of selecting existing IDF. This will provide you with option to download fresh version of ESP-IDF v3.3.5 or tag v3.3.5 from GitHub. Please, also consider upgrading to newer release of ESP-IDF, because v3.3 is relatively close to End-of-Maintenance period: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html

georgik, 2021-04-08:
@zhy102 Were you able to resolve the problem with antivirus?
Meanwhile we also updated an installer, feel free to download latest 2.7 beta release which is fixing several minor issues: https://github.com/espressif/idf-installer/releases

tom-borcin, 2021-04-09:
Hi @zhy102, thank you for reporting this. Unfortunately, we are unable to reproduce your issue and because of that we are closing it

closed 2021-04-09 by tom-borcin