espressif/esp-idf #6777: export.sh : xtensa-esp32s2-elf has no installed versions

DigKleppe, 2021-03-25:
I installed the latest idf.
runned install.sh as root, else no permission.
Tools are present.
IDF_PATH is present.
sourced export.sh:

Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.8.5
"python3" has been detected
Adding ESP-IDF tools to PATH...
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/home/dig/projecten/esp32/esp-idf/install.sh' to install it.
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '/home/dig/projecten/esp32/esp-idf/install.sh' to install it.
ERROR: tool riscv32-esp-elf has no installed versions. Please run '/home/dig/projecten/esp32/esp-idf/install.sh' to install it.
ERROR: tool openocd-esp32 has no installed versions. Please run '/home/dig/projecten/esp32/esp-idf/install.sh' to install it.
env: ‘idf.py’: No such file or directory

running install again:

dig@dig-desktop:~/projecten/esp32/esp-idf$ sudo ./install.sh
[sudo] password for dig: 
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.8.5
"python3" has been detected
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping riscv32-esp-elf@1.24.0.123_64eb9ff-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Installing Python packages from /home/dig/projecten/esp32/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring python-socketio: markers 'python_version <= "2.7"' don't match your environment
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 4)) (54.1.2)
Requirement already satisfied: click>=7.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 11)) (3.4.6)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 13)) (0.27)
Requirement already satisfied: gdbgui==0.13.2.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 18)) (0.9.0.2)
Requirement already satisfied: kconfiglib==13.7.1 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 22)) (13.7.1)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 25)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 26)) (3.1.7)
Requirement already satisfied: ecdsa>=0.16.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 27)) (0.16.1)
Requirement already satisfied: construct==2.10.54 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 31)) (2.10.54)
Requirement already satisfied: Flask<1.0,>=0.12.2 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (0.12.5)
Requirement already satisfied: gevent<2.0,>=1.2.2 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.5.0)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (2.8.1)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.9.0)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (2.9.6)
Requirement already satisfied: cffi>=1.12 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 11)) (1.14.5)
Requirement already satisfied: six>=1.9.0 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from ecdsa>=0.16.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 27)) (1.15.0)
Requirement already satisfied: pycparser in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cffi>=1.12->cryptography>=2.1.4->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 11)) (2.20)
Requirement already satisfied: itsdangerous>=0.21 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.1.0)
Requirement already satisfied: Jinja2>=2.4 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (2.11.3)
Requirement already satisfied: Werkzeug<1.0,>=0.7 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (0.16.1)
Requirement already satisfied: brotli in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Flask-Compress<2.0,>=1.4.0->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.0.9)
Requirement already satisfied: python-socketio>=1.6.1 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (5.1.0)
Requirement already satisfied: greenlet>=0.4.14 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gevent<2.0,>=1.2.2->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.0.0)
Requirement already satisfied: MarkupSafe>=0.23 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Jinja2>=2.4->Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (1.1.1)
[... cut ...] lask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (0.21.2)
Requirement already satisfied: python-engineio>=4 in /root/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /home/dig/projecten/esp32/esp-idf/requirements.txt (line 15)) (4.0.1)
All done! You can now run:

  . ./export.sh
The previous idf ( form may 2020) runs ok.
Ubuntu 20.
Regards, Dig

dobairoland, 2021-03-25:
Hi @DigKleppe. You are installing the tools into `/root/.espressif/` so they cannot be found in `/home/dig/.espressif` for using them as non-root.

Using `sudo` in this case in not recommended. The install script should install the tools for the "dig" user.

DigKleppe, 2021-03-25:
Hi
Thanks for the quick answer.
If I run the install script as non root i get a permission error. The normal (old)  tools folder has normal user rights. ???

I see the tools present in the tools folder.
like /home/dig/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/
alongside the tools of the previous installation.

dobairoland, 2021-03-25:
Could you please post the full log to see at what stage the permission error is happening? 

Yes, the `.espressif` folder should have normal user rights. You can examine the output of `ls -al` and fix the directories by `chmod` and `chown` commands. 

If you don't care about the content of `.espressif` then you can remove it and let `install.sh` re-create it.

DigKleppe, 2021-03-29:
Removed .espressif folder, installed tools.
Now everything is ok!
Thanks,
Dig

dobairoland, 2021-03-29:
Thank you @DigKleppe for the feedback. I'm happy that it got resolved.

I don't think that this require any fix. Adding any removal to these scripts would be too invasive. I'm closing this issue.

closed 2021-03-29 by dobairoland