espressif/esp-idf #6778: NimBLE mesh on ESP32 ( Provisioning and configuration data Persistance storage issuse) #951

bilalmalik76, 2021-03-25:
I am using NimBLE mesh on ESP32. But Provisioning and configuration data is not being stored on Persistance storage. I have enabled BLE_MESH_SETTINGS. But there is no config.h file.

I am getting an error of config.h file not found.

prasad-alatkar, 2021-03-25:
Hi @bilalmalik76 , can you please let me know which reference example you are trying to use ? Can you also help to share the sdkconfig as well ? Can you also please post the build error here?

bilalmalik76, 2021-03-26:
Hi @prasad-alatkar, Thanks for your reply. I am using example available in this folder: 
`esp-idf/examples/bluetooth/nimble/blemesh/`
This is the SDK config file I am using. 
`#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

#
# Security features
#
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
[... 1237 lines cut ...]
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
`
I have enabled the BLE_MESH_SETTINGS from C:\esp\esp-idf\components\bt\host\nimble\port\include\esp_nimble_cfg by making this change:
`#ifndef MYNEWT_VAL_BLE_MESH_SETTINGS
#define MYNEWT_VAL_BLE_MESH_SETTINGS (1)
#endif`
When I build the project after enabling _BLE_MESH_SETTINGS, I am getting this build error:
`In file included from C:/Users/Lenovo/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/mesh/src/glue.c:20:
C:/Users/Lenovo/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/mesh/include/mesh/glue.h:56:10: fatal error: config/config.h: No such file or directory
 #include "config/config.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.`

prasad-alatkar, 2021-04-06:
Hi @bilalmalik76 , we don't have `config/config.h` anywhere in NimBLE port. As per my understanding, the NimBLE upstream code owners have provided `BLE_MESH_SETTINGS` option to remove dependency of NimBLE port on `config/config.h`. So, it is recommended to disable `BLE_MESH_SETTINGS`. I would like to request you to take a look at `esp_ble_mesh` which supports both NimBLE and Bluedroid, ref: [esp-ble-mesh](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh). This is Espressif's official BLE Mesh stack. Please let me know if you have any other concerns (CC: @dhrishi  )

bilalmalik76, 2021-04-06:
Yes, you are right. I have already moved my application to ESP-BLE-MESH
stack and able to store provisioning data in persistence storage. I also
find
Espressif's official BLE Mesh stack more easy to use and configuration
options are also more easily accessible from menueconfig.

Thanks for your help.

prasad-alatkar, 2021-04-30:
@Alvin1Zhang, Please help to close the issue.

closed 2021-05-06 by Alvin1Zhang