espressif/esp-idf #6779: MQTT TCP example error

larisqueiroz, 2021-03-25:
Hello.
So, I'm using ESP32 (DOIT Wroom-32) to run the mqtt tcp example and I got an error (below) during the build execution. A few weeks ago, I used this example and it worked, but yesterday I noticed that the app_main.c of the tcp example in my machine was different than the current one on the repository. I believe that some changes were made and the code is even more easier to understand. Right now, the "new" tcp example is not working for me and the old one is. But I really wanted to be able to run the current one for a project. 
Thank you!
PS: I'm using the ESP-IDF Command Prompt and I didn't do any changes in the original code.

```
../main/app_main.c: In function 'mqtt_event_handler':
../main/app_main.c:96:48: error: 'MQTT_ERROR_TYPE_TCP_TRANSPORT' undeclared (first use in this function); did you mean 'MQTT_ERROR_TYPE_ESP_TLS'?
         if (event->error_handle->error_type == MQTT_ERROR_TYPE_TCP_TRANSPORT) {
                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                MQTT_ERROR_TYPE_ESP_TLS
../main/app_main.c:96:48: note: each undeclared identifier is reported only once for each function it appears in
../main/app_main.c:99:96: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             log_error_if_nonzero("captured as transport's socket errno",  event->error_handle->esp_transport_sock_errno);
                                                                                                ^~~~~~~~~~~~~~~~~~~~~~~~
                                                                                                esp_tls_stack_err
In file included from ../main/app_main.c:30:
../main/app_main.c:100:83: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:329:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:296:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../main/app_main.c:100:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
             ^~~~~~~~
../main/app_main.c:100:83: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:330:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:296:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../main/app_main.c:100:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
             ^~~~~~~~
../main/app_main.c:100:83: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:331:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:296:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../main/app_main.c:100:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
             ^~~~~~~~
../main/app_main.c:100:83: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:332:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_VERBOSE )   { esp_log_write(ESP_LOG_VERBOSE,    tag, LOG_FORMAT(V, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:296:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../main/app_main.c:100:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
             ^~~~~~~~
../main/app_main.c:100:83: error: 'esp_mqtt_error_codes_t' {aka 'struct esp_mqtt_error_codes'} has no member named 'esp_transport_sock_errno'; did you mean 'esp_tls_stack_err'?
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:333:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Users/admin/esp/esp-idf/components/log/include/esp_log.h:296:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../main/app_main.c:100:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));
             ^~~~~~~~
```

david-cermak, 2021-03-29:
@larisqueiroz Could you please share which version of IDF and `esp-mqtt` you're using? (i.e. the version which worked and the one with doesn't)

My guess is that your mqtt sub-module is out of sync. could you please check if
```
git submodule update --init --recursive
```
fixes the issue?

Dee-Fuse, 2021-04-08:
I got the same error(s) I'm using the current v4.2 Tag. 
The esp-mqtt version I use is the one bundled with esp-idf.

david-cermak, 2021-04-09:
@Dee-Fuse Could you please tell us more about your build process? CMake vs. Make,  config options, any change to the example?
The tag `v4.2` refers to c40f2590bf759ff60ef122afa79b4ec04e7633d2 (this version https://github.com/espressif/esp-idf/tree/v4.2) which contains no such macros as seen above in the error logs (e.g. `MQTT_ERROR_TYPE_TCP_TRANSPORT` ), also this version refers to this esp-mqtt: https://github.com/espressif/esp-mqtt/commit/01594bf (which again doesn't define these macros)

Are you trying to build a `v4.3` example with `v4.2` IDF? It's not always easy to achieve this forward compatibility, so the examples might not work with earlier releases (should work with future releases, though)

david-cermak, 2021-05-21:
Hi all,

We will close this, as it is believed that the issue is caused by building more recent examples with older IDFs, which is not supported. (Of course backward compatibility between minor releases is, older examples should work with newer IDFs)

Please reopen if this wasn't the case.

closed 2021-05-21 by Alvin1Zhang