espressif/esp-idf #6780: Interrupt wdt timeout after updating to ESP-IDF v4.2

leofabri, 2021-03-25:
## Environment

- Module or chip used:  ESP32-WROVER-IB
- IDF version: v4.2
- Build System:  idf.py
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- environment type: PowerShell
- Using an IDE?: VScode
- Power Supply:  USB & Battery

## Problem Description

After the update to ESP-IDF 4.2 (coming from v4.1.1), a watchdog is triggered on every boot. However, it was not happening before.

Normally, I leave both the RTC and Interrupt watchdogs enabled, like this:

```
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_ESP_INT_WDT=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=800
CONFIG_ESP_INT_WDT_CHECK_CPU1=y
CONFIG_ESP_TASK_WDT=y
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=5
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
```

But then, this happens:

```
^other logs, not relevant (the app doesn't finish loading)... ^
D (1774) efuse: coding scheme 0
D (1774) event: runninets Jun  8 2016 00:22:57     <--- (notice how the execution stops here. The watchdog gets triggered 9s after this)

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

*reboots*
```

I tried to check if there are tasks that are taking too long to return, but I couldn't find any. This actually seems to happen even before my app is able to register any task. I suppose the problem might not be related to my code, especially when considering that this wasn't happening with the old version.

Out of ideas, in order to diagnose what's going on, I played a little with the _sdkconfig_ file and performed the following:

### 1. If I disable `CONFIG_ESP_INT_WDT_CHECK_CPU1`

I thought about this because it looks like that a panic from both cores (like above) isn't diagnosed. Instead, I kept the watchdog enabled on just the core 0 (which fails as well).

Result:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008cb78  PS      : 0x00060034  A0      : 0x8008d893  A1      : 0x3ffb0750
0x4008cb78: compare_and_set_native at C:/development/v4.2/esp-idf/components/soc/include/soc/compare_set.h:28
 (inlined by) spinlock_acquire at C:/development/v4.2/esp-idf/components/soc/include/soc/spinlock.h:97
 (inlined by) vPortCPUAcquireMutex at C:/development/v4.2/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at C:/development/v4.2/esp-idf/components/freertos/xtensa/port.c:394

A2      : 0x3ffb11ec  A3      : 0x0000cdcd  A4      : 0x00060023  A5      : 0x00060021
A6      : 0x0000abab  A7      : 0xb33fffff  A8      : 0x0000abab  A9      : 0x3ffb4b4c
A10     : 0x3ffb4b44  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00060e20
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe
Core  0 was running in ISR context:
EPC1    : 0x401cf657  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008cb78
0x401cf657: uart_hal_write_txfifo at C:/development/v4.2/esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x4008cb78: compare_and_set_native at C:/development/v4.2/esp-idf/components/soc/include/soc/compare_set.h:28
 (inlined by) spinlock_acquire at C:/development/v4.2/esp-idf/components/soc/include/soc/spinlock.h:97
 (inlined by) vPortCPUAcquireMutex at C:/development/v4.2/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at C:/development/v4.2/esp-idf/components/freertos/xtensa/port.c:394

Backtrace:0x4008cb75:0x3ffb0750 0x4008d890:0x3ffb0780 0x4008caef:0x3ffb07a0 0x4008cdb1:0x3ffb07c0 0x40085329:0x3ffb07d0 0x4010be99:0x3ffbc6f0 0x4010cc12:0x3ffbc710 0x400d3fd9:0x3ffbc730 0x4000bd83:0x3ffbc750 0x4000117d:0x3ffbc770 0x400592fe:0x3ffbc790 0x4005937a:0x3ffbc7b0 0x40058bbf:0x3ffbc7d0 0x401b92bb:0x3ffbc800 0x401be89f:0x3ffbc830 0x401bea21:0x3ffbcb40 0x401cf569:0x3ffbcb70 0x40090a19:0x3ffbcba0 0x401cec0b:0x3ffbcbf0 0x4008ca1d:0x3ffbcc10
0x4008cb75: spinlock_acquire at C:/development/v4.2/esp-idf/components/soc/include/soc/spinlock.h:81
 (inlined by) vPortCPUAcquireMutex at C:/development/v4.2/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at C:/development/v4.2/esp-idf/components/freertos/xtensa/port.c:394

0x4008d890: xTaskIncrementTick at C:/development/v4.2/esp-idf/components/freertos/tasks.c:2491

0x4008caef: xPortSysTickHandler at C:/development/v4.2/esp-idf/components/freertos/xtensa/port.c:300

0x4008cdb1: _frxt_timer_int at C:/development/v4.2/esp-idf/components/freertos/xtensa/portasm.S:328

0x40085329: _xt_lowint1 at C:/development/v4.2/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4010be99: uart_tx_char at C:/development/v4.2/esp-idf/components/vfs/vfs_uart.c:164 (discriminator 1)

0x4010cc12: uart_write at C:/development/v4.2/esp-idf/components/vfs/vfs_uart.c:221

0x400d3fd9: esp_vfs_write at C:/development/v4.2/esp-idf/components/vfs/vfs.c:420 (discriminator 4)

0x401b92bb: __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:403

0x401be89f: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)

0x401bea21: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x401cf569: esp_log_writev at C:/development/v4.2/esp-idf/components/log/log.c:189

0x40090a19: esp_log_write at C:/development/v4.2/esp-idf/components/log/log.c:199

0x401cec0b: esp_event_loop_run_task at C:/development/v4.2/esp-idf/components/esp_event/esp_event.c:112

0x4008ca1d: vPortTaskWrapper at C:/development/v4.2/esp-idf/components/freertos/xtensa/port.c:143
```
That, unfortunately, neither does look very clear to me or explain clearly where the problem could be.

### 2. If I disable `CONFIG_ESP_INT_WDT`
The app just works as expected. Everything is loaded, even the WiFi+MQTT stuff seems to work. I doubt it would if a task was stuck somewhere.

The problem, however, is that I can't win an app without Interrupt watchdogs. This isn't a solution.

### Expected Behavior

The app should just run like it was doing before.

### Actual Behavior

It's boot looping. The watchdogs are being triggered every time.

### Steps to reproduce

I was unable to find a way to reproduce the issue and shrink the application due to its complexity. 
I'm not allowed to share the entire application.

## Addtional details

I tried to generate a coredump but with all the watchdogs enabled, although nothing is generated.

Any idea? Thanks in advance

Alvin1Zhang, 2021-03-26:
Thanks for reporting and sorry for the inconvenience, we will look into.

projectgus, 2021-03-28:
Hi @leofabri,

Thanks for the detailed report. Could you attach your full sdkconfig file, please?

leofabri, 2021-03-30:
> Hi @leofabri,
> 
> Thanks for the detailed report. Could you attach your full sdkconfig file, please?

Sure, here is it: **[sdkconfig](https://gist.github.com/leofabri/5e2b02af1727728efab2e78d78fd1617)**

leofabri, 2021-05-14:
Any update about this? I'd like to investigate, but I don't have a very clear about what I'm looking at.

projectgus, 2021-05-21:
Hi @leofabri,

Sorry for going quiet on you for so long.

The sdkconfig file itself seems fine, I don't see any configuration items there that are likely to cause a problem.

Given that everything works fine with INT WDT off, the most likely explanation is that something is disabling or generating a lot of interrupts for a relatively long time (800ms). That doesn't necessarily mean a task is stuck - the INT WDT will timeout if interrupts are being processed non-stop, even though nothing is necessarily stuck it's just a big queue of interrupts to handle. With PSRAM, it's possible for certain operations to run relatively slowly especially during initialization when a lot of new code is being loaded so the flash cache (which is shared with PSRAM) has a high miss rate.

To rule this out, set the CONFIG_ESP_INT_WDT_TIMEOUT_MS to something higher than the default 800ms - maybe 2000ms, for example, and then if the firmware starts to work then you can try reducing it from there. Do you notice any significant pause during startup, either with this config or when the INT WDT is disabled?

If this isn't the cause then I assume it must be something related to the particular application code. Can you give any clues about what the app does during initialization? If you selectively disable initialization for parts of the application, can you zero in one element that causes it for fail?

Also, do you have any backtrace of INT WDT failing when configured on Core 1 only (similarly to the backtrace for Core 0, only)?

Thanks,

Angus

leofabri, 2021-05-23:
Hi @projectgus,
I tried to set CONFIG_ESP_INT_WDT_TIMEOUT_MS to 2000 (and also to 5000) but the bootloop is still occurring.

> Also, do you have any backtrace of INT WDT failing when configured on Core 1 only (similarly to the backtrace for Core 0, only)?

For some reason, I was unable to backtrace the execution like I was doing before. Maybe something did change with version v.4.2.1. 

The only thing I can see now is:

```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7064
load:0x40078000,len:12176
ho 0 tail 12 room 4
load:0x40080400,len:4500
0x40080400: _init at ??:?

entry 0x400806ec
I (183) boot: ESP-IDF v4.2.1-dirty 2nd stage bootloader
I (183) boot: compile time 09:16:22
I (183) boot: chip revision: 1
I (188) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (195) boot.esp32: SPI Speed      : 80MHz
I (200) boot.esp32: SPI Mode       : DIO
I (204) boot.esp32: SPI Flash Size : 16MB
W (209) boot.esp32: PRO CPU has been reset by WDT.
W (214) boot.esp32: WDT reset info: PRO CPU PC=0x4008c9f1
0x4008c9f1: panic_handler at C:/development/esp-idf/components/esp_system/port/panic_handler.c:490 (discriminator 2)

W (220) boot.esp32: WDT reset info: APP CPU PC=0x4008f826
0x4008f826: vListInsert at C:/development/esp-idf/components/freertos/list.c:189

I (226) boot: Enabling RNG early entropy source...
```

What I'm going to try now is to disable some parts of the applicationt o idetify the problem. I'll get back to you as soon as I discover something.

projectgus, 2021-07-09:
Hi @leofabri,

Did you have any luck isolating this in a way that we can try to reproduce it?

Alvin1Zhang, 2021-07-30:
Thanks for reporting, will close now, feel free to reopen with more updates.

closed 2021-07-30 by Alvin1Zhang