espressif/esp-idf #6782: ETHERNET_EVENT_DISCONNECTED occurs when using Ethernet and SPI Master

NSTJunHara, 2021-03-26:
## Environment

- Development Kit:      ESP32-Ethernet-Kit V1.2
- Module or chip used:  ESP32-WROVER-B
- IDF version : v4.4-dev-4-g73db14240  (v4.1 v4.2 release too)
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- environment type: PowerShell.
- Using an IDE?: Yes (ESP-IDF VS Code Extension)
- Power Supply:         USB

## Problem Description

We are developing a device that reads sensor data via SPI2 and transfers it via TCP.
There was no problem with long-term operation on Wi-Fi.
On Ethernet, the ETHERNET_EVENT_DISCONNECTED event is called back in a few hours.

The problem occurs with the ESP32-Ethernet-Kit V1.2 board.
(At first, we confirmed it with our custom board using LAN8720 for PHY.)

I attach a project that adds SPI behavior based on the iperf example.
[iperfspi.zip](https://github.com/espressif/esp-idf/files/6209593/iperfspi.zip)

### Steps to reproduce
1. I am using a static IP address for cross cable. Set `192.168.4.104` on PC.
2. Run command `iperf -s`  on PC (iperf-2.0.9-win64)
3. Run command `iperf -c 192.168.4.104 -t 36000 -i 60` on the ESP32-Ethernet-Kit
4. Wait a few hours 

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6964
load:0x40078000,len:14336
ho 0 tail 12 room 4
load:0x40080400,len:3672
entry 0x40080678
I (29) boot: ESP-IDF v4.4-dev-4-g73db14240-dirty 2nd stage bootloader
I (29) boot: compile time 09:11:09
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 81 00110000 00100000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=156c0h ( 87744) map
I (151) esp_image: segment 1: paddr=000256e8 vaddr=3ffb0000 size=02e14h ( 11796) load
I (156) esp_image: segment 2: paddr=00028504 vaddr=40080000 size=07b14h ( 31508) load
I (170) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=49278h (299640) map
I (276) esp_image: segment 4: paddr=000792a0 vaddr=40087b14 size=08008h ( 32776) load
I (298) boot: Loaded app from partition at offset 0x10000
I (298) boot: Disabling RNG early entropy source...
I (309) cpu_start: Pro cpu up.
I (309) cpu_start: Starting app cpu, entry point is 0x4008140c
I (0) cpu_start: App cpu up.
[... 153 lines cut ...]
6480-6540 sec       13.80 Mbits/sec
6540-6600 sec       13.39 Mbits/sec
6600-6660 sec       13.63 Mbits/sec
6660-6720 sec       13.31 Mbits/sec
6720-6780 sec       13.63 Mbits/sec
6780-6840 sec       13.52 Mbits/sec
6840-6900 sec       14.01 Mbits/sec
6900-6960 sec       14.18 Mbits/sec
6960-7020 sec       14.13 Mbits/sec
7020-7080 sec       13.22 Mbits/sec
7080-7140 sec       13.05 Mbits/sec
7140-7200 sec       13.86 Mbits/sec
7200-7260 sec       14.12 Mbits/sec
7260-7320 sec       14.11 Mbits/sec
7320-7380 sec       14.13 Mbits/sec
7380-7440 sec       14.05 Mbits/sec
7440-7500 sec       13.57 Mbits/sec
E (7600611) event_handler: Ethernet Link Down
W (7600611) iperf: tcp client send error, error code: 113, reason: Software caused connection abort
I (7600621) iperf: iperf exit
7500-7560 sec       13.14 Mbits/sec
   0-36000 sec       0.37 Mbits/sec
I (7604611) event_handler: Ethernet Link Up
I (7604611) esp_netif_handlers: eth ip: 192.168.4.101, mask: 255.255.255.0, gw: 1.0.0.0
```

suda-morris, 2021-03-26:
Will the link down happen even without the SPI code?

NSTJunHara, 2021-03-26:
In my own project, I commented out only `spi_device_polling_transmit()` and it didn't happen.
It didn't happen in the iperf example either.
I will comment out `spi_device_polling_transmit()` in the attached project and try again.

NSTJunHara, 2021-03-29:
Even if I commented out `spi_device_polling_transmit()`, a link down occurred.
When I checked it again with iperf example, it linked down at 13800 sec.
I think the use of SPI tends to shorten the link down time.

NSTJunHara, 2021-03-31:
I had the problem on some Windows PCs, but when I tried it on a Mac, it didn't occur for more than 10 hours. I want to check other conditions, so close it once.

closed 2021-03-31 by NSTJunHara

VaishnaviPadiyappanavar, 2023-02-12:
what is the method used to solve this problem

NSTJunHara, 2023-02-13:
We are currently using a custom board that is significantly different from the evaluation board.
Although I have not received any reports of the issues from the circuit personnel, I am unsure of how to resolve them.

VaishnaviPadiyappanavar, 2023-02-13:
Ok Thankyou.