espressif/esp-idf #6785: esp32 sd card in SPI Mode

ChinmayDixit1997, 2021-03-26:
Hi,
For sd card in SPI mode i get following error everytime:
I (330) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (340) example: Initializing SD card
I (340) example: Using SPI peripheral
I (350) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (380) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (380) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (380) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (390) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
**E (400) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.**

Also I connected pull ups with external 5V supply (10K). 
Please guide.
Thanks.

Alvin1Zhang, 2021-03-26:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ChinmayDixit1997, 2021-03-26:
Thanks for the prompt reply. I am using esp32 example code for sd card.

[sd_card 1.zip](https://github.com/espressif/esp-idf/files/6211350/sd_card.1.zip)

ChinmayDixit1997, 2021-03-27:
Any Update Further?

igrr, 2021-03-28:
Hi @ChinmayDixit1997, to troubleshoot this i'd recommend doing the following:
1) Enable verbose log level in menuconfig (under component config, log output). Copy the log output you get (starting from the GPIO initialization) and attach it to this issue. From there we can see if the card is responding to at least some commands, on not responding at all.
2) Use a logic analyzer (even a low speed one will do) to check the data exchanged between the ESP32 and the SD card.

Also I wouldn't recommend connecting the pullups to 5V, since both the SD card and the ESP32 are 3.3V devices. Pulling the I/Os to 5V even through 10k resistors may latch or damage the I/O pins.

ChinmayDixit1997, 2021-03-30:
Thanks @igrr for the reply. here is my log output below after connecting the pull ups to 3.3 V (pulls up are connected to MOSI and MISO):
I (340) example: Initializing SD card
I (340) example: Using SPI peripheral
I (350) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (360) sdspi_transaction: cmd=52, R1 response: command not supported
I (390) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (390) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (390) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (400) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (410) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

Thanks in Advance.

chegewara, 2021-03-30:
Redefinition of pin mode?
```
I (350) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (400) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

ChinmayDixit1997, 2021-03-30:
No @chegewara . I am using Example code which is provide by Espressif. 

    ESP_LOGI(TAG, "Using SPI peripheral");
    
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
    slot_config.gpio_miso = PIN_NUM_MISO;
    slot_config.gpio_mosi = PIN_NUM_MOSI;
    slot_config.gpio_sck  = PIN_NUM_CLK;
    slot_config.gpio_cs   = PIN_NUM_CS;

 esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };

sdmmc_card_t* card;
    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                "If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }

above is the code which I am using.
Thanks.

igrr, 2021-03-30:
@ChinmayDixit1997 Can you switch to Verbose log level, please? The setting is in menuconfig (under component config, log output).

ChinmayDixit1997, 2021-03-30:
Yes sure.. 

I (0) cpu_start: App cpu up.
V (270) memory_layout: reserved range is 0x3f40b28c - 0x3f40b2b4
D (276) memory_layout: Checking 7 reserved memory ranges:
D (281) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (288) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb2b00
D (294) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (301) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (307) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (314) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

D (320) memory_layout: Reserved memory range 0x40080000 - 0x4008ac60
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

D (326) memory_layout: Building list of available memory regions:
V (333) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (339) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (349) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
V (355) memory_layout: Examining memory region 0x3ffb0000 - 0x3ffb8000
V (362) memory_layout: Start of region 0x3ffb0000 - 0x3ffb8000 overlaps reserved 0x3ffb0000 - 0x3ffb2b00
D (371) memory_layout: Available memory region 0x3ffb2b00 - 0x3ffb8000
V (378) memory_layout: Examining memory region 0x3ffb8000 - 0x3ffc0000
D (385) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
V (391) memory_layout: Examining memory region 0x3ffc0000 - 0x3ffc2000
D (398) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
V (404) memory_layout: Examining memory region 0x3ffc2000 - 0x3ffc4000
D (411) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
V (418) memory_layout: Examining memory region 0x3ffc4000 - 0x3ffc6000
D (424) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
V (431) memory_layout: Examining memory region 0x3ffc6000 - 0x3ffc8000
D (437) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
V (444) memory_layout: Examining memory region 0x3ffc8000 - 0x3ffca000
D (451) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
V (457) memory_layout: Examining memory region 0x3ffca000 - 0x3ffcc000
D (464) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
V (470) memory_layout: Examining memory region 0x3ffcc000 - 0x3ffce000
D (477) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
V (484) memory_layout: Examining memory region 0x3ffce000 - 0x3ffd0000
D (490) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
V (497) memory_layout: Examining memory region 0x3ffd0000 - 0x3ffd2000
D (503) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
V (510) memory_layout: Examining memory region 0x3ffd2000 - 0x3ffd4000
D (517) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
V (523) memory_layout: Examining memory region 0x3ffd4000 - 0x3ffd6000
D (530) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
V (536) memory_layout: Examining memory region 0x3ffd6000 - 0x3ffd8000
D (543) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
V (550) memory_layout: Examining memory region 0x3ffd8000 - 0x3ffda000
D (556) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
V (563) memory_layout: Examining memory region 0x3ffda000 - 0x3ffdc000
D (569) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
V (576) memory_layout: Examining memory region 0x3ffdc000 - 0x3ffde000
D (583) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
V (589) memory_layout: Examining memory region 0x3ffde000 - 0x3ffe0000
D (596) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
V (602) memory_layout: Examining memory region 0x3ffe0000 - 0x3ffe4000
V (609) memory_layout: Start of region 0x3ffe0000 - 0x3ffe4000 overlaps reserved 0x3ffe0000 - 0x3ffe0440
V (618) memory_layout: End of region 0x3ffe0440 - 0x3ffe4000 overlaps reserved 0x3ffe3f20 - 0x3ffe4350
D (628) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
V (634) memory_layout: Examining memory region 0x3ffe4000 - 0x3ffe8000
V (641) memory_layout: Start of region 0x3ffe4000 - 0x3ffe8000 overlaps reserved 0x3ffe3f20 - 0x3ffe4350
D (651) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
V (657) memory_layout: Examining memory region 0x3ffe8000 - 0x3fff0000
D (664) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
V (670) memory_layout: Examining memory region 0x3fff0000 - 0x3fff8000
D (677) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
V (684) memory_layout: Examining memory region 0x3fff8000 - 0x3fffc000
D (690) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
[... 466 lines cut ...]
D (1290) spi_master: SPI2: New device added to CS0, effective clock: 400kHz
I (1290) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
V (1300) sdmmc_cmd: sending cmd slot=1 op=52 arg=80000c08 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (1310) sdspi_host: sdspi_host_start_command: slot=1, CMD52, arg=0x80000c08 flags=0x40, data=0x0, data_size=0 crc=0x4f
D (1320) spi_master: SPI device changed from 3 to 0
D (1330) sdspi_host: sdspi_host_start_command: cmd=52 error=0x107
D (1340) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x107
V (1340) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x107
[... cut ...]
I (1440) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1450) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1460) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
V (1470) intr_alloc: esp_intr_free: Disabling int, killing handler
I (1470) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (1480) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

chegewara, 2021-03-30:
@igrr I can confirm regression in master branch.
Tested the same example with master branch 1067b28707e527f177752741e3aa08b5dc64a4d7 and b0150615dff529662772a60dcb57d5b559f480e2
This commit works:
b0150615dff529662772a60dcb57d5b559f480e2
this commit does not works
1067b28707e527f177752741e3aa08b5dc64a4d7

In my case logs are:
```
I (43) boot: ESP-IDF v4.4-dev-479-g1067b2870-dirty 2nd stage bootloader
I (44) boot: compile time 07:33:16
I (44) boot: chip revision: 0
I (47) boot.esp32s2: SPI Speed      : 80MHz
I (52) boot.esp32s2: SPI Mode       : DIO
I (57) boot.esp32s2: SPI Flash Size : 2MB
I (62) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (71) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0a2a4h ( 41636) map
I (121) esp_image: segment 1: paddr=0001a2cc vaddr=3ffbe4b0 size=027fch ( 10236) load
I (124) esp_image: segment 2: paddr=0001cad0 vaddr=40022000 size=03548h ( 13640) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=22130h (139568) map
I (165) esp_image: segment 4: paddr=00042158 vaddr=40025548 size=08f64h ( 36708) load
I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (192) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (192) cpu_start: Pro cpu up.
I (246) cpu_start: Pro cpu start user code
I (246) cpu_start: cpu freq: 160000000
I (246) cpu_start: Application information:
I (249) cpu_start: Project name:     sd_card
I (254) cpu_start: App version:      v4.4-dev-479-g1067b2870-dirty
I (261) cpu_start: Compile time:     Mar 30 2021 07:33:11
I (267) cpu_start: ELF file SHA256:  a4b2678bcc409d48...
I (273) cpu_start: ESP-IDF:          v4.4-dev-479-g1067b2870-dirty
I (280) heap_init: Initializing. RAM available for dynamic allocation:
I (287) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (293) heap_init: At 3FFC14C8 len 0003AB38 (234 KiB): DRAM
I (299) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (306) spi_flash: detected chip: generic
I (310) spi_flash: flash io: dio
W (314) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (328) cpu_start: Starting scheduler on PRO CPU.
I (333) example: Initializing SD card
I (333) example: Using SPI peripheral
I (333) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (343) sdspi_transaction: cmd=52, R1 response: command CRC error
E (353) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
E (363) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (363) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (373) example: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
```

chegewara, 2021-03-30:
And working sd card logs:
```
I (43) boot: ESP-IDF v4.4-dev-479-g1067b2870-dirty 2nd stage bootloader
I (44) boot: compile time 07:44:42
I (44) boot: chip revision: 0
I (47) boot.esp32s2: SPI Speed      : 80MHz
I (52) boot.esp32s2: SPI Mode       : DIO
I (57) boot.esp32s2: SPI Flash Size : 2MB
I (62) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (71) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0a2a4h ( 41636) map
I (121) esp_image: segment 1: paddr=0001a2cc vaddr=3ffbe4b0 size=027fch ( 10236) load
I (124) esp_image: segment 2: paddr=0001cad0 vaddr=40022000 size=03548h ( 13640) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=22130h (139568) map
I (165) esp_image: segment 4: paddr=00042158 vaddr=40025548 size=08f64h ( 36708) load
I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (192) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (192) cpu_start: Pro cpu up.
I (246) cpu_start: Pro cpu start user code
I (246) cpu_start: cpu freq: 160000000
I (246) cpu_start: Application information:
I (249) cpu_start: Project name:     sd_card
I (254) cpu_start: App version:      v4.4-dev-479-g1067b2870-dirty
I (261) cpu_start: Compile time:     Mar 30 2021 07:44:37
I (267) cpu_start: ELF file SHA256:  3e8d2ec74e5eed7f...
I (273) cpu_start: ESP-IDF:          v4.4-dev-479-g1067b2870-dirty
I (280) heap_init: Initializing. RAM available for dynamic allocation:
I (287) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (293) heap_init: At 3FFC14C8 len 0003AB38 (234 KiB): DRAM
I (299) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (306) spi_flash: detected chip: generic
I (310) spi_flash: flash io: dio
W (314) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (328) cpu_start: Starting scheduler on PRO CPU.
I (333) example: Initializing SD card
I (333) example: Using SPI peripheral
I (333) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (383) sdspi_transaction: cmd=5, R1 response: command not supported
Name: SC16G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 15193MB
I (423) example: Opening file
I (433) example: File written
I (443) example: Renaming file
I (453) example: Reading file
I (453) example: Read from file: 'Hello SC16G!'
I (463) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (463) example: Card unmounted
I (463) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (473) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (483) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
```

PS In logs for both examples esp-idf git commit is the same, because i am using example from master (the same folder), just i am doing source to different esp-idf API and toolchain

igrr, 2021-03-30:
@chegewara the issue you are seeing (ESP_ERR_INVALID_CRC during sdmmc_io_reset) is different from what @ChinmayDixit1997 is seeing (ESP_ERR_TIMEOUT during sdmmc_send_cmd_go_idle_state). The reason for your issue I am not sure about, but I'd recommend opening a separate issue, please. I see the corresponding CI test cases (init SD card in SPI mode) are passing for the master branch locally. If you see the issue doesn't appear on another commit, could you try doing `git bisect` to find the commit where the issue starts happening?

@ChinmayDixit1997 Since the card doesn't respond to the CMD0 or any of the previous commands, most likely this is a wiring error. Please check again that the SD card is functional, and check that the connections between the ESP32 and the SD card are correct. If the card is wired up correctly, it will always respond to CMD0 sent in SPI mode.

ChinmayDixit1997, 2021-03-30:
Thanks @igrr . I will check and get back to you.

chegewara, 2021-03-30:
@igrr Actually code is working in master, but due to very small issue with USB cable (lose connection) sometimes i am getting different errors (performed more tests now), not just CRC. Here is another error:
```
I (328) cpu_start: Starting scheduler on PRO CPU.
I (333) example: Initializing SD card
I (333) example: Using SPI peripheral
I (333) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (343) sdspi_transaction: cmd=52, R1 response: unexpected value 0x20
E (353) sdmmc_io: sdmmc_io_reset: unexpected return: 0x108
E (363) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (363) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (373) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.
```

I am not trying to hijack issue, just trying to help to solve @ChinmayDixit1997 issue

ChinmayDixit1997, 2021-03-30:
Hey @igrr , Just checked the wiring between ESP32 and SD Card Module.
 Its fine but the same error came. I also checked SD Card with my PC. Its working fine.
Thanks.

igrr, 2021-03-30:
@chegewara 

> sdspi_transaction: cmd=52, R1 response: unexpected value 0x20
> sdmmc_io_reset: unexpected return: 0x108

This looks interesting but again different both from @ChinmayDixit1997's issue and #6686 you have linked.

@ChinmayDixit1997 Can you connect a logic analyzer to the pins, on the SD card side, and see what is being sent? I don't have other suggestions for troubleshooting this, sorry. This step of SD initialization works for me locally and also works for @chegewara (although there is a failure later in the initialization process).

pgelinas345, 2021-04-16:
I have a similar issue. When I was using IDF v4.1, I never had problem with the SD SPI. Since I updated to v4.2, I have not been able to get it working like before. I had to put a pull up on MISO and MOSI and drop the max frequency to 10 MHz. Has there been any change in the gpio and/or spi driver from 4.1 to 4.2?

adokitkat, 2021-07-26:
Hi @ChinmayDixit1997. 
After investigating on this issue and testing `sdspi` example I have found no problem to be present with newest version of ESP-IDF. I've also tested your example code which didn't work however after some changes for it to be compliant with latest API (ie. to fixing it for IDF version v4.3) I got it to work. Is it OK for you to use the lastest release of ESP-IDF?

adokitkat, 2021-08-04:
If your problem is still not resolved it could mean you are having a hardware-related issue. Then possible solutions are (thanks @nopnop2002 for the summary):

- The connection is incorrect.
- The SD card is not formatted in FAT32.
- The connection cable is too long.
- There is no pull-up resistor.
- SPI frequency may not be compatible - try to change (lower) it.

I am closing this issue. If you are still having problems feel free to reopen it.

closed 2021-08-04 by adokitkat

engineertrooper, 2022-10-26:
Hi, @adokitkat
 
I am late, but I want to add a possibile error to your list:

- using wrong pin id for MISO, MOSI or CS.   

If the sd card example is used, it will not show an error like that as a message. For me that error occured because I was importing source code to a diffrent device.