espressif/esp-idf #6787: Unable to re-initialize netif after unitialized

persan666, 2021-03-26:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta2-2-g9a2d25191
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [YES] Eclipse
- Power Supply:         [Battery]

## Problem Description

My program have to switch between Wifi and PPPos.
If I first run PPPos and then switch to Wifi and then back to PPPos the netif layer crashes. 

// Wifi initialization:
----------------------------------------------------------------------------
```cpp
esp_netif_t* netif;

netif = esp_netif_create_default_wifi_sta();
ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
	                                                        ESP_EVENT_ANY_ID,
	                                                        &wifi_event_handler,
	                                                        NULL,
	                                                        &instance_any_id));
ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
	                                                        IP_EVENT_STA_GOT_IP,
	                                                        &wifi_event_handler,
	                                                        NULL,
	                                                        &instance_got_ip));
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
esp_wifi_init(&cfg);
esp_wifi_set_storage(WIFI_STORAGE_RAM);
esp_wifi_set_mode(WIFI_MODE_STA);
```
--------------------------------------------------------------------------------
// Wifi un-initialization:
--------------------------------------------------------------------------------
```cpp
ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));

esp_wifi_deinit();
esp_netif_deinit();
esp_netif_destroy(netif);
```
--------------------------------------------------------------------------------
// PPPos initialization
--------------------------------------------------------------------------------
```cpp
void *	modem_netif_adapter;

esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL);
esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, NULL);

esp_modem_dte_config_t config = ESP_MODEM_DTE_DEFAULT_CONFIG();
dte = esp_modem_dte_init(&config);

if(dte == NULL){
   goto err;
}

esp_modem_set_event_handler(dte, modem_event_handler, ESP_EVENT_ANY_ID, NULL);

esp_netif_config_t cfg = ESP_NETIF_DEFAULT_PPP();
netif = esp_netif_new(&cfg);
assert(netif);

esp_netif_ppp_config_t ppp_config = {
            .ppp_error_event_enabled = true,
            .ppp_phase_event_enabled = true
};

esp_netif_ppp_set_params(netif, &ppp_config);

modem_netif_adapter = esp_modem_netif_setup(dte);
esp_modem_netif_set_default_handlers(modem_netif_adapter, netif);
```
--------------------------------------------------------------------------------------
// PPPos un-initialization
--------------------------------------------------------------------------------------
```cpp
esp_event_handler_unregister(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event);
esp_event_handler_unregister(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed);

esp_modem_netif_clear_default_handlers(modem_netif_adapter);
esp_modem_netif_teardown(modem_netif_adapter);

esp_netif_destroy(netif);
```
----------------------------------------------------------------------------------------
```
IP-addresses are marked xxx

## Debug Logs
I (56409) modem: REC: CONNECT 150000000
ppp_connect[4]: holdoff=0
ppp phase changed[4]: phase=3
pppos_connect: unit 4: connecting
ppp_start[4]
ppp phase changed[4]: phase=6
pppos_send_config[4]: out_accm=FF FF FF FF
ppp_send_config[4]
pppos_recv_config[4]: in_accm=FF FF FF FF
ppp_recv_config[4]
ppp: auth protocols: PAP=0 CHAP=0 CHAP_MD5=0
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xee66d2b8> <pcomp> <accomp>]
pppos_write[4]: len=24
ppp_start[4]: finished
pppos_input[4]: got 2 bytes
pppos_input[4]: got 38 bytes
pppos_input[4]: got 16 bytes
rcvd [LCP ConfReq id=0x0 <asyncmap 0x0> <auth chap MD5> <magic 0xadfe005e> <pcomp> <accomp>]
No auth is possible
sent [LCP ConfRej id=0x0 <auth chap MD5>]
pppos_write[4]: len=13
pppos_input[4]: got 43 bytes
rcvd [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0xee66d2b8> <pcomp> <accomp>]
pppos_input[4]: got 1 bytes
pppos_input[4]: got 31 bytes
pppos_input[4]: got 14 bytes
rcvd [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xadfe005e> <pcomp> <accomp>]
sent [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0xadfe005e> <pcomp> <accomp>]
pppos_write[4]: len=24
netif_set_mtu[4]: mtu=1500
pppos_send_config[4]: out_accm=0 0 0 0
ppp_send_config[4]
pppos_recv_config[4]: in_accm=0 0 0 0
ppp_recv_config[4]
ppp phase changed[4]: phase=7
ppp phase changed[4]: phase=9
I (57025) PPP phase event: Authenticate
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[4]: len=32
I (57032) PPP phase event: Network
pppos_input[4]: got 2 bytes
pppos_input[4]: got 14 bytes
pppos_input[4]: got 33 bytes
rcvd [LCP DiscReq id=0x2 magic=0xadfe005e]
rcvd [IPCP ConfReq id=0x0]
sent [IPCP ConfNak id=0x0 <addr 0.0.0.0>]
pppos_write[4]: len=14
rcvd [IPCP ConfRej id=0x1 <compress VJ 0f 01>]
sent [IPCP ConfReq id=0x2 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[4]: len=26
pppos_input[4]: got 38 bytes
rcvd [IPCP ConfReq id=0x1]
sent [IPCP ConfAck id=0x1]
pppos_write[4]: len=8
rcvd [IPCP ConfNak id=0x2 <addr xxx.xxx.xxx.xxx> <ms-dns1 xxx.xxx.xxx.xxx> <ms-dns2 xxx.xxx.xxx.xxx>]
sent [IPCP ConfReq id=0x3 <addr xxx.xxx.xxx.xxx> <ms-dns1 xxx.xxx.xxx.xxx> <ms-dns2 xxx.xxx.xxx.xxx>]
pppos_write[4]: len=26
pppos_input[4]: got 28 bytes
rcvd [IPCP ConfAck id=0x3 <addr xxx.xxx.xxx.xxx> <ms-dns1 xxx.xxx.xxx.xxx> <ms-dns2 xxx.xxx.xxx.xxx>]
Could not determine remote IP address: defaulting to xxx.xxx.xxx.xxx
sifvjcomp[4]: VJ compress enable=0 slot=0 max slot=0
sifup[4]: err_code=0
I (57972) esp-netif_lwip-ppp: Connected
I (57977) esp-netif_lwip-ppp: Name Server1: xxx.xxx.xxx.xxx
I (57983) esp-netif_lwip-ppp: Name Server2: xxx.xxx.xxx.xxx
local  IP address xxx.xxx.xxx.xxx
remote IP address xxx.xxx.xxx.xxx
primary   DNS address xxx.xxx.xxx.xxx
secondary DNS address xxx.xxx.xxx.xxx
ppp phase changed[4]: phase=10
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4013f9c6  PS      : 0x00060b30  A0      : 0x8012f0b0  A1      : 0x3ffda0c0
0x4013f9c6: netif_set_addr at C:/source/dev/esp/esp-idf/components/lwip/lwip/src/core/netif.c:698

A2      : 0x00000000  A3      : 0x0000000c  A4      : 0x00000010  A5      : 0x00000014
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x800967ad  A9      : 0x3ffda0a0
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x3ffda144
A14     : 0x3ffdb520  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4013f9c3:0x3ffda0c0 0x4012f0ad:0x3ffda100 0x4012f03a:0x3ffda120 0x4013e5cd:0x3ffda140
0x4013f9c3: netif_set_addr at C:/source/dev/esp/esp-idf/components/lwip/lwip/src/core/netif.c:695

0x4012f0ad: esp_netif_up_api at C:/source/dev/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1215

0x4012f03a: esp_netif_api_cb at C:/source/dev/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:107

0x4013e5cd: tcpip_thread_handle_msg at C:/source/dev/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:208
 (inlined by) tcpip_thread at C:/source/dev/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

Laiany, 2021-03-29:
I'm experiencing the same problem here.

Alvin1Zhang, 2021-03-30:
Thanks for reporting and sorry for the inconvenience, we will look into.

david-cermak, 2021-03-30:
Hi @persan666 

Thanks for the report! Indeed, there's a bug in the modem network glue layer deinitialization (It fails to unregister IP related events, so whenever we receive  `IP_EVENT_PPP_GOT_IP` for the second time, we get this event with the already destroyed `esp_netif` pointer.

This patch should fix the issue:
[esp_modem-Fixed-modem-netif-deinit-to-clear-IP-event.patch.txt](https://github.com/espressif/esp-idf/files/6230891/esp_modem-Fixed-modem-netif-deinit-to-clear-IP-event.patch.txt)

persan666, 2021-03-31:
Great! Thanks for the respond!

Laiany, 2021-03-31:
It worked for me, thanks!

closed 2021-04-15 by espressif-bot