espressif/esp-idf #6788: v3.3: updated espsecure.py requires '--version' argument which is not provided by all Makefiles

bearsh, 2021-03-26:
## Environment

- Development Kit:      none
- Module or chip used:  NINA-W1
- IDF version (run ``git describe --tags`` to find it): v3.3.5-1-g85c43024c
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-97-gc752ad5
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

The esptool go updated in commit https://github.com/espressif/esp-idf/commit/54ee2f2ebc9c43e45d86779d145ef8abfb274fa9. Now `espsecure.py` requires a `--version` argument. At least the following makefiles call espsecure without that argument:
- `components/esptool_py/Makefile.projbuild` (https://github.com/espressif/esp-idf/blob/release/v3.3/components/esptool_py/Makefile.projbuild#L60)
- `components/partition_table/Makefile.projbuild` (https://github.com/espressif/esp-idf/blob/release/v3.3/components/partition_table/Makefile.projbuild#L53)

### Expected Behavior

successful build after calling make

### Actual Behavior

calling espsecure without a `--version` switch causes the build to fail

### Possible fix

```diff
diff --git a/components/esptool_py/Makefile.projbuild b/components/esptool_py/Makefile.projbuild
index 707b1a3cf..d51b64396 100644
--- a/components/esptool_py/Makefile.projbuild
+++ b/components/esptool_py/Makefile.projbuild
@@ -57,7 +57,7 @@ ifndef IS_BOOTLOADER_BUILD
 APP_BIN_UNSIGNED := $(APP_BIN:.bin=-unsigned.bin)
 
 $(APP_BIN): $(APP_BIN_UNSIGNED) $(SECURE_BOOT_SIGNING_KEY) $(SDKCONFIG_MAKEFILE)
-       $(ESPSECUREPY) sign_data --keyfile $(SECURE_BOOT_SIGNING_KEY) -o $@ $<
+       $(ESPSECUREPY) sign_data --version 1 --keyfile $(SECURE_BOOT_SIGNING_KEY) -o $@ $<
 endif
 endif
 # non-secure boot (or bootloader), both these files are the same
diff --git a/components/partition_table/Makefile.projbuild b/components/partition_table/Makefile.projbuild
index 57aeca2b9..11ee24a54 100644
--- a/components/partition_table/Makefile.projbuild
+++ b/components/partition_table/Makefile.projbuild
@@ -50,7 +50,7 @@ ifdef CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES
 PARTITION_TABLE_BIN_UNSIGNED := $(PARTITION_TABLE_BIN:.bin=-unsigned.bin)
 # add an extra signing step for secure partition table
 $(PARTITION_TABLE_BIN): $(PARTITION_TABLE_BIN_UNSIGNED) $(SDKCONFIG_MAKEFILE) $(SECURE_BOOT_SIGNING_KEY)
-       $(ESPSECUREPY) sign_data --keyfile $(SECURE_BOOT_SIGNING_KEY) -o $@ $<
+       $(ESPSECUREPY) sign_data --version 1 --keyfile $(SECURE_BOOT_SIGNING_KEY) -o $@ $<
 else
 # secure bootloader disabled, both files are the same
 PARTITION_TABLE_BIN_UNSIGNED := $(PARTITION_TABLE_BIN)
```

vonnieda, 2021-03-26:
I'm also experiencing this after upgrading from v3.3.3 to v3.3.5.

bearsh, 2021-03-26:
possibly affected file:
```
$ ag --files-with-matches ESPSECUREPY 
components/partition_table/CMakeLists.txt
components/partition_table/Makefile.projbuild
components/bootloader/Makefile.projbuild
components/bootloader/subproject/CMakeLists.txt
components/bootloader_support/CMakeLists.txt
components/bootloader_support/Makefile.projbuild
components/bootloader_support/component.mk
components/esptool_py/project_include.cmake
components/esptool_py/Makefile.projbuild
```

Alvin1Zhang, 2021-03-30:
Thanks for reporting and sorry for the inconvenience, we will look into.

dobairoland, 2021-05-03:
This was fixed in https://github.com/espressif/esp-idf/commit/f845643608ee0c47af23b1ee412fd341766fcdb8. This commit is not yet available on Github. This issue will be closed manually after the commit gets synchronized into Github.

bearsh, 2021-05-15:
@dobairoland please push, don't see a reason to hold this back as it's a **major regression for all v3.3 users**

dobairoland, 2021-05-15:
I'm sorry for the inconvenience but I cannot do anything about it. It will be pushed automatically after some regression tests pass.

You can apply the following fix until then:
```sh
cd $IDF_PATH/components/esptool_py/esptool
git checkout 5b8c2f1b02d0e4b9bac0756b3ead66e8beea428a
cd $IDF_PATH
```

dobairoland, 2021-05-31:
I believe that the fix is now in the v3.3 release branch and will be added to the v3.3.6 release. We are sorry for the delay and the inconvenience.

closed 2021-05-31 by dobairoland

vonnieda, 2021-08-09:
@dobairoland @Alvin1Zhang  Can you tell us when v3.3.6 will be released? We are now 4 months past when v3.3.5 was released broken, and nearly 3 months past when the bug was supposedly fixed.

dobairoland, 2021-08-10:
I'm sorry for the inconvenience but I'm not aware of any dates set for the release of v3.3.6. Please see my previous comments regarding manual workaround and published fix in the release/v3.3 branch.