espressif/esp-idf #6789: Flash operations panicking when SPI is initialized on APP cpu

KaeLL, 2021-03-26:
## Environment

- Module or chip used: ESP32
- IDF version: 1067b2870

## Problem Description

Title. I call `spi_bus_initialize` using `esp_ipc_call_blocking` to have the SPI interrupt trigger on core 1, and flash operations such as reading a coredump partition or writing to (unencrypted, if that matters) NVS fail as soon as they start.

## Debug Logs
Optimizations are on, but the result is identical except for the inlining with them off.
<details>
    <summary>Backtrace when about to read coredump from flash</summary>

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40095a33  PS      : 0x00060033  A0      : 0x80094854  A1      : 0x3ffe1e30  
0x40095a33: uxListRemove at /esp-idf/components/freertos/list.c:175

A2      : 0x0000001c  A3      : 0xb33fffff  A4      : 0x00060020  A5      : 0x00060023  
A6      : 0x3ffe2018  A7      : 0x3ffe201c  A8      : 0x3ffc50ec  A9      : 0x00000000  
A10     : 0x0000cdcd  A11     : 0xb33fffff  A12     : 0x00060023  A13     : 0x00060023  
A14     : 0x3ffe201c  A15     : 0x0000ffff  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x40086e14  LEND    : 0x40086e1c  LCOUNT  : 0x00000000  
0x40086e14: spi_flash_mmap at /esp-idf/components/spi_flash/flash_mmap.c:149 (discriminator 3)

0x40086e1c: spi_flash_mmap at /esp-idf/components/spi_flash/flash_mmap.c:151

Backtrace:0x40095a30:0x3ffe1e30 0x40094851:0x3ffe1e50 0x40093600:0x3ffe1e70 0x40173513:0x3ffe1eb0 0x40173611:0x3ffe1ee0 0x40086a01:0x3ffe1f00 0x40086be0:0x3ffe1f20 0x40086e28:0x3ffe1f80 0x40118b45:0x3ffe1fb0 0x40119556:0x3ffe1fd0
0x40095a30: uxListRemove at /esp-idf/components/freertos/list.c:172

0x40094851: xTaskRemoveFromEventList at /esp-idf/components/freertos/tasks.c:3493

0x40093600: xQueueSemaphoreTake at /esp-idf/components/freertos/queue.c:1495

0x40173513: esp_ipc_call_and_wait at /esp-idf/components/esp_ipc/ipc.c:119

0x40173611: esp_ipc_call_and_wait at /esp-idf/components/esp_ipc/ipc.c:107
 (inlined by) esp_ipc_call at /esp-idf/components/esp_ipc/ipc.c:140

0x40086a01: spi_flash_disable_interrupts_caches_and_other_cpu at /esp-idf/components/spi_flash/cache_utils.c:155

0x40086be0: spi_flash_mmap_pages at /esp-idf/components/spi_flash/flash_mmap.c:179
 (inlined by) spi_flash_mmap_pages at /esp-idf/components/spi_flash/flash_mmap.c:156

0x40086e28: spi_flash_mmap at /esp-idf/components/spi_flash/flash_mmap.c:151

0x40118b45: esp_partition_mmap at /esp-idf/components/spi_flash/partition.c:491
```

</details>
<details>
    <summary>Backtrace when about to write to NVS</summary>

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40095a33  PS      : 0x00060d33  A0      : 0x80094854  A1      : 0x3ffdb690  
0x40095a33: uxListRemove at /esp-idf/components/freertos/list.c:175

A2      : 0x0000001c  A3      : 0xb33fffff  A4      : 0x00000000  A5      : 0x00060d23  
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x3ffc50f0  A9      : 0x00000004  
A10     : 0x0000abab  A11     : 0xb33fffff  A12     : 0x00060d23  A13     : 0x00060d23  
A14     : 0x00000200  A15     : 0x0000001c  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x40112904  LEND    : 0x4011291f  LCOUNT  : 0x00000009  
0x40112904: nvs::HashList::find(unsigned int, nvs::Item const&) at /esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp:112

0x4011291f: nvs::HashList::find(unsigned int, nvs::Item const&) at /esp-idf/components/nvs_flash/src/intrusive_list.h:64
 (inlined by) nvs::HashList::find(unsigned int, nvs::Item const&) at /esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp:109

Backtrace:0x40095a30:0x3ffdb690 0x40094851:0x3ffdb6b0 0x40093241:0x3ffdb6d0 0x4017234e:0x3ffdb710 0x4017240d:0x3ffdb740 0x40086a01:0x3ffdb760 0x40086696:0x3ffdb780 0x4008f68d:0x3ffdb7a0 0x40086369:0x3ffdb7c0 0x40117831:0x3ffdb800 0x401114ca:0x3ffdb820 0x40112fb2:0x3ffdb840 0x4011219a:0x3ffdb8b0 0x401123b1:0x3ffdb930 0x401113d9:0x3ffdb950 0x40110faa:0x3ffdb970 0x40168d65:0x3ffdb9a0 0x401ac523:0x3ffdb9e0 0x400fdc89:0x3ffdba10 0x401a93e9:0x3ffdba40 0x401aa185:0x3ffdba60
0x40095a30: uxListRemove at /esp-idf/components/freertos/list.c:172

0x40094851: xTaskRemoveFromEventList at /esp-idf/components/freertos/tasks.c:3493

0x40093241: xQueueGenericSend at /esp-idf/components/freertos/queue.c:827

0x4017234e: esp_ipc_call_and_wait at /esp-idf/components/esp_ipc/ipc.c:128

0x4017240d: esp_ipc_call_and_wait at /esp-idf/components/esp_ipc/ipc.c:107
 (inlined by) esp_ipc_call at /esp-idf/components/esp_ipc/ipc.c:140

0x40086a01: spi_flash_disable_interrupts_caches_and_other_cpu at /esp-idf/components/spi_flash/cache_utils.c:155

0x40086696: cache_disable at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:78
 (inlined by) spi1_start at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:107

0x4008f68d: spiflash_start_default at /esp-idf/components/spi_flash/esp_flash_api.c:110

0x40086369: esp_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:652

0x40117831: esp_partition_read at /esp-idf/components/spi_flash/partition.c:347

0x401114ca: nvs::NVSPartition::read(unsigned int, void*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_partition.cpp:45
 (inlined by) nvs::NVSPartition::read(unsigned int, void*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_partition.cpp:39

0x40112fb2: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /esp-idf/components/nvs_flash/src/nvs_page.cpp:797
 (inlined by) ?? at /esp-idf/components/nvs_flash/src/nvs_page.cpp:841
 (inlined by) nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /esp-idf/components/nvs_flash/src/nvs_page.cpp:804

0x4011219a: _ZN3nvs7Storage9writeItemEhNS_8ItemTypeEPKcPKvj$part$39 at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:160
 (inlined by) ?? at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:285

0x401123b1: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:274

0x401113d9: nvs::NVSHandleSimple::set_string(char const*, char const*) at /esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:44
 (inlined by) nvs::NVSHandleSimple::set_string(char const*, char const*) at /esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:39

0x40110faa: nvs_set_str at /esp-idf/components/nvs_flash/src/nvs_api.cpp:426
```

</details>

ginkgm, 2021-08-17:
Hi @KaeLL ,
Thanks for reporting this issue, and the detailed log.

I guess it's because there are some issues that make spi_flash_disable_interrupts_caches_and_other_cpu  not compatible with IPC call.

Do you usually use IPC to install drivers on other cores? Though this is the first time I heard about it, it may help us a lot in locating related issues in the spi flash.

KaeLL, 2021-08-17:
> Do you usually use IPC to install drivers on other cores?

No. It was just a test meant to have the SPI ISR run on the APP core.

closed 2022-06-06 by KaeLL