espressif/esp-idf #6791: idf.py menuconfig blink then exit

someRandomNetizen, 2021-03-27:
I'm trying to configure my crystal to 26 MHz from the default 40 MHz using idf.py. When i hit enter, the table appear for about 1 sec, then exit. here's a picture of that: 
[image]

dobairoland, 2021-03-27:
@someRandomNetizen Please give more information about your system (ESP-IDF version, Vscode plugin version) and steps you are doing (where are you trying to open menuconfig, eg. which terminal, or built-in vscode menuconfig?).

someRandomNetizen, 2021-03-27:
Sorry, my bad, kinda moody back then. Let me edit that real quick

dobairoland, 2021-03-28:
> i'm trying to flash my project with P32 chip, first, the terminal said it couldn't find idf.py, so i add it to the invironment variable, then this happen.

I think that if it wasn't available then the Vscode setup wasn't successful or complete. Please follow the guide [here](https://github.com/espressif/vscode-esp-idf-extension) for setting up the vscode plugin.

What is the P32 chip? Do you mean ESP32?

someRandomNetizen, 2021-03-28:
@dobairoland yeah, it's ESP32. and thanks buddy, my problem's solved, well not really. Techinically i stil haven't been able to do the idf.py menuconfig yet. But i figure s.t else.
for anyone of you who have the same problem:
f1 -> esp-idf: sdk configuration editor -> esp specific -> change the box below main XTAL frequency to autodectect. that should do it

dobairoland, 2021-03-28:
@someRandomNetizen Are you able to run `idf.py --version` or `idf.py build`?

The "sdk configuration editor" is able to edit the configs and achieve the same outcome as `idf.py menuconfig`.

dobairoland, 2021-04-24:
I think the issue has been solved and I'm closing this for lack of activity.

closed 2021-04-24 by dobairoland