espressif/esp-idf #6793: abort() in esp_wifi/src/lib_printf.c:47 on connect to wifi access point when battery voltage has been read

william-ferguson-au, 2021-03-28:
## Environment

- Development Kit:      None
- Module or chip used:  ESP32S2-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.3-beta1-133-g9a2d25191
  Which becomes v4.3-beta1-134-g6f5e13dcb after applying patch for https://github.com/espressif/esp-idf/issues/6633
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes but NA. All compilation and deployment done from command line
- Power Supply:         Battery

## Problem Description

App aborts deterministically after it connects to a Wifi access point, but only if we have been reading the battery or USB voltage.

This was originally noted in https://www.esp32.com/viewtopic.php?f=2&t=19724
But perhaps not enough info was provided as I have seen no progress.

### Expected Behavior

It shouldn’t abort. The app should connect to the Wifi AP and be able to read battery levels.

### Actual Behavior

Aborts deterministically with in locks.c:138 which originates from esp_wifi/src/lib_printf.c:47
None of my code is invoked.

### Steps to reproduce

1. Idf.py -p COM<port> flash monitor

### Code to reproduce this issue

I have tried to reduce the code as much as possible.
https://gist.github.com/william-ferguson-au/dcc7ed7ffe405c0d233cf2b1669e8ec3

Essentially commenting out the esp_wifi_connect() on line 143 or the adc1_get_raw() on line 252 negate the crash (but make the app useless).

## Debug Logs

```
I (1007) main: Disabled Charging
I (1017) wifi:state: assoc -> run (10)
I (1027) wifi:connected with Wylas-Timing-2.4GHz, aid = 1, channel 9, BW20, bssid = 68:ff:7b:33:96:3b
I (1027) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (1037) wifi:pm start, type: 1

I (1037) main: wifi_event_handler eventBase='WIFI_EVENT' EventId=4
W (1047) wifi:<ba-add>idx:0 (ifx:0, 68:ff:7b:33:96:3b), tid:0, ssn:0, winSize:64
I (1017) main: VBAT - Raw: 5503, Raw Voltage: 568mV, Real Voltage: 3760mV
I (1057) main: Enabling Charging
I (1057) gpio: GPIO[15]| InputEn: 0| O
abort() was called at PC 0x400245b6 on core 0
0x400245b6: lock_acquire_generic at C:/Dev/espressif/esp-idf/components/newlib/locks.c:138

Backtrace:0x40029a4f:0x3ffd6460 0x4002a2dd:0x3ffd6480 0x40030e26:0x3ffd64a0 0x400245b6:0x3ffd6510 0x40024701:0x3ffd6540 0x400e8506:0x3ffd6560 0x400eb565:0x3ffd6870 0x400f27dd:0x3ffd68a0 0x4002f8b5:0x3ffd68d0 0x400daf2d:0x3ffd6920 0x400daf
57:0x3ffd69a0 0x400269f3:0x3ffd69f0 0x40026acd:0x3ffd6a50 0x40028554:0x3ffd6a70 0x400c72a7:0x3ffd6a90 0x40032805:0x3ffd6ad0 0x4002d021:0x3ffd6b00
0x40029a4f: panic_abort at C:/Dev/espressif/esp-idf/components/esp_system/panic.c:356
0x4002a2dd: esp_system_abort at C:/Dev/espressif/esp-idf/components/esp_system/system_api.c:112
0x40030e26: abort at C:/Dev/espressif/esp-idf/components/newlib/abort.c:46
0x400245b6: lock_acquire_generic at C:/Dev/espressif/esp-idf/components/newlib/locks.c:138
0x40024701: _lock_acquire_recursive at C:/Dev/espressif/esp-idf/components/newlib/locks.c:166
0x400e8506: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)
0x400eb565: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)
0x400f27dd: esp_log_writev at C:/Dev/espressif/esp-idf/components/log/log.c:189
0x4002f8b5: esp_log_write at C:/Dev/espressif/esp-idf/components/log/log.c:199
0x400daf2d: lib_printf at C:/Dev/espressif/esp-idf/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)
0x400daf57: phy_printf at C:/Dev/espressif/esp-idf/components/esp_wifi/src/lib_printf.c:57
0x400269f3: pll_vol_cal at ??:?
0x40026acd: wifi_track_pll_cap at ??:?
0x40028554: tx_pwctrl_background at ??:?
0x400c72a7: lmacProcessTxComplete at ??:?
0x40032805: ppTask at ??:?
0x4002d021: vPortTaskWrapper at C:/Dev/espressif/esp-idf/components/freertos/port/xtensa/port.c:168

```

## Other items if possible

- sdkconfig attached to GIST

william-ferguson-au, 2021-03-28:
If I change lib_printf.c:47 from ESP_LOGI() to ESP_EARLY_LOGI() as suggested in https://www.esp32.com/viewtopic.php?f=2&t=19724&p=73943#p73943

Then I gets lots of instances of

```
I (9440) phy: pll_cap_ext 10
```

and then the station disconnects from the wifi AP. It can't seem to maintain a connection.

newHeiko, 2021-03-29:
Might be related to #6682 ?

Heiko

william-ferguson-au, 2021-03-29:
@newHeiko looks exactly like #6682

Alvin1Zhang, 2021-03-30:
Thanks for reporting, we had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

william-ferguson-au, 2021-03-30:
Thanks Alvin. Looking forward to that fix being rolled into 4.3

vadaliya, 2021-03-31:
I also facing the same issue.

Alvin1Zhang, 2021-04-02:
Thanks for reporting, fix on master is available [here](https://github.com/espressif/esp-idf/commit/99aaebb5e49fac71a1d0fd176d471dc7c1e4e3fa), we are back porting this fix to release/4.3, will update once the fix on release/4.3 is available. Thanks.

Alvin1Zhang, 2021-06-08:
Thanks for reporting and patience in waiting for the fix, the fix on release/4.3 is available https://github.com/espressif/esp-idf/commit/b75a5fd03a4c12c7cc2ac3c7503de4a3a70accd5, feel free to reopen.

closed 2021-06-08 by Alvin1Zhang