espressif/esp-idf #6795: TFLite Hello world error in EP32-C3

Robertou2, 2021-03-28:
-
## Environment

- Development Kit:     ESP32-C3-DevKitM 1
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  ESP32-C3
- IDF version v4.4-dev-4-g73db14240
- Build System:        idf.py
- Compiler version riscv32-esp-elf-gcc.exe (crosstool-NG 1.24.0.123_64eb9ff) 8.4.0
- Operating System:     windows
- (Windows only) environment type: [PowerShell
- Using an IDE?: No, idf.py
- Power Supply:        USB

## Problem Description
Build TFLite Hello world  in EP32-C3 return 
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:37:41: error: 'round' is not a member of 'std'
 DECLARE_STD_GLOBAL_SWITCH1(TfLiteRound, round);

//Detailed problem description goes here.
When I try to build  this example https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/hello_world#deploy-to-esp32
in an ESP-32-C3, I get an error
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:37:41: error: 'round' is not a member of 'std'
 DECLARE_STD_GLOBAL_SWITCH1(TfLiteRound, round);

If the target is ESP-32 there isn´t any problem
### Expected Behavior
Get a binary file
### Actual Behavior
Get an error
### Steps to reproduce

I follow the instructions to deploy from the example. For ESP32-c3  I use  idf.py set-target esp32-c3
For ESP32 idf.py set-target esp32

After idf.py build

// If possible, attach a picture of your setup/wiring here.

https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/hello_world#deploy-to-esp32

## Debug Logs

FAILED: esp-idf/tfmicro/CMakeFiles/__idf_tfmicro.dir/tensorflow/lite/micro/kernels/activations.cc.obj
ccache C:\Users\roberto\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe   -Iconfig -I../components/tfmicro -I../components/tfmicro/third_party/gemmlowp -I../components/tfmicro/third_party/flatbuffers/include -I../components/tfmicro/third_party/ruy -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/port/riscv/include -I../../../../components/esp_hw_support/include -I../../../../components/esp_hw_support/port/esp32c3/. -I../../../../components/esp_hw_support/port/esp32c3/private_include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/include -I../../../../components/soc/esp32c3/. -I../../../../components/soc/esp32c3/include -I../../../../components/hal/esp32c3/include -I../../../../components/hal/include -I../../../../components/esp_rom/include -I../../../../components/esp_rom/esp32c3 -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/esp32c3/include -I../../../../components/driver/include -I../../../../components/driver/esp32c3/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32c3/include -I../../../../components/riscv/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/esp_pm/include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32c3/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -march=rv32imc   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-4-g73db14240-dirty\" -DESP_PLATFORM -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-type-limits -DTF_LITE_STATIC_MEMORY -DNDEBUG -O3 -Wno-nonnull -std=c++11 -Wno-return-type -Wno-strict-aliasing -MD -MT esp-idf/tfmicro/CMakeFiles/__idf_tfmicro.dir/tensorflow/lite/micro/kernels/activations.cc.obj -MF esp-idf\tfmicro\CMakeFiles\__idf_tfmicro.dir\tensorflow\lite\micro\kernels\activations.cc.obj.d -o esp-idf/tfmicro/CMakeFiles/__idf_tfmicro.dir/tensorflow/lite/micro/kernels/activations.cc.obj -c ../components/tfmicro/tensorflow/lite/micro/kernels/activations.cc
In file included from ../components/tfmicro/tensorflow/lite/kernels/internal/common.h:27,
                 from ../components/tfmicro/tensorflow/lite/micro/kernels/activations.cc:18:
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h: In function 'T tflite::TfLiteRound(T)':
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:37:41: error: 'round' is not a member of 'std'
 DECLARE_STD_GLOBAL_SWITCH1(TfLiteRound, round);
                                         ^~~~~
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:34:39: note: in definition of macro 'DECLARE_STD_GLOBAL_SWITCH1'
     return TF_LITE_GLOBAL_STD_PREFIX::std_name(x);    \
                                       ^~~~~~~~
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:37:41: note: suggested alternative: 'count'
 DECLARE_STD_GLOBAL_SWITCH1(TfLiteRound, round);
                                         ^~~~~
../components/tfmicro/tensorflow/lite/kernels/internal/cppmath.h:34:39: note: in definition of macro 'DECLARE_STD_GLOBAL_SWITCH1'
     return TF_LITE_GLOBAL_STD_PREFIX::std_name(x);    \
                                       ^~~~~~~~
[913/1028] Performing build step for 'bootloader'
[1/80] Generating project_elf_src_esp32c3.c
[2/80] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/mpu_hal.c.obj
[3/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32c3/gdma_periph.c.obj
[4/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32c3/soc_memory_layout.c.obj
[5/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32c3/rmt_periph.c.obj
[6/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/soc_include_legacy_warn.c.obj
[7/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[8/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/memory_layout_utils.c.obj
[9/80] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32c3/timer_periph.c.obj
[... 57 lines cut ...]
[67/80] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log.c.obj
[68/80] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
[69/80] Linking C static library esp-idf\log\liblog.a
[70/80] Linking C static library esp-idf\esp_rom\libesp_rom.a
[71/80] Linking C static library esp-idf\esp_hw_support\libesp_hw_support.a
[72/80] Linking C static library esp-idf\efuse\libefuse.a
[73/80] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[74/80] Linking C static library esp-idf\spi_flash\libspi_flash.a
[75/80] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[76/80] Linking C static library esp-idf\soc\libsoc.a
[77/80] Linking C static library esp-idf\hal\libhal.a
[78/80] Linking C static library esp-idf\main\libmain.a
[79/80] Linking C executable bootloader.elf
[80/80] Generating binary image from built executable
esptool.py v3.1-dev
Merged 1 ELF section
Generated C:/Users/roberto/Desktop/esp-idf/examples/hello_world/esp-idf/build/bootloader/bootloader.bin
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

projectgus, 2021-03-30:
Hi @Robertou2,

Thanks for reporting this and providing a clear description of the error. Although we don't support third party projects, I can reproduce this issue using a simple C++11 test case so I've renamed the title.

```cpp
#include <cmath>

int round_test(void)
{
    return std::round(3.33);
}
```

This seems like it will need to be fixed by updating the libstdc++ that is bundled with the ESP32-C3 toolchain. Will update once this is available.

Robertou2, 2021-03-30:
Hi @projectgus,
thanks for your fast answer.

Robertou2, 2021-03-30:
I have found a workaround using round() from cmath without std.

projectgus, 2021-04-11:
Thanks for letting us know about the workaround, @Robertou2 . This is still a bug in ESP-IDF (math headers should "just work"), so happy to keep it until we solve the root cause.

closed 2021-06-28 by espressif-bot