espressif/esp-idf #6796: Regression: ESP32 Revision 0/1 Chips BLE light sleep with 32KHz crystal not working in V4.3 (is working in 4.0)

lucazader, 2021-03-29:
## Environment

- Development Kit:      ESP32-Wrover-Kit v1 && Custom PICO-D4 module
- Module or chip used:  ESP32-WROOM-32 && ESP32-WROOM-32E && ESP32-PICO-D4
`Chip is ESP32-D0WDQ6 (revision 0)`
`Chip is ESP32-PICO-D4 (revision 1)`
`Chip is ESP32-D0WD-V3 (revision 3)`
- IDF version: v4.0.2-442-g41efdb0b34 && v4.3-beta2-2-g9a2d251912
- Build System: CMake / idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply:         USB && external 3.3V

## Problem Description
*Using the gatt_server example code*
With IDF V4.3 and the new differentiation between 32KHz crystal extra current method 1 and Method 2.
On the version (ESP32-PICO-D4 (revision 1)) of the esp32 that we are using neither of these methods results in stable operation of BLE when using light sleep power saving.
I will see the chip advertise, however if I attempt to connect, the connection error's out with a `GATT 133 (0x85)` error on the phone and `lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` showing in the device logs

However with IDF V4.0 using the 32KHz crystal with the "extra current" option is quite reliable for advertising and connections  etc.

With newer revision of chips is this not an issue as expected, however we do have a number of products out in the wild with the older chip revisions, so are obviously interested in supporting them with the newer ow power operation.

### Expected Behavior
V4.3 BLE with light sleep mode should work

### Actual Behavior
V4.3 BLE with light sleep mode fails.
V4.0 BLE with light sleep mode does work

### Steps to reproduce
In Menuconfig:
Enable 32KHz crystal With extra current settings 
Enable Free Rtos tickless idle
Enable bluetooth power saving `ORIG` mode
32KHz crystal as clock source
Clock accuracy changed to 251-500ppm (also tested tighter accuracy, but this was even more un-reliable)

### Code to reproduce this issue
added to example code to allow low power light sleep
```
esp_pm_config_esp32_t cfg = {
      .max_freq_mhz = 80,
      .min_freq_mhz = 40,
      .light_sleep_enable = true,
  };
    esp_pm_configure(&cfg);```

wolfstrassen, 2021-04-09:
I can confirm this happens also with an ESP32-PICO-D4 and IDF v4.3

negativekelvin, 2021-04-10:
It looks like `#define XTAL_32K_DAC_VAL    1` is the only thing that changed

https://github.com/espressif/esp-idf/commit/a48b5246ccd1c54f0abbc05e0759fcd5363700de

wolfstrassen, 2021-04-27:
> It looks like `#define XTAL_32K_DAC_VAL 1` is the only thing that changed
> 
> [a48b524](https://github.com/espressif/esp-idf/commit/a48b5246ccd1c54f0abbc05e0759fcd5363700de)

tried this, kind of work unless the timer used for gettimeofday is RTC instead of RTC and high-resolution timer. Can't connect if using RTC

xiewenxiang, 2021-06-09:
@lucazader 

Please enable `ESP32_TIME_SYSCALL_USE_RTC`

```c
Name: ESP32_TIME_SYSCALL_USE_RTC
Prompt: RTC
Type: bool
Value: y

Direct dependencies (=y):
     <choice ESP32_TIME_SYSCALL>

Kconfig definition, with parent deps. propagated to 'depends on'
================================================================

At /home/xiewenxiang/esp/esp-idf/components/esp32/Kconfig:528
Included via /home/xiewenxiang/esp/esp-idf/Kconfig:382 -> kconfigs.in:8
Menu path: (Top) -> Component config -> ESP32-specific -> Timers used for gettimeofday function

  config ESP32_TIME_SYSCALL_USE_RTC
        bool "RTC" if IDF_TARGET_ESP32(=y)
        select ESP_TIME_FUNCS_USE_RTC_TIMER(=y)
        depends on <choice ESP32_TIME_SYSCALL>

```

Alvin1Zhang, 2021-07-01:
@lucazader Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

therealergo, 2021-07-13:
I'm working on a different issue (my revision 3 chips seem to need this option for 32.768k to oscillate reliably) but I may have found the issue that's preventing the CONFIG_ESP32_RTC_EXT_CRYST_ADDIT_CURRENT_V2 option from working properly after [a48b524](https://github.com/espressif/esp-idf/commit/a48b5246ccd1c54f0abbc05e0759fcd5363700de)

[Line 1367 of sleep_modes.c](https://github.com/espressif/esp-idf/blob/5b5e46971abb223d7034508eb11ca4e40bea7845/components/esp_system/sleep_modes.c#L1367) prevents the RTC_PERIPH power domain from being powered down in sleep mode. It only checks the  CONFIG_ESP32_RTC_EXT_CRYST_ADDIT_CURRENT macro, and not the CONFIG_ESP32_RTC_EXT_CRYST_ADDIT_CURRENT_V2 macro. This means that unless RTC_PERIPH is manually kept on with:
`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);`
The touch peripheral will no longer provide power to the crystal in sleep and the crystal oscillation may die.

AxelLin, 2022-06-18:
> @lucazader Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

https://github.com/espressif/esp-idf/issues/6621 seems the same or similar issue.