espressif/esp-idf #6797: AP disconnects every few minutes

0xFEEDC0DE64, 2021-03-29:
I am using a 4.4-dev based esp-idf with an ESP32 and I experience WiFi disconnects to a laptop from the AccessPoint every ~5 minutes or so. NetworkManager on the laptop just pops up with "Connection lost" and a few seconds later "[AccessPointName] connected".

When that happens, I see a DHCP log message from esp-idf DHCP server. I was recording the log output over night and it seems to have a regular interval:

```
        I (24402279) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24462719) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24523079) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24583599) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24644089) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24704929) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24765189) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24825509) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24886399) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (24946529) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (25007499) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (25068479) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (25128939) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (25189249) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
        I (25227259) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182
```

What components of the esp-idf could I look into to find out why this happens? I have logging set to information in menuconfig, I have some websocket and httpd traffic which gets interupted every time this happens, which results in a reconnect a loss of live data for some seconds.

xueyunfei998, 2021-03-31:
hi @0xFEEDC0DE64 
 Do you use a laptop to do hot spots, and then use ESP to connect the computer problems ?

0xFEEDC0DE64, 2021-03-31:
ESP is connected to my home router (STA), laptop is connected with LAN cable (ethernet) to the same router, laptop is also connected to the ESP's AccessPoint

xueyunfei998, 2021-04-01:
hi @0xFEEDC0DE64 

- You set ESP32 to SoftAP + sta mode，
- Esp32 sta connection router,
- The laptop is connected to the router via Ethernet
- Laptop connected to ESP hotspot
Through the above actions，
WiFi disconnects to a laptop from the AccessPoint every ~5 minutes .

This should be the reproduction process right？

xueyunfei998, 2021-04-01:
- esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.128.250.182.

The SoftAP segment should be 192.168.4.1，Why does log show 10.128.250.182.

0xFEEDC0DE64, 2021-04-01:
We changed the ip address range as we had customers with a 192.168.4.0/24 network at home and when STA and AP have the same ip range, I have problems with internet access and UDP broadcasts on the right interface (to find other equipment).

I use `tcpip_adapter_dhcps_option` and `esp_netif_dhcps_start` to change the default range

peterpoetzi, 2021-04-01:
I also have the problem that my ESP32 always loose the connection to the STA network every 300.00 seconds (exactly 5 minutes) for ~ 2 seconds. Maybe we can find this interval somewhere in the idf, maybe some kind of timeout.

0xFEEDC0DE64, 2021-04-01:
Maybe unrelated, but since I increased the log level from INFO to DEBUG, setting our custom ip range for AP suddenly fails,
and I "fixed" this by calling `esp_netif_dhcps_stop()` multiple times, until `esp_netif_set_ip_info()` doesn't return `ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED` anymore.

I have absolutley no idea why the DHCP is not stopped anymore with DEBUG loglevel and even less an idea why stopping it again helps.

```
        if (const auto err = esp_netif_dhcps_get_status(esp_netif, &status); err == ESP_OK)
            ESP_LOGI(TAG, "esp_netif_dhcps_get_status() succeeded");
        else
        {
            ESP_LOGE(TAG, "esp_netif_dhcps_get_status() failed with %s", esp_err_to_name(err));
            return err;
        }

        for (int i = 0; ; i++)
        {
            if (const auto err = esp_netif_dhcps_stop(esp_netif); err == ESP_OK)
                ESP_LOGI(TAG, "esp_netif_dhcps_stop() succeeded");
            else
            {
                ESP_LOGE(TAG, "esp_netif_dhcps_stop() failed with %s", esp_err_to_name(err));
                if (err != ESP_ERR_ESP_NETIF_DHCP_ALREADY_STOPPED)
                    return err;
            }

            if (const auto err = esp_netif_set_ip_info(esp_netif, &info); err == ESP_OK)
            {
                ESP_LOGI(TAG, "esp_netif_set_ip_info() for AP succeeded");
                break;
            }
            else
            {
                ESP_LOGE(TAG, "esp_netif_set_ip_info() for AP failed with %s", esp_err_to_name(err));
                if (err == ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED && i < 20)
                {
                    vTaskDelay(std::chrono::floor<espcpputils::ticks>(50ms).count());
                    continue;
                }
                return err;
            }
        }

        dhcps_lease_t lease;
        lease.enable = true;
        lease.start_ip.addr = static_cast<uint32_t>(local_ip) + (1 << 24);
        lease.end_ip.addr = static_cast<uint32_t>(local_ip) + (11 << 24);

        if (const auto err = tcpip_adapter_dhcps_option((tcpip_adapter_dhcp_option_mode_t)TCPIP_ADAPTER_OP_SET, (tcpip_adapter_dhcp_option_id_t)REQUESTED_IP_ADDRESS, (void*)&lease, sizeof(dhcps_lease_t));
                err != ESP_OK)
        {
            ESP_LOGE(TAG, "DHCPS Set Lease Failed! %s", esp_err_to_name(err));
            return err;
        }

        if (const auto err = esp_netif_dhcps_start(esp_netif); err != ESP_OK)
        {
            ESP_LOGE(TAG, "DHCPS Start Failed! %s", esp_err_to_name(err));
            return err;
        }
```

```
D (00:00:02.679) WIFI_GOE: go-e WiFi Event: 10 - WIFI_AP_START[0m
D (00:00:02.690) wifi_init_default: wifi_start esp-netif:0x3ffd1b98 event-id12[0m
D (00:00:02.690) esp_netif_lwip: check: remote, if=0x3ffd1b98 fn=0x4015bdf8
[0;33m0x4015bdf8: esp_netif_dhcps_stop_api at /home/feedc0de/mobile-esp8266-firmware/build_hw4/../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1111
[0m
[0m
D (00:00:02.703) wifi_init_default: WIFI mac address: 10 52 1c 89 3b e9[0m
D (00:00:02.710) esp_netif_lwip: esp_netif_dhcps_stop_api esp_netif:0x3ffd1b98[0m
[... 31 lines cut ...]
[0;33m0x4015bdf8: esp_netif_dhcps_stop_api at /home/feedc0de/mobile-esp8266-firmware/build_hw4/../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1111
[0m
[0m
D (00:00:02.919) esp_netif_lwip: esp_netif_dhcps_stop_api esp_netif:0x3ffd1b98[0m
D (00:00:02.922) esp_netif_lwip: dhcp server stop successfully[0m
D (00:00:02.928) esp_netif_lwip: call api in lwip: ret=0x0, give sem[0m
[0;32mI (00:00:02.935) WIFI_GOE: esp_netif_dhcps_stop() succeeded[0m
D (00:00:02.941) esp_netif_lwip: check: remote, if=0x3ffd1b98 fn=0x4015c2f8
[0;33m0x4015c2f8: esp_netif_set_ip_info_api at /home/feedc0de/mobile-esp8266-firmware/build_hw4/../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1342
[0m
[0m
D (00:00:02.948) esp_netif_lwip: esp_netif_set_ip_info_api esp_netif:0x3ffd1b98[0m
netif: GW address of interface ap set to 10.128.250.181
netif_set_ipaddr: netif address being changed
D (00:00:02.964) esp_netif_lwip: call api in lwip: ret=0x0, give sem[0m
[0;32mI (00:00:02.970) WIFI_GOE: esp_netif_set_ip_info() for AP succeeded[0m
D (00:00:02.977) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd1b98[0m
D (00:00:02.984) esp_netif_lwip: check: remote, if=0x3ffd1b98 fn=0x4015bc90
[0;33m0x4015bc90: esp_netif_dhcps_start_api at /home/feedc0de/mobile-esp8266-firmware/build_hw4/../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1074
[0m
[0m
D (00:00:02.991) esp_netif_lwip: esp_netif_dhcps_start_api esp_netif:0x3ffd1b98[0m
D (00:00:02.998) esp_netif_lwip: DHCP server started successfully[0m
D (00:00:03.004) esp_netif_lwip: call api in lwip: ret=0x0, give sem[0m
```

xueyunfei998, 2021-04-02:
hi @0xFEEDC0DE64

- I finally know the problem scene.
- You set SoftAP and router as a network segment.(router: 10.128.250.1 .  softap(192.168.4.1(default)--->10.128.250.181))

- Laptop connected to SoftAP, dropped every 5 minutes.

- I think it may appear in the same network segment, there may be address conflict。

Can you provide a demo to reproduce the problem? I will try to reproduce the problem locally.

0xFEEDC0DE64, 2021-04-18:
Your hint helped me fixing it already, thank you :)

closed 2021-04-18 by 0xFEEDC0DE64