espressif/esp-idf #6798: BLE-MESH Fast Provisioning NVS

MrCurio, 2021-03-30:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.2-47-g2532ddd9f + multiple ble mesh patches
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [Yes, VsCode]
- Power Supply:         [USB]

## Problem Description
Hello, i am developing a full custom application based on Bluetooth Mesh using Fast Provisioning and i had encountered certain issues during development. So far, my networks are correctly built using fast provisioning and everything works as expected. 

However, now i want to implement a new feature, which consists of: once the network has been formed, send a command to all  nodes to reactivate its fast provisioning in order to add new nodes (out of the range of the main provisioner) to the network. This feature would allow me to add nodes, wherever they are,  to the network without having to destroy and restart all the fast provisioning process.

To do so i do not consider sending a SET_FAST_PROV message to every node with a specific new range of addresses. Instead, i would like to use their old Fast Provisioning info and fill in the gaps left in the last process. This way, nodes that already provisioned their maximum, should not reactivate FP (fast provisioning) and node with free spots in their first address range would reactivate FP. 

The problem comes when reactivating FP info in the nodes. 

If no reset occurs in the node, a simply call to the ``` esp_ble_mesh_set_fast_prov_action(FAST_PROV_ACT_ENTER);``` API would take the latest FP info and continue provisioning as desired **blocking overlapping adresses** and filling the addresses firstly assigned until reaching its maximum number of nodes. 

However, if there's a reset all the fast provisioning info except ``` prov_ctx.curr_alloc_addr``` is gone (its all static and not stored in NVS), which means FP info must be set, managed and stored by the user . I think it would be good to store FP info (at least unicast_max and unicast_min) in the NVS via stack to avoid further overlapping issues. Otherwise the user can set any range after reset with no restrictions at all. Below there's an example of what i'm talking about: 

```
/mesh_core/fast_prov.c
u8_t bt_mesh_set_fast_prov_unicast_addr_range(u16_t min, u16_t max)
{
    .....

    if (min <= prov_ctx.fast_prov.unicast_addr_max) { <------------ This line is always false when a reset 
        BT_ERR("Unicast address overlap");
        return 0x03; /* status: address overlaps with current value */
    }

    prov_ctx.fast_prov.unicast_addr_min = min;
    prov_ctx.fast_prov.unicast_addr_max = max;

    prov_ctx.curr_alloc_addr = prov_ctx.fast_prov.unicast_addr_min;
    .....
``` 

I think if no more NVS info wants to be stored, at least the range of addresses should be checked e.g. as follows:
```
u8_t bt_mesh_set_fast_prov_unicast_addr_range(u16_t min, u16_t max)
{
    .....

    if (min < prov_ctx.curr_alloc_addr) { <------------ This line changes
        BT_ERR("Unicast address overlap");
        return 0x03; /* status: address overlaps with current value */
    }

    prov_ctx.fast_prov.unicast_addr_min = min;
    prov_ctx.fast_prov.unicast_addr_max = max;

    prov_ctx.curr_alloc_addr = prov_ctx.fast_prov.unicast_addr_min;
    .....
``` 

In fact, i have a doubt, is any provisioning information besides netKey and appKey and stack information stored in the nodes that use fast provisioning such as number of provisioned nodes? Is the mesh_nodes (provisioner_main.c) array filled and saved while using Fast provisioning? I am debugging and i cannot see that information at stack level. Shouldn't it be stored? 

### Expected Behavior
Much more information store in NVS allowing to continue form the previous point before reset
### Actual Behavior
Missing fast provisioning information after reset
### Steps to reproduce

1. Take Fast provisioning examples and try to reactivate FP after a reset
2.  Do not assign a new range of addresses to provision 

## Debug Logs

```
esp_err_t fp_server_recv_reactivate(esp_ble_mesh_msg_ctx_t *ctx){

    if (srv->prov_node_cnt >= srv->max_node_num)
    {
        ESP_LOGW(TAG, "Unable to reactivate FP. Not enough address!");
        return ESP_OK;
    }

    uint16_t num_nodes_stack = 0;
     const esp_ble_mesh_node_t **entry = esp_ble_mesh_provisioner_get_node_table_entry(); 
     for (int i = 0; i < CONFIG_BLE_MESH_MAX_PROV_NODES; i++) { 
         const esp_ble_mesh_node_t *node = entry[i]; 
         if (node) { 
             num_nodes_stack++;
         } 
     } 
    ESP_LOGI(TAG, "Nodes stack: %d", num_nodes_stack); --------> always zero even if it provisioned a node
    uint16_t number_nodes = esp_ble_mesh_provisioner_get_prov_node_count(); //!!This API does not work after reset
    ESP_LOGI(TAG, "Nodes stack 2: %d", number_nodes); --------> always zero even if it provisioned a node
```

## Other items if possible

# Override some defaults so BT stack is enabled
# by default in this example
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BTDM_MODEM_SLEEP=n
CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE=y
CONFIG_BTDM_BLE_MESH_SCAN_DUPL_EN=y
CONFIG_BT_NIMBLE_ENABLED=y

# Override some defaults of ESP BLE Mesh
CONFIG_BLE_MESH=y
CONFIG_BLE_MESH_FAST_PROV=y
CONFIG_BLE_MESH_MAX_PROV_NODES=6
CONFIG_BLE_MESH_PBA_SAME_TIME=3
CONFIG_BLE_MESH_TX_SEG_MSG_COUNT=10
CONFIG_BLE_MESH_RX_SEG_MSG_COUNT=10
CONFIG_BLE_MESH_CFG_CLI=y
CONFIG_BLE_MESH_NOT_USE_NO_OOB=y

#Enable NVS Storage
CONFIG_BLE_MESH_SETTINGS=y

Campou, 2021-06-04:
Hi @MrCurio 

Sorry for the late reply.

For storing the fast provisioning info in NVS, we could add this operation in the application layer. 

About storing the stack-layer information, e.g., the provisioned node count, because the node just works as a Provisioner **temporarily** in the fast provisioning mode, currently we didn't store any temporary information of the node.

Thanks.

MrCurio, 2021-06-13:
Hi @Campou 
I think there's no need to add the operation from your side. I ended up saving the information on my own along with other relevant configurations in the NVS.

Thanks anyway!

closed 2021-06-13 by MrCurio