espressif/esp-idf #6800: Bluetooth pairing fails for Android phones with MediaTek chipset

rj747, 2021-03-30:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): [v4.1.1 @ 4ebe0d8a2f0fabf577ce03ab9dec976dd3ce3198 ]
- Build System:         [idf.py]
- Compiler version: [(crosstool-NG esp-2020r3) 8.4.0]
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description
The ESP32 Bluetooth cannot pair with certain Android phones that use MediaTek chips that support Bluetooth 5.0. The affected phones include LG Stylo 6, LG Phoenix 5, and Alcatel 3V. Phones/tablets that do not use the MediaTek chipset for Bluetooth will pair with no issues such as the Pixel, Pixel4 (qualcom) and Hytab-plus-10wb1 (xradio). 

When capturing the Bluetooth pairing packets, I noticed that there was LL_REJECT_IND_EXT opcode from the device that was common for all the affected phones which may indicate why the connection was dropped. Also tried ESP-IDF Pre-release v4.3-beta2 with different bluetooth module options but the issues still persists.

### Expected Behavior
ESP32 should pair with the android phones with no issues.

### Actual Behavior
ESP32 attempts to pair but will fail regardless of retries.

### Steps to reproduce

1. Enable ESP and attempt to pair via one of the affected phones. A modified version of the ESP-IDF Gatt Security Server Demo at the link below can be run on the Wrover kit to reproduce.

### Code to reproduce this issue
https://gist.github.com/rj747/3c686faffea20d258875413165afe8ef

## Debug Logs
https://gist.github.com/rj747/3c686faffea20d258875413165afe8ef
```
D (1391) BT_HCI: HCI Enqueue Comamnd opcode=0x2008

D (1401) BT_HCI: Receive packet event_code=0xe

D (1401) BT_HCI: btu_hcif_command_complete_evt

D (1401) BT_HCI: HCI Enqueue Comamnd opcode=0x2009

D (1411) BT_HCI: Receive packet event_code=0xe

D (1411) BT_HCI: btu_hcif_command_complete_evt

D (1421) BT_SMP: static passkey 123456
D (1421) BT_HCI: HCI Enqueue Comamnd opcode=0x2005

D (1431) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (1431) BT_HCI: Receive packet event_code=0xe

D (1441) BT_HCI: btu_hcif_command_complete_evt

D (1441) BT_HCI: Receive packet event_code=0xe

D (1451) BT_HCI: btu_hcif_command_complete_evt

D (1451) BT_HCI: HCI Enqueue Comamnd opcode=0x2006

D (1461) BT_HCI: Receive packet event_code=0xe

D (1461) BT_HCI: btu_hcif_command_complete_evt

D (1471) BT_HCI: HCI Enqueue Comamnd opcode=0x200a

D (1471) BT_HCI: Receive packet event_code=0xe

D (1471) BT_HCI: btu_hcif_command_complete_evt

[... 244 lines cut ...]
D (16501) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 62a27aca9037 not connected
D (16501) BT_SMP: smp_cb_cleanup

D (16511) BT_SMP: result state = SMP_STATE_IDLE

E (16511) BT_BTM: Device not found

I (16521) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (16521) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8
I (16541) SEC_GATTS_DEMO: remote BD_ADDR: 62a27aca9037
I (16541) SEC_GATTS_DEMO: address type = 1
I (16541) SEC_GATTS_DEMO: pair status = fail
I (16551) SEC_GATTS_DEMO: fail reason = 0x66
I (16551) SEC_GATTS_DEMO: Bonded devices number : 0
```

## Other items if possible
Elf file for Gatt Demo project:
[sec_gatts_demo_elf.zip](https://github.com/espressif/esp-idf/files/6229736/sec_gatts_demo_elf.zip)

Bluetooth capture for demo project:
[esp_ble_security_capture.zip](https://github.com/espressif/esp-idf/files/6229728/esp_ble_security_capture.zip)

Bluetooth capture for original project:
[ble_stylo6_capture.zip](https://github.com/espressif/esp-idf/files/6229812/ble_stylo6_capture.zip)

chewhs00, 2021-04-20:
Any feedback on this issue? From the Wireshark capture, after Scan, Feature, and Version req/resp without any problem, then the Master's (mobile phones) Fragment Start packets start to show bad crc and there were no Slave (esp device) responses on all 3 models as specified. The project source code is from 
esp-idf-v4.1.1/examples/bluetooth/bluedroid/ble/gatt_security_server

xulongzhuang, 2021-06-03:
@rj747 
Could you clear all the bound devices on the phone side first, then erase the flash of the esp32 and re-burn the code, and then perform the pairing, will the same result still appear on those phones?

rj747, 2021-06-04:
I tested this on multiple new phones that had no prior paired devices and clearing the bound devices was one of the first things I did when I ran into this issue. The attached program was also loaded onto a fresh dev kit as the issue showed up on a custom board with a ESP32. There should be no issues reproducing this with one of the phones listed and we can ship one of them to you if that would help.

xiewenxiang, 2021-06-09:
@rj747 

The pairing fail is beacause the connection is broken during pairing process. But we can't reproduce this reason by our test phones.

We will purchase these mobile phones.

chewhs00, 2021-07-06:
@xiewenxiang , any update on this? If necessary, we can provide phones for you to test

tomptz, 2021-09-21:
@xiewenxiang any update? We are facing a similar problem.

xiewenxiang, 2021-09-26:
@chewhs00 @tomptz 

This issue has been fixed and will be synchronized out recent.

tomptz, 2021-10-01:
@xiewenxiang will you inform us in this conversion when the fix is available?

xiewenxiang, 2021-10-01:
@tomptz The fix has been submitted and is waiting for merged. It is the National Day holiday at present, so it should be after October 7th.

tomptz, 2021-10-14:
@xiewenxiang any update on this? Is there a development branch that we could test?

chewhs00, 2021-11-05:
> @chewhs00 did you find a solution or workaround for this issue?

As far as I have tested, Nimble seems to work fine with these phones, but not with Bluedroid

AxelLin, 2021-11-05:
> @tomptz The fix has been submitted and is waiting for merged.

@xiewenxiang 
Do you mean this fix?
master: a7247db6b4069c7cc0dd67d3cc03b4e03e72c3ed
v4.4: 3187b9b5a4c2eedcfc2be2b83539f74954d9724f

xiewenxiang, 2021-11-05:
@AxelLin  Yes,  with this commit, can the test pass?

chewhs00, 2021-11-08:
> > @tomptz The fix has been submitted and is waiting for merged.
> 
> @xiewenxiang Do you mean this fix? master: [a7247db](https://github.com/espressif/esp-idf/commit/a7247db6b4069c7cc0dd67d3cc03b4e03e72c3ed) v4.4: [3187b9b](https://github.com/espressif/esp-idf/commit/3187b9b5a4c2eedcfc2be2b83539f74954d9724f)

The fix solves the pairing issue for LG Stylo 6, LG Phoenix 5, and Alcatel 3V.
The fix currently is in release/v4.1 as well . Any scheduled date for the release?

chewhs00, 2022-08-29:
Has this fix applied to v4.3.2? Is this the one?

AxelLin, 2023-05-18:
@xiewenxiang 
v4.3 branch does not include this fix, please backport the fix.
This was reported https://github.com/espressif/esp-idf/issues/8532#issuecomment-1551819583

closed 2024-12-30 by Alvin1Zhang