espressif/esp-idf #6802: how to disable only the RF part in ESP32-S2

edbek, 2021-03-31:
Help, how to disable only the RF part in ESP32-S2 ? This is to reduce the power consumption of the chip.
For this, can I use the command "esp_wifi_set_mode (WIFI_MODE_NULL);" ?

EspHuifeng, 2021-04-01:
The RF part won't work if you don't initialize and start wifi.

negativekelvin, 2021-04-01:
@EspHuifeng it is not clear if they want to disable it after using it or not use it at all. It would be good to know the expected power draw if the WiFi is initialized and inactive vs uninitialized.

edbek, 2021-04-01:
I need to turn it off after initialization and then turn it on again.
Having rummaged a little in the codes, I have so far come to the conclusion that it is necessary to call the function to turn off the WiFi RF-part:

"esp_wifi_stop ()"

Is it so ? Can someone reliably answer this question?

neg2led, 2021-04-02:
Yes, [`esp_wifi_stop();`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv413esp_wifi_stopv) will shut off the RF block in an ESP32-S2. 

If you want to free as many resources as you can, [`esp_wifi_deinit();`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_deinitv) will shut down the entire WiFi manager and free up the heap memory it uses.

Some more info can be found on the [Power Management](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers) page

edbek, 2021-04-02:
neg2led, Thank you !

Alvin1Zhang, 2021-09-02:
Thanks for reporting and sharing the notes, feel free to reopen.