espressif/esp-idf #6804: Coap implementation with cxx example

Mohits2103, 2021-03-31:
Hello all,

I tried to implement [coap_client](https://github.com/espressif/esp-idf/tree/master/examples/protocols/coap_client) to one of the [CXX example](https://github.com/espressif/esp-idf/tree/master/examples/cxx/experimental/sensor_mcp9808)  

I could successfully add the coap components but am getting the following error whe I try to build the project

````
:~/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808$ idf.py build
Executing action: all (aliases: build)
Running make in directory /home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/build
Executing "make -j 6 all"...
[  0%] Built target custom_bundle
[  0%] Built target _project_elf_src
[  0%] Built target partition_table_bin
[  1%] Built target esp32_linker_script
[  1%] Performing build step for 'bootloader'
[  1%] Built target __idf_ulp
[  1%] Built target __idf_esp_serial_slave_link
[  4%] Built target __idf_log
[  5%] Built target _project_elf_src
[  2%] Built target __idf_sdmmc
[  9%] Built target __idf_esp_rom
[ 12%] Built target __idf_esp_common
[  2%] Built target __idf_esp_https_ota
[ 15%] Built target __idf_xtensa
[  3%] Built target __idf_esp_http_server
[ 26%] Built target __idf_esp_hw_support
[  3%] Built target __idf_esp_http_client
[  3%] Built target __idf_tcp_transport
[ 35%] Built target __idf_efuse
[  4%] Built target __idf_esp-tls
[ 61%] Built target __idf_bootloader_support
[ 63%] Built target __idf_spi_flash
[  6%] Built target __idf_nghttp
[ 65%] Built target __idf_micro-ecc
[ 90%] Built target __idf_soc
[  6%] Built target __idf_app_trace
[  6%] Built target __idf_cxx
[ 93%] Built target __idf_hal
[ 95%] Built target __idf_main
[  7%] Built target __idf_newlib
[ 98%] Built target bootloader.elf
[  9%] Built target __idf_freertos
[  9%] Built target __idf_esp_timer
[100%] Built target gen_project_binary
[  9%] Built target __idf_esp_common
[100%] Built target app
[  9%] No install step for 'bootloader'
[ 10%] Built target __idf_esp32
[ 10%] Completed 'bootloader'
[ 10%] Built target __idf_perfmon
[ 10%] Built target bootloader
[ 10%] Built target __idf_xtensa
[ 13%] Built target __idf_driver
[ 13%] Built target __idf_esp_ringbuf
[ 13%] Built target __idf_esp_pm
[ 14%] Built target __idf_esp_hw_support
[ 16%] Built target __idf_soc
[ 17%] Built target __idf_heap
[ 17%] Built target __idf_log
[ 26%] Built target __idf_lwip
[ 28%] Built target __idf_esp_wifi
[ 37%] Built target __idf_wpa_supplicant
[ 38%] Built target __idf_esp_event
[ 39%] Built target __idf_esp_netif
[ 40%] Built target __idf_tcpip_adapter
[ 41%] Built target __idf_esp_eth
[ 42%] Built target __idf_vfs
[ 45%] Built target __idf_hal
[ 46%] Built target __idf_esp_rom
[ 49%] Built target __idf_esp_system
[ 49%] Built target __idf_espcoredump
[ 49%] Built target __idf_esp_gdbstub
[ 49%] Built target __idf_pthread
[ 51%] Built target __idf_nvs_flash
[ 53%] Built target __idf_spi_flash
[ 53%] Built target __idf_esp_ipc
[ 53%] Built target __idf_app_update
[ 55%] Built target __idf_bootloader_support
[ 56%] Built target __idf_efuse
[ 57%] Built target mbedtls
[ 64%] Built target mbedcrypto
[ 65%] Built target mbedx509
[ 66%] Built target __idf_mbedtls
[ 67%] Built target __idf_cbor
[ 67%] Built target __idf_console
[ 67%] Built target __idf_asio
[ 67%] Built target __idf_unity
[ 67%] Built target __idf_qrcode
[ 70%] Built target __idf_coap
[ 70%] Built target __idf_protobuf-c
[ 71%] Built target __idf_esp_adc_cal
[ 71%] Built target __idf_esp_hid
[ 72%] Built target __idf_depthai-spi-api
[ 72%] Built target __idf_esp_websocket_client
[ 72%] Built target __idf_mdns
[ 73%] Built target __idf_expat
[ 73%] Built target __idf_wear_levelling
[ 73%] Built target __idf_jsmn
[ 73%] Built target __idf_json
[ 77%] Built target __idf_freemodbus
[ 78%] Built target __idf_mqtt
[ 79%] Built target __idf_openssl
[ 80%] Built target __idf_spiffs
[ 91%] Built target __idf_libsodium
[ 92%] Built target __idf_experimental_cpp_component
[ 93%] Built target __idf_led_strip
[ 94%] Built target __idf_protocol_examples_common
[ 94%] Built target __idf_cmock
[ 95%] Built target __idf_protocomm
[ 96%] Built target __idf_esp_local_ctrl
[ 97%] Built target __idf_fatfs
[ 98%] Built target __idf_wifi_provisioning
Consolidate compiler generated dependencies of target __idf_main
[ 98%] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/sensor_mcp9808.cpp.obj
In file included from /home/mohit/esp/esp-idf/components/depthai-spi-api/common/esp32_spi_impl.h:21,
                 from /home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:7:
/home/mohit/esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
In file included from /home/mohit/esp/esp-idf/components/coap/libcoap/include/coap2/async.h:18,
                 from /home/mohit/esp/esp-idf/components/coap/port/include/coap/coap.h:27,
                 from /home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:28:
/home/mohit/esp/esp-idf/components/coap/libcoap/include/coap2/net.h:668:6: error: conflicting declaration of 'void coap_ticks(coap_tick_t*)' with 'C' linkage
 void coap_ticks(coap_tick_t *);
      ^~~~~~~~~~
In file included from /home/mohit/esp/esp-idf/components/coap/port/include/coap/coap_dtls.h:14,
                 from /home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:27:
/home/mohit/esp/esp-idf/components/coap/libcoap/include/coap2/coap_time.h:110:6: note: previous declaration with 'C++' linkage
 void coap_ticks(coap_tick_t *t);
      ^~~~~~~~~~
/home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:35:20: warning: 'PREVIEWSTREAM' defined but not used [-Wunused-variable]
 static const char* PREVIEWSTREAM = "spipreview";
                    ^~~~~~~~~~~~~
In file included from /home/mohit/esp/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:9:
/home/mohit/esp/esp-idf/components/depthai-spi-api/spi_api.hpp:14:20: warning: 'dai::NOSTREAM' defined but not used [-Wunused-variable]
 static const char* NOSTREAM = "";
                    ^~~~~~~~
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/sensor_mcp9808.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4754: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2

````

Please help me on this. Thanks in advance.

0xjakob, 2021-04-01:
HI @Mohits2103, could you try to include `coap.h` as `extern "C"`?

E.g. like this:
```
extern "C" {
#include "coap.h"
}

0xjakob, 2021-04-02:
@Mohits2103 In the long run, we will change the coap header files to properly declare functions as `extern "C"`.

Mohits2103, 2021-04-19:
I did the changes as suggested by you and now I am getting this error.

````
home/mohit/Desktop/esp-idf/examples/cxx/experimental/sensor_mcp9808/build/config/sdkconfig.h:93:39: error: invalid conversion from 'int' to 'coap_log_t' [-fpermissive]
 #define CONFIG_COAP_LOG_DEFAULT_LEVEL 4
                                       ^
/home/mohit/Desktop/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:62:40: note: in expansion of macro 'CONFIG_COAP_LOG_DEFAULT_LEVEL'
 #define EXAMPLE_COAP_LOG_DEFAULT_LEVEL CONFIG_COAP_LOG_DEFAULT_LEVEL
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mohit/Desktop/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:203:24: note: in expansion of macro 'EXAMPLE_COAP_LOG_DEFAULT_LEVEL'
     coap_set_log_level(EXAMPLE_COAP_LOG_DEFAULT_LEVEL);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mohit/esp/esp-idf/components/coap/port/include/coap/coap.h:34,
                 from /home/mohit/Desktop/esp-idf/examples/cxx/experimental/sensor_mcp9808/main/sensor_mcp9808.cpp:36:
/home/mohit/esp/esp-idf/components/coap/libcoap/include/coap2/coap_debug.h:65:36: note:   initializing argument 1 of 'void coap_set_log_level(coap_log_t)'
 void coap_set_log_level(coap_log_t level);
                         ~~~~~~~~~~~^~~~~

````

Please someone help me on this

0xjakob, 2021-04-19:
@Mohits2103 I'll come back to you tomorrow. This looks like an incompatibility of the way we use `sdkconfig.h` with C++. Could you try to use a static case around `CONFIG_COAP_LOG_DEFAULT_LEVEL`?
Like `static_cast<coap_log_t>(CONFIG_COAP_LOG_DEFAULT_LEVEL)`.

Mohits2103, 2021-04-19:
Hey @0xjakob, Thank you very much,  It worked great and I was able to build it using  ````static_cast<coap_log_t>(CONFIG_COAP_LOG_DEFAULT_LEVEL)````

I had one doubt, Why do we use "static_cast" (Sorry I am a beginner in ESP-IDF) I tried to do it with just ````<coap_log_t>(CONFIG_COAP_LOG_DEFAULT_LEVEL)```` and it threw error...

Also are there any COAP-Client "POST" example available ? since the coap-client example available is for "GET"

0xjakob, 2021-04-20:
@Mohits2103 Great to hear it works :)

You can imagine `static_cast<coap_log_t>()` as a kind of template function. It takes the target type as template argument (inside the `<...>`). The function parameter is the original data and the return value is the data converted to the target type. Using the angular brackets alone is not valid C++, hence it throws a compile error. This is actually not IDF-specific. You can read more about templates in [cppreference](https://en.cppreference.com/w/cpp/language/templates) website. Although I suggest to find a quick online tutorial for that as well.

Regarding a COAP-Client POST example: I don't know. If you want a follow-up on that, it's probably the best idea to make a forum post at [esp32.com](esp32.com).

0xjakob, 2021-04-20:
@Mohits2103 One more question: Did you put all your headers as `extern "C"` or only the ones referring to coap?

Mohits2103, 2021-04-20:
Hey @0xjakob , Sorry My bad... Shall go through the cppreference website suggested by you.
I have put only headers referring to coap under ```extern "C"```

hmalpani, 2021-09-22:
Hi @Mohits2103, I have updated the internal codebase, it should be updated on GitHub in a couple of days,
Thanks for your contribution in fixing this.