espressif/esp-idf #6807: Contriburiton Guidelines Popup on Github Links Not Useful

cnlohr, 2021-04-01:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  N/A
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-480-g68ce83dff
- Build System:        N/A
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?:  N/A
- Power Supply:         N/A

## Problem Description

When attempting to make a pull-request, the following popup appears linking the contribution guidelines on github.  
[image]

It links this page here: https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/CONTRIBUTING.rst

But many of the links on that page are broken. All of the links using the :prefix: appear to point back at the page but nowhere on the page.

For instance, clicking :idf:`examples' links to here: https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/CONTRIBUTING.rst#id7  (note #id7) 

### Expected Behavior

The github documentation page for contributions can 

### Actual Behavior

### Steps to reproduce

1. Start a new PR for ESP-IDF from a personal repo.
2. Click "Contributing guidelines"
3. Click a link within the "Contributing guidelines" document.

(See problem description for pictures/etc)

### Code to reproduce this issue

N/A

## Debug Logs

N/A

## Other items if possible

N/A

projectgus, 2021-04-05:
Hi @cnlohr,

Thanks for the heads-up. This ReST file is built into the main ESP-IDF docs here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/index.html - which is what all of the extra markup is for. GitHub renders some of it correctly, but clearly not all of it.

I'm not sure what we can do to fix this, I don't think we can override the GitHub generated link, and we do like the fact that the file is (mostly) readable in plaintext in the repo as well as being in the docs.

Something we might be able to do is add some kind of header on the .rst file that links to the "real" docs pages, but isn't rendered into the "real" docs page itself. Although this is also a bit fiddly because we maintain multiple docs versions and we have no way of knowing which branch someone is reading.

Will consult some colleagues and think about this some more.

Alvin1Zhang, 2022-05-06:
Thanks for reporting and sorry for the very slow turnaround, there is a fix under internal reviewing, and the issue will close automatically once the fix is available on GitHub, thanks.

closed 2022-05-15 by espressif-bot